Koenkk/zigbee2mqtt #6281: [RFC] Availability refactor

Koenkk, 2021-02-14:
This issue contains a set of improvements and proposals for the availability feature.
Feel free to provide feedback or suggest any additional improvements.

## Improvements
- [x] Prevent devices always being pinged at exactly the same time (#6133)
- [ ] Investigate possibilty to use reporting for availabilty instead of polling. This would make the availabilty much more scalable for larger networks.
- [ ] Allow to configure availability timeout per device
- [ ] Allow to configure amount of failed checks before device is marked offline
- [ ] Don't mark devices as online after startup but wait until checked
- [ ] Home Assistant: add `availability_mode`

## Configuration
Current:

```yaml
advanced:
    availability_timeout: 60
    availability_blocklist:
        - my_device
    availability_passlist:
        - my_other_device
```

Proposed:

```yaml
availability:
    # Same as advanced.availability_timeout
    timeout: 60
    # Same as advanced.availability_blocklist
    blocklist:
        - my_device
    # Same as advanced.availability_passlist
    passlist:
        - my_other_device
    # New: device will be marked offline after 1 missed timeout
    attempts: 1

devices:
    '0x12345678':
        friendly_name: my_bulb
        availability:
            timeout: 20 # <- New: overrides availability.timeout
            attempts: 2 # <- New: overrides availability.attempts
            # With this config the device will be marked as offline when 2 consecutive availability checks fail.

```

Note: will not be a breaking change, config will automatically be migrated.

## Related issues
- https://github.com/Koenkk/zigbee2mqtt/issues/6188
- https://github.com/home-assistant/core/issues/44671
- https://github.com/Koenkk/zigbee2mqtt/issues/5430

sjorge, 2021-02-14:
Don’t we have part of 2 already, sort of. When a device currently sends a report for one of its attributes it resets the timeout?

I guess for bulbs we could just always configure genOnOff reporting, with a max of the timeout and a min of the timeout - random offset. Unless a remote is bound, then we can perhaps increase the frequency.

We need to add info to this in devices.js though, e.g. hue does not support reporting at all.

Something along the line of an ‘availReportClusterAttribute’, and when under er fall back to the current methode.

Koenkk, 2021-02-15:
> Don’t we have part of 2 already, sort of. When a device currently sends a report for one of its attributes it resets the timeout?

Yes this is already in place, but for some devices like IKEA it's possible to entirely depend on reporting instead of polling.

sjorge, 2021-02-15:
> > Don’t we have part of 2 already, sort of. When a device currently sends a report for one of its attributes it resets the timeout?
> 
> Yes this is already in place, but for some devices like IKEA it's possible to entirely depend on reporting instead of polling.

I think most bulbs aside Hue and Gledopto support genOnOff reporting so that covers a lot, not sure for other routers, the Niko socket/plugs also should support that one as do the Ubisys S1/S2... I guess we need to get a list of candidate targets we could use.

GitOnHub, 2021-03-25:
I've not been following the availability feature too much yet, but on reading through the issue logs I noticed my mental model about battery powered sensor availability differs from the way z2m seems to approach availability now. Admittedly, Z2M has mostly been focusing on mains powered devices.  

Regarding battery-powered sensor availability: I've seen quite consistent timing between check-in messages depending on the model of the sensor. As my limited understanding of the availability feature is now, there seems to be a global setting and a per-device override. (1) Wouldn't it make sense to have a per-model default as well? 

I've also noticed in the past that some of these check-ins don't arrive, so in the following hour(s) availability is uncertain. Is the model of `availability_timeout `and `attempts` therefore the right abstraction for battery powered devices? (2) Maybe (something like) `expected_checkin_interval` and `missed_checkins_allowed` would fit better for them, with defaults per device model and overridable per device.

To illustrate, all my aqara motion sensors check in every 54-55 minutes, while all my aqara temperature sensors are at 54-58 minutes, all aqara contact sensors at 50 minutes and all heiman smoke detectors up to 17 hours. All my devices for most Tradfri models don't check in at all.

Masterz69, 2021-03-26:
IMHO there is a fundamental issue with device availability. Let me illustrate.
At moment Sonoff SNZB-02 still "available" in HA even after 23 (!!!) hours being actually offline. 

I'm not very deep in all that staff, but will share.
Just checked SNZB-02 device "Reporting" section - I see min/max reporting interval for temperature/humidity/battery.
Not sure - is it something gathered from sensor during interview or from some other sources ?

Anyway, at least for SNZB-02, all required to mark device "unavailable" already known: max reporting intervals are known.
If maximum interval reached -> device unavailable.

P.S. Xiaomi devices built quite well, even water leak sensor "ping" hub every ~50 munutes.... Xiaomi hub claim device "offline" after 2 hours. Can I trust dat water leak sensor ? Answer: not. 2 hours of water leakage in case of sensor failure (it will fail, any appliance will fail, question only one - when) - too much.

[image]
^^ SNZB-02 / TH-01

[image]
^^ 
lumi.sensor_wleak.aq1 ^^ some default settings ?

bkupidura, 2021-03-26:
@Masterz69 by design battery powered devices are in deep sleep. AFAIK there is no way to ping and force device to wake.
So if device is reporting every 60m (like aqara), you can only mark device as unavailable after N*60, where N is yours decision how crucial device is.

Masterz69, 2021-03-27:
> @Masterz69 by design battery powered devices are in deep sleep. AFAIK there is no way to ping and force device to wake.
> So if device is reporting every 60m (like aqara), you can only mark device as unavailable after N*60m (N>0).
> 
> In my env i mark devices as unavailable based on device type:
> 
> * default 120m
> * routers 5m (they report every 1m)
> * lights 48h (i dont care)

Good. Let's code it into Z2M integration, please. IMHO it must be in infrastructure level.

Koenkk, 2021-04-17:
I've created a detailed proposal here: https://github.com/Koenkk/zigbee2mqtt.io/blob/availability/docs/information/availability.md

sjorge, 2021-04-17:
> I've created a detailed proposal here: https://github.com/Koenkk/zigbee2mqtt.io/blob/availability/docs/information/availability.md

What do you mean by `Zigbee2MQTT configures the device to check in every 75% of the availability timeout, so by default 3,75 minutes.` specifically?

Is the intent to configure reporting on say genBasic.someatrib?

Koenkk, 2021-04-18:
@sjorge yes, this reduces the stress on the coordinator

sjorge, 2021-04-18:
> @sjorge yes, this reduces the stress on the coordinator

Probably also need a mechanism to override which cluster/attribute and a fallback, afaik hue bulbs don't seem to do any reporting.

Koenkk, 2021-04-18:
In case no reporting comes in the device will be pinged (point 3 of https://github.com/Koenkk/zigbee2mqtt.io/blob/availability/docs/information/availability.md#algorithm-only-applicable-for-routers)

michael-lewis-rft, 2021-04-18:
The proposal looks good. However, would it be possible to add "retry_count" and "retry_interval" whilst you're at it?  
I've noticed others like myself having interest in this since we treat availability='offline' as an indication of tampering, and thus trigger "all the alarms", so loss of a single ping can cause a disproportionate response.

I've created a kludge to cater for this higher up the stack in HA, but fixing it at the point of origin would solve it for everyone.

Koenkk, 2021-04-18:
@michael-lewis-rft this is possible with he proposal. e.g. if you want 5 pings to fail you set the timeout to 5 minutes and ping interval to 1 minute.

timoline, 2021-05-02:
I assume these new settings will not be under advaced, but get's its own section?

And what is going to happen with Availability Blocklist / passlist / blacklist / whitelist, or is this out of scope with this refactor?

Koenkk, 2021-05-02:
@timoline will not be under advanced, disabling availability for certain devices can now be controlled by setting `availability: false` (read comments in https://github.com/Koenkk/zigbee2mqtt.io/blob/availability/docs/information/availability.md#advanced-configuration)

snippem, 2021-08-17:
@Koenkk I wanted to try this but i added :
availability: true
and under advanced: availability_test: true
in the superuser log off home assistant i get the following error.

21-08-17 14:33:16 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability_new' for Zigbee2mqtt Edge (45df7312_zigbee2mqtt_edge)
21-08-17 14:33:16 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2mqtt Edge (45df7312_zigbee2mqtt_edge)

Am i testing the new feature in the right way? or am i doing something wrong.

Koenkk, 2021-08-17:
@snippem this feature is not available for testing through the HA addon, I will add this later.

Koenkk, 2021-08-28:
The initial implementation has been finished and is also available for testing through the HA addon (see OP on how to enable). Please provide your feedback, I will aim to replace the current availability implementation in the 1 October release.

@ciotlosm for the ha addon currently only `availability: true` is allowed however this can also be an object see https://github.com/Koenkk/zigbee2mqtt.io/blob/availability/docs/information/availability.md#advanced-configuration , it it possible to specify this in the `config.json`?

copystring, 2021-08-28:
Allow to configure amount of failed checks before device is marked offline
Is this coming in the future? I really need this.

sjorge, 2021-08-28:
@Koenkk I'll give it another go, in one of the earlier WIP I had all my battery powered contact sensors marked as unavailable. (Coincidentally they also always shows as on the network map without a route, however they do work and report battery every 24-36ish hours) 

I took note with the current implementation and they never showed offline aside from once but that was because one of my S1 switches turned off and killed power to 2 bulbs, they were the closest to the device so it was porbably routed through those, so that one was correct.

Will report back later.

Koenkk, 2021-08-28:
> Allow to configure amount of failed checks before device is marked offline
> Is this coming in the future? I really need this.

The checking mechanism has been changed, first attempt will be light weight, if that fails are more aggressive ping is done (with route discovery, checking network address), please check if with the current implementation devices are still marked offline incorrectly.

@sjorge 

>  in one of the earlier WIP I had all my battery powered contact sensors marked as unavailable. (Coincidentally they also always shows as on the network map without a route, however they do work and report battery every 24-36ish hours)

the timeout is 25 hours, so it's expected they will be marked offline if they report in after > 25 hours, try increasing the timeout for these devices.

copystring, 2021-08-28:
> The checking mechanism has been changed, first attempt will be light weight, if that fails are more aggressive ping is done (with route discovery, checking network address), please check if with the [...]

How would I know devices are incorrectly marked as offline? Since I've been testing the new feature all my ikea routers were marked as offline after a few days. restarting z2m does not make them "online" again.

Koenkk, 2021-08-28:
@copystring the debug logs tell wether ping fails or succeeds, if it fails they are offline.

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

copystring, 2021-08-28:
@Koenkk enabled debugging. The device is actually offline. I have IKEARouter1-4
Replugging IKEARouter3 made that instantly come back online.

I do wonder why there is only 1/1 attempt for the offline device (IKEARouter1) and 1/2 attempts for the online device (IKEARouter3)?

[image]
[image]

Koenkk, 2021-08-29:
@copystring devices which are online will have 2 attempts, where the second attempt is more agressive (with recovery). When a device is already offline only 1 attempt is made to prevent too much stress on the network.

snippem, 2021-08-29:
> @Koenkk I wanted to try this but i added :
> availability: true
> and under advanced: availability_test: true
> in the superuser log off home assistant i get the following error.
> 
> 21-08-17 14:33:16 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability_new' for Zigbee2mqtt Edge (45df7312_zigbee2mqtt_edge)
> 21-08-17 14:33:16 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2mqtt Edge (45df7312_zigbee2mqtt_edge)
> 
> Am i testing the new feature in the right way? or am i doing something wrong.(i have read somewhere if i add those lines i could test this.)

> @snippem this feature is not available for testing through the HA addon, I will add this later.

For the records from what I read in this thread.
For home assistant the configuration for availability is :

availability: true 

To activate the new feature ?
Also I want to now if my settings are correct, because I hope the it resolves the push the button twice 😄 for a light to go on.
Because I still see the same issues in the supervisor logs it  recognizes availability as a unknown option.
(Maybe i am doing something wrong 🤣)

sjorge, 2021-08-29:
@Koenkk small update, so far so good.
I set the active timeout to 15 like I had before, and for the passive timeout I did 28h to give the contact sensors a bit more  wiggle room.

ciotlosm, 2021-09-02:
> @ciotlosm for the ha addon currently only `availability: true` is allowed however this can also be an object see https://github.com/Koenkk/zigbee2mqtt.io/blob/availability/docs/information/availability.md#advanced-configuration , it it possible to specify this in the `config.json`?

This can be an object, but the maximul levels allowed are just two, and this has 3:
```
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
```

This would work with something like:
```
availability:
  active_timeout: 10
  passive_timeout: 1500
```

Koenkk, 2021-09-02:
@ciotlosm how does the configuration writing work? Does the addon overwrite everything in the `configuration.yaml`? Otherwise we can allow just `availability: boolean?` from the config and allow more advanced configuration via the frontend

Koenkk, 2021-10-02:
The reworked availability implementation has now replaced the current one in the dev branch and will be included in the next release (1 November)

timoline, 2021-10-02:
lookslike it broke something in frontend

[image]

Koenkk, 2021-10-03:
@nurikk can you check this: https://github.com/Koenkk/zigbee2mqtt/issues/6281#issuecomment-932731100 ? I believe it is due to the composite type: https://github.com/Koenkk/zigbee2mqtt/blob/fb96551bebd93e57dae4b0ff173f80040ffee2b0/lib/util/settings.schema.json#L30

nurikk, 2021-10-04:
> @nurikk can you check this: [#6281 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/6281#issuecomment-932731100) ? I believe it is due to the composite type:
> 
> https://github.com/Koenkk/zigbee2mqtt/blob/fb96551bebd93e57dae4b0ff173f80040ffee2b0/lib/util/settings.schema.json#L30

Underlying library doesn't support this type of syntax, refactoring to use [oneOf, anyOf, and allOf](https://react-jsonschema-form.readthedocs.io/en/latest/usage/oneof/) will help.
Here is the example
https://github.com/Koenkk/zigbee2mqtt/blob/fb96551bebd93e57dae4b0ff173f80040ffee2b0/lib/util/settings.schema.json#L260

Koenkk, 2021-10-04:
@nurikk thanks, done!

timoline, 2021-10-04:
thx
looking good.....I think
[image]

lolorc, 2021-10-04:
Just tried it, it's not reporting as online a ikea bulb I electricaly switched on after starting z2m. And that despite being able to toggle the bulb with the frontend

Koenkk, 2021-10-12:
@sjorge not that I'm aware of, could you provide the debug log?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

sjorge, 2021-11-05:
Sadly I'm still seeing this, of course not for the short periode where I run with all debugging enabled :( and leaving it all enabled for long just isn't feasable.

```
Zigbee2MQTT:error 2021-11-05 13:23:15: Failed to ping 'outlet/pantry/air_circulator' (attempt 1/1, Read 0x60a423fffef83be6/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
```

I wonder if it is because I have an advanced configuration and not just true set?

```yaml
availability:
  active:
    timeout: 15
  passive:
    timeout: 1680
```

Koenkk, 2021-11-05:
@sjorge with advanced the default values are changed, behaviour isn't so I don't expect this to be the issue.

closed 2021-12-13 by github-actions[bot]