Koenkk/zigbee2mqtt #6284: Please add this to devices.js

agoralive, 2021-02-14:
{
        zigbeeModel: ['4023330P6'],
        model: '4023330P6',
        vendor: 'Philips',
        description: 'Hue white ambiance suspension Amaze',
        meta: {turnsOffAtBrightness1: true},
        extend: hue.light_onoff_brightness_colortemp,
        ota: ota.zigbeeOTA,
    },

agoralive, 2021-02-14:
I have added this because my lamp did not work with this:

https://www.zigbee2mqtt.io/devices/4023330P7.html

sjorge, 2021-02-14:
Do you happen to know the colorTempMin and Max reported by the device?

agoralive, 2021-02-14:
No - This I'm unaware how to get such info, but controls are working "just" like the P7 edition which was already in the settings.js file.....

agoralive, 2021-02-14:
I have setup a node to catch all from the lamp, I will see tomorrow what it's reported.

sjorge, 2021-02-14:
It should show up in the database.db entry, as we read it during the initial configure call.

~ sjorge

agoralive, 2021-02-21:
So this is the thing about the lamp from database.db:

{"id":21,"type":"EndDevice","ieeeAddr":"0x001788010873b29e","nwkAddr":51011,"manufId":4107,"manufName":"Philips","powerSource":"Battery","modelId":"RWL021","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2096,"inCluster$

agoralive, 2021-02-21:
This is everything from the database.db file, I can't see anything :)

{"id":19,"type":"Router","ieeeAddr":"0x00178801067f5e02","nwkAddr":38998,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"4023330P6","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"4023330P6","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20200124","swBuildId":"1.65.9_hB3217DF4"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20200124","swBuildId":"1.65.9_hB3217DF4","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1613905761267}

sjorge, 2021-02-21:
@Koenkk I'm only seeing colorCapabilities ... I guess the release doesn't have the colorTemp reading yet? Or are those skipped for the hue.xxx presets?

Koenkk, 2021-02-21:
Could be that @agoralive is not on the latest zigbee2mqtt (1.17.1), if not update to the latest and reconfigure by clicking the gear icon in the frontend.

agoralive, 2021-02-22:
No I did update, this is why I needed this to be included because I added it between christmas and new year the first time, and after the update I had to do it again :)

In Node-red I do get Color_temp notifications from the lamp, and I am able to control them also.

But has this been added, because the I could try to update once more?

closed 2021-04-02 by github-actions[bot]