Koenkk/zigbee2mqtt #6286: Image is truncated not long enough to contain a valid Tag

885narrow, 2021-02-14:
I tried to update my IKEA Gunnarp lamp. 
It fails with this message.

Zigbee2MQTT:info  2021-02-14 22:28:52: MQTT publish: topic 'zigbee2mqtt/badezimmerlampe/update-state', payload 'available'
Zigbee2MQTT:info  2021-02-14 22:28:52: MQTT publish: topic 'zigbee2mqtt/badezimmerlampe/update_available', payload 'true'
Zigbee2MQTT:info  2021-02-14 22:28:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'badezimmerlampe' failed (Image is truncated: not long enough to contain a valid tag)","meta":{"device":"badezimmerlampe","status":"update_failed"},"type":"ota_update"}'
Zigbee2MQTT:info  2021-02-14 22:28:52: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x60a423fffeeecd67"},"error":"Update of 'badezimmerlampe' failed (Image is truncated: not long enough to contain a valid tag)","status":"error"}'
Zigbee2MQTT:info  2021-02-14 22:28:52: MQTT publish: topic 'zigbee2mqtt/badezimmerlampe/linkquality', payload '28'
Zigbee2MQTT:error 2021-02-14 22:28:52: Update of 'badezimmerlampe' failed (Image is truncated: not long enough to contain a valid tag)

sorry , I am totally new to zigbee2mqtt
thank you very much..

# Bug Report
## What happened
IKEA Gunarp update failed due to truncated file?

## Debug Info
Zigbee2MQTT version: latest ...
Adapter hardware: CC2531

885narrow, 2021-02-15:
Will try to use testserver Ikea

885narrow, 2021-02-15:
[https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/information/ota_updates.md#using-the-ikea-tradfri-test-server](url)

885narrow, 2021-02-15:
advanced:
ikea_ota_use_test_url: true

With this setting it does not start and give an script error.
Any help appreciated.. Thx

Wauxadd, 2021-02-24:
Having same problem with some of my ikea bulbs.
`Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Keittiö Kahvi L"},"error":"Update of 'Keittiö Kahvi L' failed (Image is truncated: not long enough to contain a valid tag)","status":"error","transaction":"nmdve-1"}'
Error Update of 'Keittiö Kahvi L' failed (Image is truncated: not long enough to contain a valid tag)
Debug Received Zigbee message from 'Keittiö Kahvi L', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":553788977,"imageType":16901,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Keittiö Kahvi L', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":553788977,"imageType":16901,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Keittiö Kahvi L', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":553788977,"imageType":16901,"manufacturerCode":4476}' from endpoint 1 with groupID 0 `

Wauxadd, 2021-02-24:
> advanced:
> ikea_ota_use_test_url: true
> 
This worked!

closed 2021-05-03 by github-actions[bot]

JsBergbau, 2021-06-06:
Also for update of TRADFRI bulb E27 WW 806lm it is currently necessary to use
```
advanced:
  ikea_ota_use_test_url: true
```

pklebba, 2021-08-24:
Is IKEA still didn't fix the images?

mmakaay, 2021-08-24:
Yeah, I'm still getting the same error (tried it just now) when not using the test URL.
After switching to the test URL, it works.

Coedy, 2021-10-30:
> WARNING: Use at your own risk!
> 
> advanced: ikea_ota_use_test_url: true

Thanks for this! Have been struggling for ages to get my oldest bulb to update!

flatsiedatsie, 2022-01-06:
My bulbs are updating ok without the test server so far, but I'm seeing this issue with a Tradfri Control Outlet instead (v. 2.0.022).

akentner, 2022-01-26:
I have the same issue with Dresden Elektronik | Mega23M12. 
Any ideas on that device? It's not IKEA, so I can't switch to `ikea_ota_use_test_url`

```
Update of 'terraceroom22_background_highboard' failed (Image is truncated: not long enough to contain a valid tag)
```

gavilaso, 2022-02-07:
when you said to add
advanced:
ikea_ota_use_test_url: true

are you talking to added to the zigbee2mqtt configuration.yaml or to the regular one... 

can you provide a screenshot on how it should look once it added?

pklebba, 2022-02-07:
@gavilaso to the configuation.yaml file:

```
data_path: /config/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
serial:
  port: /dev/ttyACM0
  adapter: deconz
advanced:
  log_level: warn
  ikea_ota_use_test_url: true
frontend:
  port: 8099
```