Koenkk/zigbee2mqtt #6287: Mijia light illuminance sensor, no data sent/received

golddragon007, 2021-02-14:
## What happened
No measurements are received/sent from xiaomi Mijia light illuminance sensor.
I have a small problem with one of my previous device that worked previously. I see this in the bind tab
[image]
it connects to the network, but there's no data (illuminance) sent, while this is how it looks like a working one
[image]
it doesn't seem that it want to work by it's own
[image]
I also tried to add the missing bind, but I get this:
[image]
and this is the only one way that I can press the bind button
[image]
I've also tried to repair, but nothing. The same issue. How can I fix it? Or the device became faulty?

## What did you expect to happen
Receive illuminance data.

## How to reproduce it (minimal and precise)
Pass.

## Debug info
Zigbee2MQTT version: 1.17.1-1
Adapter hardware: cc2652rb
Adapter firmware version: zStack3x0 20201113

closed 2021-03-24 by github-actions[bot]

golddragon007, 2021-04-16:
This is still an existing problem...

vomov, 2021-04-24:
I'm seeing the same issue; three sensors that used to work, do not work anymore. However:

- When Zigbee2MQTT restarts, I get one (old?) message
- When I press and hold the pairing button, I do see 'device announced'. Nothing else, and I cannot seem to change this behaviour (as in; it'll never say 'left network' or 'paired' or anything else.

golddragon007, 2021-05-26:
Ok, a little update, I think this was a bug in the controller firmware, I've updated my controller, and now it started to work again with the previously not working sensor. I don't know if a fix was or the flash process itself solved this issue, but now it seems it works.
And I can confirm previously I had the same that @vomov wrote down in his comment.