Koenkk/zigbee2mqtt #6288: Error Command 1 genLevelCtrl.moveToLevelWithOnOff

kitopopo, 2021-02-14:
Dear friend @Koenkk ,

After some of the last updates I have been observing the following error in the zigbee2mqtt log:

```
Feb 14 22:49:48 raspberrypi npm[1596]: Zigbee2MQTT:error 2021-02-14 22:49:48: Publish 'set' 'state' to 'lamps' failed: 'Error: Command 1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}) failed (Data request failed with error: 'MAC channel access failure' (225))'

feb 14 22:49:48 raspberrypi npm[1596]: Zigbee2MQTT:info  2021-02-14 22:49:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'lamps' failed: 'Error: Command 1 
genLevelCtrl.moveToLevelWithOnOff({\"level\":254,\"transtime\":10}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"lamps"},"type":"zigbee_publish_error"}'

```
Lamps is a group with my 4 bulbs philips hue White and Color Ambiance GU10 (929001953101)

When the problem occurs, the orders sent to my philips bulbs seem to go very slow and when I make a network map I get the LQI error:

```
feb 14 22:51:03 raspberrypi npm[1596]: Zigbee2MQTT:error 2021-02-14 22:51:03: Failed to execute routing table for 'Denon plug'
feb 14 22:51:14 raspberrypi npm[1596]: Zigbee2MQTT:error 2021-02-14 22:51:14: Failed to execute LQI for 'Router plug'
feb 14 22:51:24 raspberrypi npm[1596]: Zigbee2MQTT:error 2021-02-14 22:51:24: Failed to execute routing table for 'Router plug'
feb 14 22:51:24 raspberrypi npm[1596]: Zigbee2MQTT:info  2021-02-14 22:51:24: Network scan finished
```
 I only manage to solve the problem in 2 ways:

- physically disconnecting the CC2531 from the usb and reconnecting it
- completely turning off my raspberry (removing power). If I reset my raspberry the problem persists.

any idea what is this bug so annoying?

I will do more tests to see if I get more data

thanks in advance

## How to reproduce it (minimal and precise)

At the moment I don't know how to reproduce the problem, it is random and it seems to happen when I turn on or off the power to the philips hue.

## Debug info
Zigbee2MQTT version: version 1.17.1-dev
Adapter hardware: CC2531

Koenkk, 2021-02-15:
`MAC channel access failure' (225))` indicated there is too much interference, see https://www.zigbee2mqtt.io/information/FAQ.html#common-error-codes

kitopopo, 2021-02-19:
thanks @Koenkk ,

I had near of zigbe2mqtt stick a sonoff 1ch diy , after Ward off the sonoff the mac error disappear, thanks very much

closed 2021-02-19 by kitopopo