Koenkk/zigbee2mqtt #6291: unsupported device with Zigbee model 'TS0014' after update

ttocbvn, 2021-02-15:
## What happened
Hi !
I updated the Zigbee2mqtt Edge add-on 1.17.1 on hassio
When I repair the device, model "TS0014" (Tuya Switch 4 gang) is not supported
I see the new "devices.js" file that supports the "TS0014" model, not my device
earlier I used by editing "devices.js" with "TS0004" model
How do support my device
sorry for my english
Zigbee2mqtt Edge log: https://pastebin.com/qYWZCk8q
database.db: https://pastebin.com/1aRnJmyu
## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2531

Koenkk, 2021-02-15:
Added back your device.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-15 by Koenkk

WildeRNS, 2021-04-16:
hello, please add one more manufacturer to switch:
_TZ3000_r0pmi2p3
this is 4 gang switch with neutral: https://a.aliexpress.com/_msQqvwj

kbecao85, 2021-06-14:
@Koenkk https://pt.aliexpress.com/item/4000908029622.html, this is 4 gang switch with neutral.

jocamane, 2022-01-06:
> hello, please add one more manufacturer to switch: _TZ3000_r0pmi2p3 this is 4 gang switch with neutral: https://a.aliexpress.com/_msQqvwj

this switch work now ?
My don't work ok
Thanks

WildeRNS, 2022-01-06:
> > hello, please add one more manufacturer to switch: _TZ3000_r0pmi2p3 this is 4 gang switch with neutral: https://a.aliexpress.com/_msQqvwj
> 
> this switch work now ? My don't work ok Thanks

Yes, my work good

jocamane, 2022-01-06:
> > > hello, please add one more manufacturer to switch: _TZ3000_r0pmi2p3 this is 4 gang switch with neutral: https://a.aliexpress.com/_msQqvwj
> > 
> > 
> > this switch work now ? My don't work ok Thanks
> 
> Yes, my work good

ok, thanks. I try again in weekend.

jocamane, 2022-01-08:
I test again e my don't work with conbee ii

jocamane, 2022-01-08:
Device '0x5c0272fffeae7628' with Zigbee model 'TS0014' and manufacturer name '_TZ3000_r0pmi2p3' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0014' and manufacturer name '_TZ3000_r0pmi2p3'