Koenkk/zigbee2mqtt #6292: Cannot find 'data/tuya.dump.txt' mentioned in guide

Faustino-ai, 2021-02-15:
## What happened
I couldn't find 'data/tuya.dump.txt' as mentioned in the guide: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html

## What did you expect to happen
After including 'fz.tuya_data_point_dump ' line in devices.js as mentioned in the guide, restarting zigbee2mqtt should have some kind of messages being appended in the text file tuya.dump.txt (according to the guide, this file should exist in the data folder). 

## How to reproduce it (minimal and precise)
Doing everything exactly the same as shown in the guide.
## Debug info

Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2531
Adapter firmware version: 20201127

closed 2021-03-26 by github-actions[bot]

li-soft, 2021-05-04:
I just faced with the same issue. I have HA hosted on the Raspberry (no docker or any other vm). So my story is that I bought some thermostatic valves with Unitec label which seems to be unsupported. However I'd like to be this good one ;) and discover the device exposed dp's.

So following the instructions I have added 'data/tuya.dump.txt' in to my converter and then

I got:
`Zigbee2MQTT:info  2021-05-04 21:28:58: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffe855a2e', payload '{"away_mode":"OFF","child_lock":"LOCKED","current_heating_setpoint":6,"linkquality":60,"local_temperature":22.2,"preset":"schedule","system_mode":"auto"}'
/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:3101
                if (err) throw err;
                         ^
[Error: ENOENT: no such file or directory, open 'data/tuya.dump.txt'] {
  errno: -2,
  code: 'ENOENT',
  syscall: 'open',
  path: 'data/tuya.dump.txt'
}`

under the zigbe20mqtt directory I have created 'data/tuya.dump.txt' file thinking that it may help - however with no luck ...

Faustino-ai, 2021-05-05:
Your issue isn't as similar as mine though. Mine never mentioned any directory at all. Just being there silent.

magicmatt007, 2021-05-18:
I have the same problem...

magicmatt007, 2021-05-19:
With a bit of trial and error, I was able to get it running on a HASSIO based Zigbee2Mqtt plug in:

Fix was to add "/" before "data/tuya.dump.txt":

With the plug in running, 

- SSH into zigbee2MQTT container: docker exec -it $(docker ps -f name=addon_45df7312_zigbee2mqtt -q) bash
- Then locate fromZigbee.js  (app/... Converters/.... ...herdsman converteres..... And add a "/" in front of data in the line "fs.appendFile('data/tuya.dump.txt', dataStr, (err)"    --> "fs.appendFile('/data/tuya.dump.txt', dataStr, (err)"
- Then save and exit the file
- cd /data
- tuya.dump.txt is visible there

Note: This is all a bit cumbersome. Not sure, what would need to be done to safe the file in Hassio's config folder for easier access....

flatsiedatsie, 2022-01-02:
I encountered the same issue. I expected the `data/tuya.dump.txt` file to be relative to the location of `configuration.yaml`. In my case that configuration file is not in the standard location.

Faustino-ai, 2022-01-04:
> I encountered the same issue. I expected the `data/tuya.dump.txt` file to be relative to the location of `configuration.yaml`. In my case that configuration file is not in the standard location.
> 
> More details here: [Koenkk/zigbee-herdsman-converters#3303](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3303)

Unfortunately, door lock works so differently compared to the rest that this implementation will not work. So there will be no data available by using this method. I've tested on other several tuya products and everything works as intended. Just not door locks. It seems like they are using a very different approach for the door lock and so far I didn't see anyone successfully reverse engineer the zigbee version.

james-fry, 2022-09-10:
> With a bit of trial and error, I was able to get it running on a HASSIO based Zigbee2Mqtt plug in:
>
> Fix was to add "/" before "data/tuya.dump.txt":
>
> With the plug in running,
>
> * SSH into zigbee2MQTT [...]

Thanks that - works well. I used to use the hassio portainer addon to connect to running containers, but now that has been removed from the public addons repo. This is a bit more long winded but works just as well. I used the /share/zigbee2mqtt/ path to allow access to the file from samba.

mikeg-de, 2022-11-16:
> With a bit of trial and error, I was able to get it running on a HASSIO based Zigbee2Mqtt plug in:
>
> Fix was to add "/" before "data/tuya.dump.txt":
>
> With the plug in running,
>
>     * SSH into [...]

The issue is still persistent. Interestingly it once worked, then I adjusted the converter for testing purposes but reverted which then resulted int he issue. I had to log into SSH on port 22222 (bcs using hassio) and found two files:

```
# find / -type f -name "fromZigbee.js"
/mnt/data/docker/overlay2/1d9353452fdb44db420bc90ebd3e31c94526c3ec10d645ea8ab65686a9586000/diff/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js
/var/lib/docker/overlay2/1d9353452fdb44db420bc90ebd3e31c94526c3ec10d645ea8ab65686a9586000/diff/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js
# vi /var/lib/docker/overlay2/1d9353452fdb44db420bc90ebd3e31c94526c3ec10d645ea8ab65686a9586000/diff/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js
```

Adjusted only the second and it worked. Guess the mnt directory is a symlink.