Koenkk/zigbee2mqtt #6295: IKEA LED1837R5 erros bit still 'works'

heggink, 2021-02-15:
## What happened
After a while, some IKEA light will timeout when switched but still function (the command times out but the lamp still responds). Have seen this with both LED1837R5 and LED1732G11. Same lamps, if switched through group/remote also still work.
Only re-pairing fixes the issue. Restartsing z2m makes no difference.

## What did you expect to happen
No timeout error in the log

## How to reproduce it (minimal and precise)
Wait for a while (couple of days typically) until errors for a light appear in the log

## Debug info
Zigbee2MQTT version: latest dev (updated this morning)
Adapter hardware: zzh with latest router FW (CC26X2R1)
Adapter firmware version: latest

Log:
```
Debug Received MQTT message on 'zigbee2mqtt/Buitenlamp 1/set' with data '{"state": "ON"}'
Error Publish 'set' 'state' to 'Buitenlamp 1' failed: 'Error: Command 0xec1bbdfffea4e6f8/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53466 - 1 - 12 - 6 - 11 after 10000ms)'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Buitenlamp 1' failed: 'Error: Command 0xec1bbdfffea4e6f8/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 53466 - 1 - 12 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"Buitenlamp 1"},"type":"zigbee_publish_error"}'
```

Koenkk, 2021-02-15:
Can you provide a sniff of this? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

heggink, 2021-02-15:
Will do. I don;t have a sniffer lying around so just ordered one. Should arrive this weekend. Any suggestion on what to sniff specifically or just all traffic from/to this device?

Koenkk, 2021-02-15:
Start sniffing just before you send the command, wait 60 seconds and stop. Based on this I can filter and see what goes wrong.

MattWestb, 2021-02-16:
Its very likely the famous Silabs stack bug that is fixed in 6.7.8.0 (parent accouterments is crashing one buffet) but not implemented in the IKEA and all other Silabs (Smartthing, Tuya, Sonoff and so on) firmware.

heggink, 2021-02-19:
[ikea issue capture.zip](https://github.com/Koenkk/zigbee2mqtt/files/6010011/ikea.issue.capture.zip)
This should be it (I hope ;-)). I followed your instructions on how to sniff traffic. Installed zboss and wireshark. changed channel to 11 and pressed start. Saw wireshark being opened. Switched a light that errored out yesterday, waited another 60secs after domoticz reported the error and then stopped. Then went to wireshark and pressed save as -> pcap file. zipped and attached here.

Koenkk, 2021-02-19:
Can you also provide your network key? Otherwise I cannot decrypt the sniff.

heggink, 2021-02-19:
```
  network_key:
    - 208
    - 233
    - 105
    - 65
    - 218
    - 3
    - 211
    - 19
    - 161
    - 9
    - 247
    - 95
    - 99
    - 39
    - 198
    - 179
```
like this?

Koenkk, 2021-02-19:
Looks like the problem is indeed on the side of the bulb:
- Frame 48: Zigbee2MQTT sends the ON command, no acknowledgment is received.
- Frame 64: command timed out, Zigbee2MQTT tries to discover the route to the bulb
- Frame 65: this is the strange part, bulb responds with the route, there is a direct connection between the coordinator and the bulb (no hops/routers in between).
- Later zigbee2mqtt sends the ON command again, but no response again.

[image]

Updating your bulb may solve the issue: https://www.zigbee2mqtt.io/information/ota_updates.html

heggink, 2021-02-19:
Thanks for checking this out! Unfortunately, this happens with multiple (types) lamps and none of them have updates :-(.
Ah well, just wait until IKEA releases an update I guess. Until then, unplugging is the only fix.
Good w-e!

closed 2021-03-30 by github-actions[bot]