Koenkk/zigbee2mqtt #6296: IKEA E1812 does not change from ON

tibbors, 2021-02-15:
## What happened: IKEA E1812 does not change state

## What did you expect to happen: Change from on to off (and vice versa)

## How to reproduce it (minimal and precise):

- Pair IKEA E1812 with Zigbee2MQTT
- observe that the state does not change from "on"

## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2531

Geertvdc, 2021-02-27:
Same issue here.

Pairing the Tradfi Shortcut button (E1812) worked fine. but i just dont see ANY events happening when i press the button.

```
Zigbee2MQTT:info  2021-02-27 11:10:14: Device '0x680ae2fffe31aab8' joined
Zigbee2MQTT:info  2021-02-27 11:10:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe31aab8","ieee_address":"0x680ae2fffe31aab8"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-02-27 11:10:14: Starting interview of '0x680ae2fffe31aab8'
Zigbee2MQTT:info  2021-02-27 11:10:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe31aab8","ieee_address":"0x680ae2fffe31aab8","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-02-27 11:10:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x680ae2fffe31aab8"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-02-27 11:10:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x680ae2fffe31aab8"},"type":"pairing"}'
Zigbee2MQTT:info  2021-02-27 11:10:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe31aab8","ieee_address":"0x680ae2fffe31aab8"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-02-27 11:10:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x680ae2fffe31aab8"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-02-27 11:11:51: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe319af6', payload '{"battery":100,"linkquality":44,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-02-27 11:12:20: Successfully interviewed '0x680ae2fffe31aab8', device has successfully been paired
Zigbee2MQTT:info  2021-02-27 11:12:20: Device '0x680ae2fffe31aab8' is supported, identified as: IKEA TRADFRI shortcut button (E1812)
Zigbee2MQTT:info  2021-02-27 11:12:20: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"TRADFRI shortcut button","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["on","brightness_move_up","brightness_stop"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E1812","supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x680ae2fffe31aab8","ieee_address":"0x680ae2fffe31aab8","status":"successful","supported":true},"type":"device_interview"}'
```

Info: Zigbee2mqtt version:  1.17.1-4
Hardware: cc2531

I'm  running zigbee2mqtt in home assistant. an entity was created automatically and linkquality and battery are working fine.

I have 2 of these buttons and both have exactly the same issue. 

any ideas?

antti365, 2021-03-24:
@Koenkk I believe that since the button supports group binding, the device's "on" value should be replaced with "toggle", then this button can be used with sockets and lamps, in such a form as now I can not think of how to use it. ..
The second question - binding to the group turned out to be done without removing the device from the default group (901), in contrast to the Ikea switch - should it be so? On the Bind tab in the frontend, the clusters connected to the default group differ from the switch settings, for example the PowerCfg cluster is connected to the default group instead of the coordinator.

Koenkk, 2021-03-24:
@antti365 this device is indeed not bound to the default group (not needed for this device). I've already ordered it and will look into it.

MattWestb, 2021-03-25:
Its sending one on command on on /off cluster with unicast and / or broadcast (depends of cluster bindings).

If like it sending toggle you need doing on custom firmware for it ;-)

antti365, 2021-03-25:
@MattWestb I have created a pair of this button with an outlet using bindings and the button only turns it on, it does not turn it off.  I can't think of an application for this functionality, it's nonsense ... I hope Koen will come up with something, otherwise I'll have to hand it back to the store.

MattWestb, 2021-03-25:
I was still writing its only sending on **on** command = **not** seeding **toggle** so still  you need writing on new firmware if you like chaining function from one scene button to one On/Off button.
From wire shark:
```
Frame 5771: 112 bytes on wire (896 bits), 112 bytes captured (896 bits) on interface \Device\NPF_Loopback, id 0
Null/Loopback
Internet Protocol Version 4, Src: 127.0.0.1, Dst: 127.0.0.1
User Datagram Protocol, Src Port: 17754, Dst Port: 17754
ZigBee Encapsulation Protocol, Channel: 20, Length: 48
IEEE 802.15.4 Data, Dst: 0x0304, Src: 0xd79d
ZigBee Network Layer Data, Dst: 0x0304, Src: 0xd79d
    Frame Control Field: 0x2248, Frame Type: Data, Discover Route: Enable, Security, End Device Initiator Data
    Destination: 0x0304
    Source: 0xd79d
    Radius: 30
    Sequence Number: 74
    [Extended Source: SiliconL_ff:fe:d0:9f:82 (ec:1b:bd:ff:fe:d0:9f:82)]
    [Origin: 3324]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 54
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
    Sequence Number: 14
    Command: On (0x01)
```
You need changing the last command `on` (0x01) to `toggle` (0x02).

But i think its easier and less expensive baying on On/Off button for 6€ then the Silab devkit is costing 100€ for getting access to the Zigbee stack.

But you can always making some "magic automatons" that is only working the your system is 100% on line.

Koenkk, 2021-03-26:
Received my E1812 today:

- Initially pairing failed (interview failed), this was due to a bug in zigbee-herdsman, this has been fixed now. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). If you don't get actions or battery, try force removing the device via the frontend and repair.
- The remote only sends the following zigbee command:
  - Single press: `on`
  - Hold: `brightness_move_up`
  - Hold release: `brightness_stop`
  - The remote doesn't send any other zigbee commands like `off` or `brightness_move_down` (I guess this makes sense as the button was not made for this purpose)
- Since it only sends these commands, binding it to an other device makes no sense (since it cannot dim the brightness or turn a device off)

tibbors, 2021-03-26:
I am a noob at this. Does this mean that the device doesn‘t send any other commands? Why would IKEA do this? I am confused.

MattWestb, 2021-03-26:
@Koenkk Take one look on de(F)CONZ  then its little tricky then like sending multiple commands the being bonded or sending uni and broadcasts.

Its more or less the same problems with syfonisk and the new Styrbar https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4572

Koenkk, 2021-03-26:
> I am a noob at this. Does this mean that the device doesn‘t send any other commands? Why would IKEA do this? I am confused.

Because it's not meant to control lights, for this use: https://www.zigbee2mqtt.io/devices/E1743.html

@MattWestb it looks to be working correctly

sjorge, 2021-03-26:
I think IKEA kept it simple for them selves and used the same firmware as the E1743 but just the 'top' side of the button hardware is implemented on the E1812.

I'd be honest, I'd expect it to send a scene_recall's with different ID's for short and long press, but I guess the are implementing them purely in software (shortcuts).

MattWestb, 2021-03-26:
The hardware is only little modified from the E1743 (one button in the middle of the PCB) but the firmware is more one Syponisk / Styrbar and its being paired as one "shortcut button" and not on normal controller in IKEA GW.
Its sending both unicast and broadcast if binding one group (if binding more cluster is sending more commands) as the other 2 do but the old is not doing.

I think its 100% software but i was not trying configuring scenes in the IKEA GW then i was sniffing it.

So one symponisk with our turning sensor ;-)

877dev, 2021-03-28:
> Received my E1812 today:
> 
> * Initially pairing failed (interview failed), this was due to a bug in zigbee-herdsman, this has been fixed now.

@Koenkk would this also affect IKEA E1743?
Mine always fails interview in 1.8.1 master branch, 

Will you be adding this fix to the next release?
Thanks

Koenkk, 2021-03-28:
@877dev I don't know, but since they seem to share almost the same FW it could be.

Fix will be in the next release (on 1 april)

ratsoid, 2021-04-01:
I updated to the latest version but I'm experiencing the same issue that @Geertvdc mentioned (nothing happens when clicking)

Koenkk, 2021-04-01:
@ratsoid try force removing the device via the frontend en repair it.

antti365, 2021-04-02:
I wrote a letter with a proposal to update the firmware in support of Ikea, let's see what they answer.  Otherwise, you will have to forget about this device ...

877dev, 2021-04-05:
> Fix will be in the next release (on 1 april

Yup it's working now, thanks!

closed 2021-04-05 by Koenkk