Koenkk/zigbee2mqtt #6297: Siren outdoor SMaBiT (Bitron Video) AV2010/29A

samuprofe, 2021-02-15:
Siren outdoor SMaBiT (Bitron Video) AV2010/29A
https://manuals.smabit.eu/av2010_29a.html

Interesting device since there are few (or no) outdoor sirens that support ziggbee. Following the steps of https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html and doing multiple tests I have managed to connect to zigbee2mqtt by adding the following to devices.js (in home assistant):

```
      {   
		zigbeeModel: ['AV2010/29A'], 
		model: 'AV2010/29A', 
		vendor: 'SMaBiT', 
		description: 'SMaBiT Zigbee Outdoor Siren', 
		supports: 'warning',
        fromZigbee: [fz.battery, fz.ias_contact_alarm_1, ],
        toZigbee: [tz.warning],
        meta: {disableDefaultResponse: true, configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
			await reporting.batteryVoltage(endpoint);
        },
        exposes: [e.warning(), e.battery(), e.contact(), e.battery_low(), e.tamper()],
	
	},
```

I have also discovered the following:
```
topic: 'zigbee2mqtt /friendly name/ set'

//Turn on the siren and lights, duration can be indicated in seconds. The strobe parameter must always be false for the siren to turn on:
payload: '{"warning": {"duration": 3, "strobe": false}}'

//5 short beeps. If the alarm is on, it stops:
payload: '{"warning": {"duration": 0}}'
payload: '{"warning": {"level": "medium"}}'

//5 long beeps. If the alarm is on, it stops:
payload: '{"warning": {"level": "low"}}'
payload: '{"warning": {"level": "high"}}'

//1 short beeps. If the alarm is on, it stops:
payload: '{"warning": {"level": "very_high"}}'
```

I have not got:
- Deactivate tamper contact (to change the battery for example)
- Get the battery % (I do not know if it is possible, at the moment it says that the battery is 0% and that battery_low is false with new batteries)
Any clues how to try to do it?

I don't know if what I put in the configure section is correct (I have copied it from other devices) in case someone can review it before adding it to the documentation.

Thanks.

database.db entry of the device
`{"id":29,"type":"EndDevice","ieeeAddr":"0x00124b00207f60d6","nwkAddr":16557,"manufId":0,"manufName":"SMaBiT","powerSource":"Battery","modelId":"AV2010/29A","epList":[1,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,3,1280,1282],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"AV2010/29A","manufacturerName":"SMaBiT","powerSource":3,"zclVersion":1,"appVersion":11,"hwVersion":0,"dateCode":"20170421        "}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001949c1df","zoneState":1}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001949c1df","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":1026,"inClusterList":[],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":11,"hwVersion":0,"dateCode":"20170421        ","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1613384672295}`

what I get from the device: 
ac_status: false
battery_low: false
contact: true
enrolled: false
linkquality: 49
restore_reports: false
smoke: false
supervision_reports: true
tamper: false
trouble: false

Koenkk, 2021-02-15:
> Deactivate tamper contact (to change the battery for example)

Do you know if this is possible via the original gateway? I've never seen it before for a Zigbee device

> Get the battery % (I do not know if it is possible, at the moment it says that the battery is 0% and that battery_low is false with new batteries)

I don't think it's possible for this the device should have `genPowerCfg` (number ` 1`) in the `inClusterList` but it's not. I think it only support `battery_low`.

Question: why did you use `ias_contact_alarm_1` ? This device doesn't provide `contact` (like a window/door sensor)?

samuprofe, 2021-02-15:
> > Deactivate tamper contact (to change the battery for example)
> 
> Do you know if this is possible via the original gateway? I've never seen it before for a Zigbee device

I think so from what it says in the manual: "Sabotage alert: Opening the siren connected to the smart home system triggers a sabotage alarm. To prevent the alarm from triggering, whether on installation, make sure the tamper contact (4) plug is not connected to the main board, or remove the siren from the configuration of your Smart Home system before opening it."

> I don't think it's possible for this the device should have `genPowerCfg` (number ` 1`) in the `inClusterList` but it's not. I think it only support `battery_low`.

ok, thank you
 
> Question: why did you use `ias_contact_alarm_1` ? This device doesn't provide `contact` (like a window/door sensor)?

I don't really know, I must have copied it from some other siren or device. I understand that I have to delete it.

Koenkk, 2021-02-15:
> I think so from what it says in the manual: "Sabotage alert: Opening the siren connected to the smart home system triggers a sabotage alarm. To prevent the alarm from triggering, whether on [...]

You will have to sniff traffic with the original gateway to support this (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

I've added the config. Can you check if it works correctly?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

samuprofe, 2021-02-16:
It works with dev version. Thank you.

I don't have the original gateway, sorry.

I'll add the information on the documentation.

Regards and thanks again.

closed 2021-02-16 by Koenkk

mgabetta, 2021-07-06:
Ehi!
thanks for integrating this! I've received it today and it works as described in @samuprofe messages.
Are there any updates on how to use it at full extent (tone level, tone type for example)?

samuprofe, 2021-07-09:
> Ehi!
> thanks for integrating this! I've received it today and it works as described in @samuprofe messages.
> Are there any updates on how to use it at full extent (tone level, tone type for example)?

I'm glad it works. I only found out what I have already commented and put in the documentation. But if you discover something please comment here.

Greetings.

Rockman18, 2021-07-15:
Hello, thanks for your work, I've received mine last week and it works well to alert.
But the attributes tamper and battery_low are always N/A. Is it the same for you ?

[image]

mgabetta, 2021-07-15:
[image]
 yes, unfortunatelly!

mgabetta, 2021-07-16:
Do you think that trying to reach out the manufacturer to have some detail on how the siren is supposed to be used by BRIDGE/Gateway could be useful?

Rockman18, 2021-07-20:
We can have a try, it could consider some benefit to be on open source platforms to sell more devices, as it is the only zigbee siren on the market as far as I know.

Rockman18, 2021-07-22:
@Koenkk : In the logs for this device we can sometime find : 
`No converter available for 'AV2010/29A' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":20}'`

Any idea what it could be ?

mgabetta, 2021-09-01:
> We can have a try, it could consider some benefit to be on open source platforms to sell more devices, as it is the only zigbee siren on the market as far as I know.

What I wrote was:

`The support is only partial because it is possible to operate only a limited number of actions, as described in the "Controlling" section here https://www.zigbee2mqtt.io/devices/AV2010_29A.html#controlling `

I just received an answer, unfortunately I'm not enough skilled in Z2M to understand if it is useful...
Here's what SMaBiT Customer Service replied:

`When looking at the Controlling section, the description is actually quite confusing. The siren - just as also the smoke
detector 24A which can be used as a indoor siren - uses the WD cluster (warning device) which allows you to control 
all the functions of the siren as explained in the "exposed" section.
In case your request is still of interest, could you please - considering what I have written about the WD cluster - describe
 more in detail what you need and I will pass this on to development?
`
Can anybody help?

Rockman18, 2021-09-01:
@mgabetta : Before your message i've commited a new converter for AV2010/29A on the dev branch. I think battery and tamper will work.
If you are on the dev branch, try to pair your device again. If not, this should be released at the start of september and deployed really soon after.

mgabetta, 2021-09-02:
Great! I will try it ASAP.
Regarding "Warning", is there any update? I'm referring to the possibility of tuning "mode", "level", "strobe_level" etc. as they are meant to be controlled and not as reported in https://www.zigbee2mqtt.io/devices/AV2010_29A.html#controlling (which seems as a workaround)

Rockman18, 2021-09-02:
I think i understand why our device is strange : https://github.com/Koenkk/zigbee2mqtt/issues/8310

mgabetta, 2021-10-01:
@Rockman18 any updates on this? especially related to issue #8310

Rockman18, 2021-10-01:
A fix was published at the beginning of September and for home assistant the release occurs once a month so it will be released during the next days.