Koenkk/zigbee2mqtt #6299: please add support for TS0504B (v similar to TS0503B) LED strip controller

netweaver1970, 2021-02-15:
As I couldn't find this device in the latest Dev z2m, nor in the github issue list, I've put the request here:
Original source: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues/561 

**Device description**
Tuya TS0504B led strip controller

**Zigbee2mqtt page**
https://www.zigbee2mqtt.io/devices/TS0503B.html
Note: The TS0504B it's similar to TS0503B but it also includes the while light (RGBW)

{
fingerprint: [{modelID: 'TS0504B', manufacturerName: '_TZ3000_ukuvyhaa'}],
model: 'TS0504B',
vendor: 'TuYa',
description: 'RGBW LED controller',
extend: preset.light_onoff_brightness_colortemp_colorxy(),
},

**MQTT message with device information**
MQTT publish: topic 'zigbee2mqtt/Led-Strip-Controller', payload '{"color":{"x":0.3125,"y":0.32894736842105265},"color_temp":157,"last_seen":"2021-02-03T09:54:27.860Z","linkquality":0,"state":"ON"}'

**MQTT messages from device**
Put here examples of MQTT messages with different states you've received from device, as example

```
info MQTT publish: topic 'zigbee2mqtt/Led-Strip-Controller', payload '{"color":{"x":0.3125,"y":0.32894736842105265},"color_temp":157,"last_seen":"2021-02-03T09:58:11.740Z","linkquality":15,"state":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Led-Strip-Controller', payload '{"color":{"x":0.3125,"y":0.32894736842105265},"color_temp":157,"last_seen":"2021-02-03T09:58:11.755Z","linkquality":15,"state":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Led-Strip-Controller', payload '{"brightness":63.75,"color":{"x":0.3125,"y":0.32894736842105265},"color_temp":157,"last_seen":"2021-02-03T09:58:13.412Z","linkquality":15,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/Led-Strip-Controller', payload '{"brightness":63.75,"color":{"x":0.3125,"y":0.32894736842105265},"color_temp":157,"last_seen":"2021-02-03T09:58:13.438Z","linkquality":15,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/Led-Strip-Controller', payload '{"brightness":63.75,"color":{"x":0.526676280311873,"y":0.41329727450763},"color_temp":500,"last_seen":"2021-02-03T09:58:14.865Z","linkquality":15,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/Led-Strip-Controller', payload '{"brightness":63.75,"color":{"x":0.4513888888888889,"y":0.4652777777777778},"color_temp":254,"last_seen":"2021-02-03T09:58:16.204Z","linkquality":15,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/Led-Strip-Controller', payload '{"brightness":63.75,"color":{"x":0.41397849462365593,"y":0.5},"color_temp":294,"last_seen":"2021-02-03T09:58:20.560Z","linkquality":21,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/CC2530-Router', payload '{"last_seen":"2021-02-03T09:58:22.152Z","led":true,"linkquality":47}'
info MQTT publish: topic 'zigbee2mqtt/Led-Strip-Controller', payload '{"brightness":63.75,"color":{"x":0.3125,"y":0.3289473
```

Koenkk, 2021-02-15:
Done!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-15 by Koenkk

netweaver1970, 2021-02-16:
mmm, strange ...
I've recreated the z2m container, with retrieval of last image, again this morning but that LED strip RGBW controller TB0504B, but the device is still unknown after deleting & re-pairing.

my database.db entry for this controller.
`{"id":66,"type":"Router","ieeeAddr":"0x60a423fffe113527","nwkAddr":506,"manufId":4098,"manufName":"_TZ3000_ukuvyhaa","powerSource":"Mains (single phase)","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"manufacturerName":"_TZ3000_ukuvyhaa","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613461670932}`

Koenkk, 2021-02-16:
Can you try repairing  (first remove the device) ? I'm missing the modelID from the database, perhaps some hickup during pairing.

netweaver1970, 2021-02-17:
I did multiple times already, same result (unsupported) every time but at least now the modelId is filled in the database: 

`{"id":66,"type":"Router","ieeeAddr":"0x60a423fffe113527","nwkAddr":43606,"manufId":4098,"manufName":"_TZ3000_ukuvyhaa","powerSource":"Mains (single phase)","modelId":"TS0504B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0504B","manufacturerName":"_TZ3000_ukuvyhaa","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613555678051}`

Koenkk, 2021-02-17:
With the modelID now; it should be correctly recognised in the latest dev branch.

netweaver1970, 2021-02-19:
Thanks, it worked

rmickael62, 2021-02-22:
I just bought the same model of controller "TS0504B" but I do not understand how to control the white channel? 
[image]