Koenkk/zigbee2mqtt #6302: CC2652RB Error: Coordinator failed to start, probably the panID is already in use after restart

robertodocampo, 2021-02-15:
Previus to migrate from my old CC2531 to a new CC2652 i create a laboratory test with only 1 coordinator (CC2652) and 1 device (Blitwoiif BW-SHP13). 

When i start zigbee2mqtt no problems, i pair the device without problems and the system seems to be ok. My problems occurs after restart zigbee2mqtt. After the restart the zigbee2mqtt fails with "Error: Coordinator failed to start, probably the panID is already in use"

I try to delete the config, change several panId's network keys.. and the problems is the same. panId in use after reboot.
I simulate the same scenary with the CC2531 and no problems after Reboot.

I'm ussing zigbee2mqtt@1.17.1  and the latest firmware.

Koenkk, 2021-02-15:
Are you flashing the sick with windows?

robertodocampo, 2021-02-15:
No, i'm using mac os x.

I reproduce the logs and are available on this url's:

Initial start (and pairing):
https://pastebin.com/bpQEsU6E

After restart the zigbee2mqtt:
https://pastebin.com/VS0jhj7n

Regards!

Koenkk, 2021-02-15:
Can you try flashing the stick on a different computer? I've spend lots of time investigating the exactly the same issue in https://github.com/Koenkk/zigbee-herdsman/issues/282 . Turned out somehow his setup flashed the firmware incorrectly.

robertodocampo, 2021-02-15:
Great!! I try to reflash the CC2652 on a W10 (instead of OS X) and the CC2652 survive to the restart!

Thanks a lot @Koenkk !

Koenkk, 2021-02-15:
What python version is installed on windows/osx?

robertodocampo, 2021-02-15:
Sorry i deleted the OS X installation but it's possible that i used an old version 3.6 (or similar). On windows i use 3.9.1

regards!

closed 2021-02-16 by robertodocampo

benek984, 2021-05-09:
@Koenkk I have the same problem, I have tried everything. Re-flashing on MAC and Windows, tried to wrap stick in the aluminium foil (film) still the same. When I start with new PANID I am able to start z2m but soon as I add one device and restart the z2m I am not able to run it. I have CC2652RB from SLAE.SH 
Before I used Texas Instruments CC2531 and all was perfect until my network grow above 30 devices so I decided to go for CC2652RB from SLAE.SH but since then nothing but trouble. Running z2m as add-on in Home assistant. 

Please help

kotjaraxx, 2021-06-01:
Тоже имею такую проблему. Перепрошивка на разных пк, замена стиков(3шт), проблему не решает.

heapxor, 2022-05-26:
What is the solution for that?

ovizii, 2022-06-13:
Another user with CC2652RB from SLAE.SH here. Flashed like this: https://slae.sh/projects/cc2652/#flashing 
I tried flashing from Win 10, later on via Debian. Flashing was successful, I can see the correct firmware on the stick but zigbee2mqtt kees failing with:

```
Zigbee2MQTT:error 2022-06-13 19:19:36: Exiting...
Zigbee2MQTT:error 2022-06-13 19:19:37: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
```
I definitely do not have the same panID. I am trying to set up this new stick on a raspi for a holiday home. My main HA with Z2M is running on my NAS and has a different panID.

Anyway, I'll try and debug this with the instructions provided here https://slae.sh/projects/cc2652/#flashing I just thought I'd post here too, maybe someone has meanwhile figured out what the issue was.

jerrychong25, 2022-09-19:
> Another user with CC2652RB from SLAE.SH here. Flashed like this: https://slae.sh/projects/cc2652/#flashing I tried flashing from Win 10, later on via Debian. Flashing was successful, I can see the [...]

Hi @ovizii ,

I had same issue.

Do you found a solution for this issue?

If yes, mind to share full steps?

Thank you!

ovizii, 2022-09-19:
> Hi @ovizii ,
> 
> I had same issue.
> 
> Do you found a solution for this issue?
> 
> If yes, mind to share full steps?
> 
> Thank you!

I got it working, but I can't remember what the fix was. I had 3 sticks around, I eventually ended up flashing all 3 of them from a Windows machine, if I remember correctly I used this tool https://www.ti.com/tool/UNIFLASH I found referenced here: https://slae.sh/projects/cc2652/#flashing - it was a pain in the a** to get all of that working. Good luck.

csirek, 2022-09-22:
I have a different modul but the sympthons are the same.

I was able to resolve this problem here: https://github.com/Koenkk/zigbee2mqtt/issues/10209#issuecomment-1251479014)

ustari28, 2025-04-22:
I upgraded recently y I had the same problem, I restarted one time more and the coordinator came back online.