Koenkk/zigbee2mqtt #6304: Hue motion occupy/clear message doesn't go through (very random and very occasionally)

ASNNetworks, 2021-02-15:
## What happened
So very occasionally, like maybe once every week, sometimes longer than that, I notice the Hue motion sensor doesn't trigger.  I noticed that when I then stop moving, wait 10 seconds and then move again, it does detect. Also I had that the 'clear' command didn't trigger after I left the area. Going back in the area and then back out again for 10 seconds, did send the 'clear' command. So my thoughts are: the Hue motion sensor does trigger it internally, but somehow the message doesn't go through. Which explains why it does trigger after waiting for 10 seconds. And why it sends the clear command after I re-enter the room and wait 10 seconds after it didn't send the 'clear' message before. 

## What did you expect to happen
I expect it to trigger normally and all messages to go through. 

## How to reproduce it (minimal and precise)
This is very hard to reproduce, since it happens completely at random. What I do know though, is that me and my brother (he also has HA with same setup) both have this issue. We both use zigbee2mqtt with a Conbee II (on a usb 2,0 extension cable). I try to look for the logs, but I don't know what to look for. I am aware that Conbee II is still experimental. But I don't know it this is a Conbee II issue, or something else. I don't think this is related to Philips Hue Motion per se, because I experienced this behaviour in the past with Sonoff SNZB-03 as well (but perhaps both have this issue coincidentally? 

## Debug info
Zigbee2MQTT version:  1.17.1 and edge release
Adapter hardware: Conbee II
Adapter firmware version: latest

Extra: I have 8 Philips Hue Motion sensors and 2 of them have showed this behaviour. But it could be a coincidence, since those two are used the most (so higher chance to re-produce because they get used more).

jimmyeao, 2021-02-16:
When did you last replace the battery? I had issues with a couple doing weird things like this until I replaced the battery, then it cleared up

ASNNetworks, 2021-02-16:
> When did you last replace the battery? I had issues with a couple doing weird things like this until I replaced the battery, then it cleared up

They are brand new, as in just a week old lol. The sensors are all brand new. Also my brothers are brand new, all little over 10 days old as of today. It has nothing to do with the battery. As I explained, I believe Hue motion sensor somehow does trigger internally. Because when I keep moving, it will remain off (because no new message will be sent). But if I stop, wait 10s and then move, it will turn on. Meaning internally it went off and sent a new 'occupied' message. 

This is also the case when it was on and didn't turn off when there was no motion. Entering the area and standing still for 10 seconds, sent the 'clear' message which then did turn it off.

jimmyeao, 2021-02-16:
I'll follow this with interest. The delay between the sensor 'clearing' and being able to re-detect motion is something that annoys me greatly in Hue (I get the same 10s delay you mention). Im testing some Aqara sensors at the moment which initially appear a little more responsive.

ASNNetworks, 2021-02-16:
> I'll follow this with interest. The delay between the sensor 'clearing' and being able to re-detect motion is something that annoys me greatly in Hue (I get the same 10s delay you mention). Im testing some Aqara sensors at the moment which initially appear a little more responsive.

I think you are not understanding my problem correctly. The 10 second delay is NOT between clearing and detecting again. That is just the 10 second cooldown. When the sensor sees no motion for 10 seconds, it will send a clear signal. When I move right then, it will immediately send a occupy signal again. If you keep moving in that first 10 seconds, it will keep extending the occupy, untill no motion. That part works fantastic and way better than Aqara (that’s the one of the reasons I ditched 14 Aqara en Sonoff sensors).

My only issue is that on very rare occasions, somehow the occupied OR clear message doesn’t go through zigbee2mqtt. Though it appears the sensor itself does trigger internally (hence standing still for 10s re-triggers).

Please keep my topic about this only issue. If you have other issues, like the 10 second delay between clear message and a new trigger, create a new topic about that. Because I do not have that issue with all my Hue motion sensors.

ASNNetworks, 2021-03-10:
Nobody else having issues with messages not going through? I'm still having this issue unfortunately. Not sure if it's actually Z2M, or maybe MQTT. But the weird thing is: only motion sensors have this issue, all other devices work fine 100% of the time.

hirsts, 2021-03-31:
I am having similar issues. I recently migrated from ZHA to Z2M. I have been using Philips Hue motion sensors for a long time and they have been very reliable and accurate. Initial testing with Z2M worked and so I migrated. Since then I am getting random "occupied" states remaining rather then clearing, Ive not yet done detailed analysis but came across this issue.

Zigbee2MQTT version
1.18.1 commit: 1.18.1
Coordinator type
ConBee2/RaspBee2
Coordinator revision
0x26680700
Frontend version
0.3.76

The sensor info below is for a sensor that is currently incorrectly reporting the room as occupied.

[image]
[image]

ASNNetworks, 2021-03-31:
@hirsts I have also found these tickets in the meantime: 

https://github.com/Koenkk/zigbee2mqtt/issues/6684

and 

https://github.com/Koenkk/zigbee2mqtt/issues/4694

Could be related?

ASNNetworks, 2021-05-01:
Commenting for visibility,  still present. Not sure it's related to the tickets I mentioned yet.

ASNNetworks, 2021-06-01:
Still present.

ASNNetworks, 2021-07-03:
Still present

ASNNetworks, 2021-08-03:
Still present.

ASNNetworks, 2021-09-03:
Still present

ASNNetworks, 2021-10-05:
Still present

closed 2021-11-12 by github-actions[bot]