Koenkk/zigbee2mqtt #6310: IKEA LED Driver and Bulb spontaneously not working until power cycle

CBismark, 2021-02-15:
## What happened
Sometimes my IKEA LED Driver 30W (ICPSHC24-10EU-IL-1) and my IKEA Bulb (LED1732G11) are not reacting anymore. A hardreset (power cycle) and they work again. This seems to have begun since I changed my coordinator to a Conbee 2.
The error log from zigbee2mqtt says
`error 2021-02-15 21:07:12: Publish 'set' 'state' to 'Licht_Kommode' failed: 'Error: Command 0x842e14fffe403e3b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'`

## What did you expect to happen
They should work all the time.

## How to reproduce it (minimal and precise)
They spontaneously don't work anymore, hard to reproduce.

## Debug info
Zigbee2MQTT version: 1.17.0-dev
Adapter hardware: Conbee II
Adapter firmware version: coordinator revision 0x26680700

Did anyone have a simliar problem to this? Is there anything I can do to help you identify the problem further?

Kind Regards,

Christoph

MattWestb, 2021-02-16:
Have you updating the firmware o the driver ??
The EU version have getting on large update to Zigbee 3 for one week ago.

CBismark, 2021-02-16:
I updated the firmware of the ICPSHC24-10EU-IL-1 and will see if the problem occurs again or not.

CBismark, 2021-02-27:
The problem has happened again. The LED driver and the bulb both don't work anymore, until I hard reset them. Any ideas what do do about it?

closed 2021-04-06 by github-actions[bot]