Koenkk/zigbee2mqtt #6313: Tuya TS0041 'wall switch' injecting error(s) in z2m logs regularly

netweaver1970, 2021-02-15:
Hi,
using latest 1.17.1-dev version (also happens with older versions), I'm getting regular error in the logs.
The switch works fine most of the time. Sometimes I need to click a 2nd time, as if the 1st action didn't register.
Only bound to the coordinator, cluster PowerCfg.
No Reporting. 
The model photo shown for the TS0041 is slightly different than my real switch, I'm having big full-face, hinged (at the top) mechanical switch while the picture shows a touch type small circle as contact area. Might be the same inside.
Battery shows 100% in the GUI but only got installed 2 weeks ago. So it could be right.

Stilll ...

`Error Failed to configure 'Office / door', attempt 2 (Error: ConfigureReporting 0x5c0272fffecaa025/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32700 - 1 - 21 - 1 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:554:17) at processTimers (internal/timers.js:497:7))`

Why trying to configure reporting? This is an Action Type, no use in trying to set up reporting I guess... Only for the battery level maybe. But maybe this one only supports polling? Or transfers the battery percentage only as part of the Action message?

Thanks.

Koenkk, 2021-02-16:
Can you provide the data/database.db for this device?

netweaver1970, 2021-02-17:
with pleasure. This is the device I bought: https://www.aliexpress.com/item/4001095272467.html?spm=a2g0s.9042311.0.0.27424c4dUKSqZc

database.db entry 
`{"id":64,"type":"EndDevice","ieeeAddr":"0x5c0272fffecaa025","nwkAddr":32700,"manufId":4098,"manufName":"_TZ3000_peszejy7","powerSource":"Battery","modelId":"TS0041","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,10,1,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\f\u0000\u0000\u00                                      00\u0000\u0011","65506":28,"modelId":"TS0041","manufacturerName":"_TZ3000_peszejy7","powerSource":3,"appVersion":66,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":"attributes":"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca15655","endpointID":1}],"configuredReportings":],"meta":}}},"appVersion":255,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613547186913}
`

Koenkk, 2021-02-17:
Issue should be fixed now. Looks that the configure reporting for battery percentage remaining isn't always needed since you got a battery level anyway.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

hrosak88, 2021-02-21:
Hi! I have the same problem with TS0041. After a few minutes must I click second time to do an action. It is possible to repair in next stable version? Thx.

closed 2021-04-01 by github-actions[bot]