Koenkk/zigbee2mqtt #6314: problems connecting to a Mosquitto V2 in early moments of the Z2M container boot.

netweaver1970, 2021-02-15:
version 1.17.1-Dev, errors when booting up the docker container :
Upgraded to Mosquitto V2 recently.
Sometimes the Z2M container stops with error 137.

But below error is repeatable, when restarting the containter.

```
Zigbee2MQTT:error 2021-02-15 22:32:37: Cannot send message: topic: 'zigbee2mqtt/Aqara PIR Office', payload: '{"battery":100,"elapsed":4,"illuminance":26,"illuminance_lux":26,"last_seen":"2021-02-15T22:32:36+00:00","linkquality":144,"occupancy":true,"temperature":24,"voltage":3015}
Zigbee2MQTT:error 2021-02-15 22:32:37: Not connected to MQTT server!
Zigbee2MQTT:error 2021-02-15 22:32:37: Cannot send message: topic: 'zigbee2mqtt/Aqara PIR Office', payload: '{"battery":100,"elapsed":6,"illuminance":30,"illuminance_lux":30,"last_seen":"2021-02-15T22:32:36+00:00","linkquality":144,"occupancy":true,"temperature":24,"voltage":3015}
Zigbee2MQTT:error 2021-02-15 22:32:37: Not connected to MQTT server!
Zigbee2MQTT:error 2021-02-15 22:32:37: Cannot send message: topic: 'zigbee2mqtt/Aqara PIR Office', payload: '{"battery":100,"elapsed":4,"illuminance":30,"illuminance_lux":30,"last_seen":"2021-02-15T22:32:36+00:00","linkquality":144,"occupancy":true,"temperature":24,"voltage":3015}
Zigbee2MQTT:error 2021-02-15 22:32:37: Not connected to MQTT server!
Zigbee2MQTT:error 2021-02-15 22:32:37: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"announce","meta":{"friendly_name":"0x60a423fffe113527"},"type":"device_announced"}
Zigbee2MQTT:error 2021-02-15 22:32:38: Not connected to MQTT server!
Zigbee2MQTT:error 2021-02-15 22:32:38: Cannot send message: topic: 'zigbee2mqtt/TH_Fridge', payload: '{"battery":62,"elapsed":0,"humidity":66.67,"last_seen":"2021-02-15T22:32:36+00:00","linkquality":66,"pressure":1023,"temperature":7.85,"voltage":2935}
Zigbee2MQTT:info  2021-02-15 22:32:39: Connected to MQTT server
Zigbee2MQTT:info  2021-02-15 22:32:39: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

```

after a while all 'falls through' and connects fine to the Mosquitto.

Any idea?

Thanks.

Koenkk, 2021-02-16:
Regarding startup, could it be that z2m starts faster than mosquitto?

IngoHenning, 2021-02-16:
Starting with the release of Mosquitto version 2.0.0 the default config will only bind to localhost and will not accept external connections. More information can be found in the release notes https://mosquitto.org/blog/2020/12/version-2-0-0-released/

I had the same issue and downgraded to Mosquitto 1.63 for a quick fix.

Pofilo, 2021-02-16:
@IngoHenning I suppose you talk about Mosquitto 1.6.13 !
But yeah, it works with a downgrade, thanks for the tip !

IngoHenning, 2021-02-16:
You are right, I mean 1.6.13 sorry

IngoHenning, 2021-02-16:
@Pofilo You are right, I mean 1.6.13 sorry

netweaver1970, 2021-02-17:
I've got Mosquitto v2 running with this conf file (from Andreas Spiess IOTstack fork) : https://github.com/SensorsIot/IOTstack/blob/master/.templates/mosquitto/mosquitto.conf
and this docker-compose service file snippet :  https://github.com/SensorsIot/IOTstack/blob/master/.templates/mosquitto/service.yml

It kind of works fine now, only the booting up of z2m acts up and sometimes in the middle of a session it also stops connecting. And it recovers again later.

very weird.

Pofilo, 2021-03-21:
I think this issue can be closed.
The problem was because the 2.0 version of Mosquitto binds to localhost.
A solution is to add `listener 1883` to the conf as explained in the documentation.

closed 2021-03-21 by Koenkk