Koenkk/zigbee2mqtt #6321: Please add support for Tuya Smart water/gas valve _TZ3000_o4cjetlm

oetelaar24, 2021-02-16:
Hi Koen,

I've just bought a ZN231392 Smart water/gas valve (https://nl.aliexpress.com/item/4001051095315.html)
This seems exactly the Tuya valve: https://www.zigbee2mqtt.io/devices/TS011F_valve.html however, upon pairing, it is identified as a TS0001

`Zigbee2MQTT:info  2021-02-16 15:17:02: Device '0x60a423fffed222c0' joined
,Zigbee2MQTT:info  2021-02-16 15:17:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffed222c0","ieee_address":"0x60a423fffed222c0"},"type":"device_joined"}'
,Zigbee2MQTT:info  2021-02-16 15:17:02: Starting interview of '0x60a423fffed222c0'
,Zigbee2MQTT:info  2021-02-16 15:17:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffed222c0","ieee_address":"0x60a423fffed222c0","status":"started"},"type":"device_interview"}'
,Zigbee2MQTT:info  2021-02-16 15:17:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x60a423fffed222c0"},"type":"device_connected"}'
,Zigbee2MQTT:info  2021-02-16 15:17:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x60a423fffed222c0"},"type":"pairing"}'
,Zigbee2MQTT:info  2021-02-16 15:17:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffed222c0","ieee_address":"0x60a423fffed222c0"},"type":"device_announce"}'
,Zigbee2MQTT:info  2021-02-16 15:17:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x60a423fffed222c0"},"type":"device_announced"}'
,Zigbee2MQTT:info  2021-02-16 15:17:06: Successfully interviewed '0x60a423fffed222c0', device has successfully been paired
,Zigbee2MQTT:info  2021-02-16 15:17:06: Device '0x60a423fffed222c0' is supported, identified as: TuYa 1 gang switch (TS0001)`

After trying to remove and re-pair the device a couple of times, I got the unsupported device message:
`,Zigbee2MQTT:warn  2021-02-16 15:27:10: Device '0x60a423fffed222c0' with Zigbee model 'undefined' and manufacturer name '_TZ3000_o4cjetlm' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

Looking in the dev version of devices.js, i couldn't find support for this model just yet, but I'm quite confident that adding a copy of:
`    {
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_rk2yzt0u'}],
        model: 'TS011F_valve',
        vendor: 'Lonsonho',
        description: 'Smart water/gas valve (upgrade version)',
        extend: preset.switch(),
    },`

I assume (I know I shouldn't) that this would work?:
`    {
        fingerprint: [{modelID: 'ZN231392', manufacturerName: '_TZ3000_o4cjetlm'}],
        model: 'ZN231392',
        vendor: 'Tuya',
        description: 'Smart water/gas valve (Tuya)',
        extend: preset.switch(),
    },`

(I've gotten the modelID from the sticker on the box).

I hope I've provided enough information. Please let me know if I missed something/cut to many corners here :thinking: 

Kind regards,
-Bart

Koenkk, 2021-02-16:
Should work now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

oetelaar24, 2021-02-16:
Thanks!
It works now, but there is also something funny going on.
When I enable Joining and then push the button on the device for 5secs (as per manual, the led starts blinking..). it joins and is recognized as a TS001, it does seem to work just fine.
When I then remove the device via de webinterface and let it rejoin, it still gets joined as the TS001.
If I press the button on the device for 5 secs again, it leaves the network, rejoins, and then is correctly detected as the valve.

It seems to me the device is not identifying itself correctly the first times?
I can repeat this and get a consitent behaviour out of it. It only joins/get recognised correctly as the valve after leaving/rejoining via pressing of the button on the device itself.

Koenkk, 2021-02-17:
Can you provide the database.db entry of this device when it's recognised as the `TS001`?

oetelaar24, 2021-02-17:
Ofcourse:

> {"id":45,"type":"Router","ieeeAddr":"0x60a423fffed222c0","nwkAddr":50698,"manufId":4098,"manufName":"_TZ3000_o4cjetlm","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TZ3000_o4cjetlm","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00215ff75d","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1613586617862}

Koenkk, 2021-02-17:
Should be fixed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

oetelaar24, 2021-02-18:
I've updated to the latest-dev image. However I see no change in behaviour.
When it the container starts, it logs:
Zigbee2MQTT:info  2021-02-18 19:26:42: Starting Zigbee2MQTT version 1.17.1-dev (commit #67bb215)
,Zigbee2MQTT:info  2021-02-18 19:26:42: Starting zigbee-herdsman (0.13.71)

Koenkk, 2021-02-18:
I forgot to update indeed, please try again in a few hours from now.

oetelaar24, 2021-02-18:
Awesome, works well every time now :) Thanks!!
Only thing missing is the image of the device, but I guess that'll get added on the new release, because the documentation is also not added?

Koenkk, 2021-02-19:
Picture will be available with next release (1 March)

closed 2021-02-19 by Koenkk

sator86, 2022-05-18:
I have the same issue
this valve ZN231392   is detected as switch TS0011
any help please :)
thank you

sator86, 2022-05-18:
how can I get it ?
I don't have any idea

Koenkk, 2022-05-20:
The file is next to the zigbee2mqtt `configuration.yaml`.

sator86, 2022-05-20:
hello,
is this correct ?

Koenkk, 2022-05-21:
@sator86 should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

sator86, 2022-05-22:
thank you @Koenkk for your help
I got the dev version, but facing to a problem fatal: can only run as pid 1
so I have to wait until the new release of stable
thank you for your time :)
I realy appreciate

uowis, 2023-04-05:
Hi have similar device not supported with this database entry : 
{"id":32,"type":"Router","ieeeAddr":"0xa4c13809602682ac","nwkAddr":41195,"manufId":4417,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1680703373514,"defaultSendRequestWhen":"immediate"}