Koenkk/zigbee2mqtt #6326: Distribute container (Docker) via the GitHub Container Registry

frenck, 2021-02-17:
## Is your feature request related to a problem? Please describe

DockerHub is imposing API limits, which, as a developer using Docker a lot, run into from time to time.

## Describe the solution you'd like

I would love to be able to get the image for the Zigbee2MQTT container via the GitHub Container Registry

## Describe alternatives you've considered

🤷 

## Additional context

They can be distributed in parallel.

Koenkk, 2021-02-17:
Hi @frenck :smile:

Good suggestion, I've adapted the ci to also push to ghcr.

Link: https://github.com/users/Koenkk/packages/container/package/zigbee2mqtt

Currently only the `latest-dev` is available, with the next release (on 1 March) `latest`/`1.18.0` will also be available.

frenck, 2021-02-17:
👋  @Koenkk

Awesome! Thanks 👍

closed 2021-02-17 by Koenkk