Koenkk/zigbee2mqtt #6327: RIALTO (4-NOKS) PLUG AND THERMOSTAT

moskyow, 2021-02-17:
Hi,

I'm trying to add 2 new devices : one smart plug and one thermostat from Rialto (Astrel Group)
Link at products : https://www.astrelgroup.com/soluzioni/smart-home-iot/prodotti-rialto-active/

The products are similar to Bitron, therefore I have created an external converters with the Bitron definition.

for plug
 {
        zigbeeModel: ['ZR-PLUG-RB-HA'],
        model: 'SMART-PLUG-EU-RI',
        vendor: '4-noks s.r.l.',
        description: 'Video wireless socket',
        fromZigbee: [fz.on_off, fz.metering],
        toZigbee: [tz.on_off],
        exposes: [e.switch(), e.power()],
        meta: {configureKey: 4},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await reporting.instantaneousDemand(endpoint);
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 10000, multiplier: 1});
        },

For thermostat
{
        zigbeeModel: ['ZED-TTR2-HA'],
        model: 'ZED-TTR-RI',
        vendor: '4-noks s.r.l.',
        description: 'Wireless wall thermostat with relay',
        fromZigbee: [fz.legacy.bitron_thermostat_att_report, fz.battery],
        toZigbee: [tz.thermostat_occupied_heating_setpoint, tz.thermostat_local_temperature_calibration, tz.thermostat_local_temperature,
            tz.thermostat_running_state, tz.thermostat_temperature_display_mode, tz.thermostat_system_mode],
        exposes: [e.battery(), exposes.climate().withSetpoint('occupied_heating_setpoint', 7, 30, 0.5).withLocalTemperature()
            .withSystemMode(['off', 'auto', 'heat']).withRunningState(['idle', 'heat', 'cool']).withLocalTemperatureCalibration()],
        meta: {configureKey: 2, battery: {voltageToPercentage: '3V_2500_3200'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const binds = [
                'genBasic', 'genPowerCfg', 'genIdentify', 'genPollCtrl', 'hvacThermostat', 'hvacUserInterfaceCfg',
            ];
            await reporting.bind(endpoint, coordinatorEndpoint, binds);
            await reporting.thermostatTemperature(endpoint, {min: 900, max: repInterval.HOUR, change: 1});
            await reporting.thermostatTemperatureCalibration(endpoint);
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
            await reporting.thermostatRunningState(endpoint);
            await reporting.batteryAlarmState(endpoint);
            await reporting.batteryVoltage(endpoint);
        },

For the plug I have the follow error :
Zigbee2MQTT:error 2021-02-17 10:39:06: Failed to configure '0x000d6f0005bccea3', attempt 1 (ReferenceError: reporting is not defined
    at Object.configure (evalmachine.<anonymous>:18:9)
    at Configure.configure (/app/lib/extension/configure.js:131:45)
    at Configure.onZigbeeStarted (/app/lib/extension/configure.js:99:28)
    at Controller.callExtensionMethod (/app/lib/controller.js:382:44))

Below my database.db (i don't know why i have two coordinator CC2531, but the two xiaomi devices work good)

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e1a599","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}

{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d00025eee31","nwkAddr":31523,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,65535,1026,1027,1029],"outClusterList":[0,4,65535],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2011}},"msRelativeHumidity":{"attributes":{"measuredValue":5285}},"msPressureMeasurement":{"attributes":{"measuredValue":984,"scale":-1,"scaledValue":9844}},"genBasic":{"attributes":{"65281":{"1":2945,"4":5032,"5":365,"6":[0,1],"10":0,"100":2014,"101":5270,"102":98431},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":30,"modelId":"lumi.weather","dateCode":"20161129","swBuildId":"3000-0001"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":2,"hwVersion":30,"dateCode":"20161129","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1613554205802}

{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d000279c6a8","nwkAddr":2506,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_smoke","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,18,1280,12,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3065,"3":29,"4":17320,"5":4108,"6":[0,1],"8":4100,"10":0,"100":7,"150":0},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":16,"dateCode":"20170314","swBuildId":"3000-0001","modelId":"lumi.sensor_smoke"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0018e1a599","zoneState":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":16,"dateCode":"20170314","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1613550134335}

{"id":4,"type":"EndDevice","ieeeAddr":"0x000d6f000cda7330","nwkAddr":39802,"manufId":4209,"manufName":"4-noks s.r.l.","powerSource":"Battery","modelId":"ZED-TTR2-HA","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,10,32,513,516,2821],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZED-TTR2-HA","manufacturerName":"4-noks s.r.l.","powerSource":3,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"2017032717086110","swBuildId":"V1b232-20160120"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"2017032717086110","swBuildId":"V1b232-20160120","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1613513815959}

{"id":5,"type":"Router","ieeeAddr":"0x000d6f0005bccea3","nwkAddr":2926,"manufId":4209,"manufName":"4-noks s.r.l.","powerSource":"Mains (single phase)","modelId":"ZR-PLUG-RB-HA","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,6,1794],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"ZR-PLUG-RB-HA","manufacturerName":"4-noks s.r.l.","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"180591088","swBuildId":"V1b242-20160419"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0012 [... cut ...] tings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}

closed 2021-03-28 by github-actions[bot]

mircob85, 2022-09-15:
@moskyow Hai qualche novità con l'integrazione di questi termostati Rialto 4-noks? Avrei bisogno anch'io. grazie

moskyow, 2022-09-15:
> @moskyow Do you have any news with the integration of these Rialto 4-noks thermostats? I need it too. thank you

I've integrated with ZHA and Sonoff ZIGBEE coordinator

mircob85, 2022-09-15:
Have you got telegram? Can you help me? Thank you

mircob85, 2022-10-15:
> > @moskyow Do you have any news with the integration of these Rialto 4-noks thermostats? I need it too. thank you
> 
> I've integrated with ZHA and Sonoff ZIGBEE coordinator

Can you help me? have you got telegram? thank you

masp31, 2023-04-26:
Hi,  Did you have the integration of the  4-noks plug?  thank you

moskyow, 2023-04-26:
> Hi, Did you have the integration of the 4-noks plug? thank you

Use ZHA and do the pairing as writtern on the pairing instructions of the plug. Very simple

masp31, 2023-04-26:
OK. Thank you. But, can i have ZHA in parallel with Z2M in the same dongle?

moskyow, 2023-04-26:
> OK. Thank you. But, can i have ZHA in parallel with Z2M in the same dongle?

I think no

masp31, 2023-05-02:
I still dont get it... Can you tell me the n. of the chanel, do you have.
Thank you

FragMenthor, 2023-05-03:
> > Hi, Did you have the integration of the 4-noks plug? thank you
> 
> Use ZHA and do the pairing as writtern on the pairing instructions of the plug. Very simple

Hello! Can you please tell us what is tour channel number in ZHA? Probably this device only works in some channels, because I am also testing one and only reports "last_seen", doen't pair; the channel must be the limiter...
Thanks in advance!