Koenkk/zigbee2mqtt #6330: Lidl HG06668 reports multiple actions after pressed once.

Danielson187, 2021-02-17:
## What happened
When i press the Lidl HG06668 Zigbee doorbell once it reports multi "pressed" actions, most of the time around 3/4 times. I tought the device was bad so i bought a new one, the new one acts te same;

`info  2021-02-16 10:56:40: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"pressed","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
info  2021-02-16 10:56:40: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
info  2021-02-16 10:56:40: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell/action', payload 'pressed'
info  2021-02-16 10:56:42: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"pressed","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
info  2021-02-16 10:56:42: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
info  2021-02-16 10:56:42: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell/action', payload 'pressed'
info  2021-02-16 10:56:45: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"pressed","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
info  2021-02-16 10:56:45: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
info  2021-02-16 10:56:45: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell/action', payload 'pressed'`

## What did you expect to happen
Just a single "pressed" action.

## How to reproduce it (minimal and precise)
Just press the doorbell once.

## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2021-02-17:
Did an attempt to fix this, please let me know if it works.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Danielson187, 2021-02-18:
Hi Koen,

I've updated to: 1.17.1-dev (commit #a151d76)

But the issue is still there!

Koenkk, 2021-02-18:
Can you provide the herdsman debug logging of the duplicate events?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Danielson187, 2021-02-18:
Like this?

`[34mZigbee2MQTT:debug[39m 2021-02-18 20:38:36: Received Zigbee message from 'lidl_silvercrest_smart_wireless_door_bell', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:36: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"pressed","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:36: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:36: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell/action', payload 'pressed'
[34mZigbee2MQTT:debug[39m 2021-02-18 20:38:38: Received Zigbee message from 'lidl_silvercrest_smart_wireless_door_bell', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:38: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"pressed","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:38: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:38: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell/action', payload 'pressed'
[34mZigbee2MQTT:debug[39m 2021-02-18 20:38:39: Received Zigbee message from 'xiaomi_door_window_contact_sensor_2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:39: MQTT publish: topic 'zigbee2mqtt/xiaomi_door_window_contact_sensor_2', payload '{"battery":100,"contact":true,"linkquality":26,"temperature":23,"voltage":3005}'
[34mZigbee2MQTT:debug[39m 2021-02-18 20:38:40: Received Zigbee message from 'lidl_silvercrest_smart_wireless_door_bell', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:40: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"battery":100,"battery_low":false,"linkquality":28,"tamper":false}'
[34mZigbee2MQTT:debug[39m 2021-02-18 20:38:41: Received Zigbee message from 'lidl_silvercrest_smart_wireless_door_bell', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:41: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"pressed","battery":100,"battery_low":false,"linkquality":31,"tamper":false}'
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:41: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"","battery":100,"battery_low":false,"linkquality":31,"tamper":false}'
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:41: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell/action', payload 'pressed'
[34mZigbee2MQTT:debug[39m 2021-02-18 20:38:44: Received Zigbee message from 'lidl_silvercrest_smart_wireless_door_bell', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:44: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"pressed","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:44: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"","battery":100,"battery_low":false,"linkquality":26,"tamper":false}'
[32mZigbee2MQTT:info [39m 2021-02-18 20:38:44: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell/action', payload 'pressed'`

Danielson187, 2021-02-18:
Or this:

`  zigbee-herdsman:controller:log Disable joining +9s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":2,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,3,2,11,0,0]}} +9s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,2,0,30,6,0,25,3,2,11,0,0,20] +9s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +9s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +9s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,2,55] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,2,55] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,2] - 55 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":2} +9s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":0,"tcsignificance":0} +215ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,0,0,26] +216ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82] +211ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +218ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:log Port closed +11s
Zigbee2MQTT:info  2021-02-18 20:37:46: zigbee-herdsman stopped
Zigbee2MQTT:info  2021-02-18 20:37:46: Stopped Zigbee2MQTT

danielson@SRV01:/opt/zigbee2mqtt$ DEBUG=zigbee-herdsman* npm start > /home/danielson/z2m.log
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +7ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +459ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,1,26] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,1,26] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,1] - 26 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":377} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +18ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,0,2,6,3,144,21,52,1,2,0,0,0,0] - 218 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608} +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2 +0ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +10ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +11ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,0,0,26] +217ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82] +211ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +219ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:log Port closed +2m
`

Koenkk, 2021-02-19:
Thanks, but this log contains a lot of events. Can you provide the log of just a single click leading into multiple publishes?

Danielson187, 2021-02-19:
Here you go:

`Zigbee2MQTT:info  2021-02-19 19:32:21: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"pressed","battery":100,"battery_low":false,"linkquality":26,"tamper":true}'
Zigbee2MQTT:info  2021-02-19 19:32:21: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell', payload '{"action":"","battery":100,"battery_low":false,"linkquality":26,"tamper":true}'
Zigbee2MQTT:info  2021-02-19 19:32:21: MQTT publish: topic 'zigbee2mqtt/lidl_silvercrest_smart_wireless_door_bell/action', payload 'pressed'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +9s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,233,1,11,36] +56ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,233,1,11,36] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [233,1,11] - 36 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":233,"endpoint":1,"transid":11} +76ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x842e14fffef50809:59801,233,0) +72ms
  zigbee-herdsman:adapter:zStack:adapter Wait 2000ms +0ms
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x842e14fffef50809:59801/1 (0,1,1) +2s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":59801,"destendpoint":1,"srcendpoint":1,"clusterid":1280,"transid":12,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,94,11,0,0]}} +2s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,153,233,1,1,0,5,12,0,30,5,24,94,11,0,0,5] +2s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,153,233,1,205,50,10] +36ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,153,233,1,205,50,10] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [153,233,1,205,50] - 10 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":59801,"relaycount":1,"relaylist":[13005]} +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,233,1,12,35] +19ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,233,1,12,35] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [233,1,12] - 35 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":233,"endpoint":1,"transid":12} +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x842e14fffef50809:59801,233,1) +74ms
  zigbee-herdsman:adapter:zStack:adapter Discovering route to 59801 +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":59801,"options":0,"radius":30} +74ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,153,233,0,30,10] +74ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33] +27ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0} +84ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,29,68,129,0,0,0,5,153,233,1,1,0,28,0,229,108,60,0,0,9,9,95,0,5,0,0,23,0,0,205,50,10,188] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,153,233,1,1,0,28,0,229,108,60,0,0,9,9,95,0,5,0,0,23,0,0,205,50,10,188] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 29 - 2 - 4 - 129 - [0,0,0,5,153,233,1,1,0,28,0,229,108,60,0,0,9,9,95,0,5,0,0,23,0,0,205,50,10] - 188 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":59801,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":28,"securityuse":0,"timestamp":3960037,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[9,95,0,5,0,0,23,0,0]}} +39ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":95,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"zonestatus":5,"extendedstatus":0},"Command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}},"address":59801,"endpoint":1,"linkquality":28,"groupID":0}' +2s
[... cut ...]
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":59801,"options":0,"radius":30} +65ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,153,233,0,30,10] +66ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33] +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0} +63ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
`

Koenkk, 2021-02-19:
Thanks, I currently don't see how we can filter out the duplicates (since the `transactionSequenceNumber` changes every time). 

@dasfuu  do you also have this issue? (since you added support for it in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2121)

dasfuu, 2021-02-19:
@Koenkk Updated to latest dev branch and I don't have this issue. I only get one event per button press.

dasfuu, 2021-02-19:
For reference here is a photo of my device: 
[image]

I noticed you have very low linkquality. Could it be because of this? It's the only difference I can see.  I am using Slaesh's CC2652RB stick.

Danielson187, 2021-02-20:
I'm having exactly the same version:

[image]

The link quality outside is a bit poor indeed, but inside is great and then it's happenig to! I'm using a CC2531 as coordinator and one CC2531 as a router that is close to the doorbell!

dasfuu, 2021-02-22:
Whats your firmware build date on the doorbell? 
[image]

I'm very curious whats causing this. Can you test the doorbell in a separate network without any other devices? 
(is that even possible that another device causes this?)

Danielson187, 2021-02-23:
What is that interface that you are using? I'm using Zigbee2MqttAssistant but that isn't showing the Firmware version:

[image]

I don't have a separate network to test the doorbell :(

dasfuu, 2021-02-23:
That is zigbee2mqtts own UI. You can enable it in the config. I'm on the latest dev version.

```
frontend:
  port: 8080
```

Danielson187, 2021-02-23:
Nice!

But the firmware seems te be the same:

[image]

dasfuu, 2021-02-23:
I don't know enough about zigbee. I could offer you, to send/lend me one of your buttons and I test it in my setup. But I don't think it makes a difference.

@Koenkk do you accept sponsored zigbee items to integrate/debug them?

Koenkk, 2021-02-23:
@dasfuu I don't accept them, because this typically takes a lot of time and it's not 100% sure that I can reproduce this issue.

Instead of that you can sniff the traffic of this, in this way we can debug further: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

vertex-github, 2021-03-30:
Perhaps related - Im seeing 2 messages for a Xiaomi button single click, despite having ` homeassistant_legacy_triggers: false` in my `advanced` section

closed 2021-05-09 by github-actions[bot]