Koenkk/zigbee2mqtt #6331: Tuya TS0042 - no battery

Mr-Groch, 2021-02-17:
## What happened
[Tuya TS0042](https://www.zigbee2mqtt.io/devices/TS0042.html#tuya-ts0042) device does not send battery values (no voltage and no percentage, but battery is exposed). As can I see in device clusters it should support battery information:

[image]

I think this is related to #6180 - same device family, but with 2 buttons...

## What did you expect to happen
Battery level should be reported

## How to reproduce it (minimal and precise)
Just pair it and observe no battery values

## Debug info
Zigbee2MQTT version: 1.17.1-dev commit: 8eb1a98
Adapter hardware: CC26X2R1 zzh!
Adapter firmware version: 20210120

Mr-Groch, 2021-02-17:
After adding this to devices.js, and after z2m restart battery percentage is sended, like in #6180:

```
    {
        zigbeeModel: ['TS0042'],
        model: 'TS0042',
        vendor: 'TuYa',
        description: 'Wireless switch with 2 buttons',
        whiteLabel: [{vendor: 'Smart9', model: 'S9TSZGB'}, {vendor: 'Lonsonho', model: 'TS0042'}],
        exposes: [e.battery(), e.action(['1_single', '1_double', '1_hold', '2_single', '2_double', '2_hold'])],
        fromZigbee: [fz.tuya_on_off_action, fz.battery],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
        },
    },
```

Koenkk, 2021-02-17:
Added your changes. thanks!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-17 by Koenkk

robertalexa, 2021-08-03:
@Mr-Groch hi, sorry i don't mean to revive an old thread, i was wondering what is your battery life on these devices. I have 4 of them and they die pretty much every 3 4 weeks with little to no use. The devices keep announcing themselves once every hour too, which might be why they drain. I would appreciate somw feedback. Thx

Mr-Groch, 2021-08-04:
Yes, I had same problem with draining when reporting was enabled for those devices (by default). When I manually disabled reporting (you must press the switch before clicking on reporting remove button). I've replaced baterry few months ago and no draining now.

robertalexa, 2021-08-04:
Hi, thanks for coming back to me. Really appreciate it :)

I am just wondering if instead of removing reporting, which will lose the battery status altogether, it might be worth altering the reporting period.
Maybe sometime like min 0, max 0, change 1 (maybe even 5).

According to spec this will only report on value change and not on time interval
[image]

For me the device fails to change the reporting values and even completely remove the binding from the Frontend, so I guess i shall run some tests on my testing instance changing the reporting in the code.

PS: my device is the Zemismart one
[image]

robertcsakany, 2022-12-11:
Hi! I have the same reporting problem, is any update about that?

robertalexa, 2022-12-11:
> Hi! I have the same reporting problem, is any update about that?

You will need to be a lot more specific than this.

On the latest version of z2m there are no reporting issues. The device is not meant to do reporting, instead it will expose the value from time to time. If you updated recently you will need yo re pair your device to ensure it configures correctly