Koenkk/zigbee2mqtt #6334: Exposes - add support for leak sensors?

flatsiedatsie, 2021-02-17:
## Is your feature request related to a problem? Please describe

I have a user of the Zigbee2MQTT addon for the WebThings gateway who has a leak sensor and a door sensor. WebThings gateway can show these devices is a special way (with optimal icons). But currently the [exposes API](https://www.zigbee2mqtt.io/information/exposes) doesn't seem to reflect if a device is a leak sensor (or door sensor?).

## Describe the solution you'd like
If leak sensors are a zigbee device category, being able to recognise such devices via the exposes API would be wonderful.

## Describe alternatives you've considered
I'm not sure, since I don't personally own a leak or door sensor device.

sjorge, 2021-02-17:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/0c34d244d97d0213ae56274f4da18a4a65b4ca3c/lib/exposes.js#L507

They are already expose like this, at least mine is in the z2m frontend

flatsiedatsie, 2021-02-17:
! wow

maybe the exposes documentation page could link to that github page as well?

closed 2021-03-28 by github-actions[bot]