Koenkk/zigbee2mqtt #6338: Lidl HG06104A color does not change unless toggle sent twice

globulus2, 2021-02-17:
## What happened
I have strange behaviors for this led strip
1 - color changing
As in this issue https://github.com/Koenkk/zigbee2mqtt/issues/5716, the color change does not work (no effect, strip remains white). But if i send twice the toggle state, the color change. I also tried sending OFF, then ON, but in this case it remains white.
example : 
zigbee2mqtt/Bandeau_bar/set {"state":"OFF"}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color_temp":0,"linkquality":39,"state":"OFF"}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color_temp":0,"linkquality":39,"state":"OFF"}
--> strip OFF
zigbee2mqtt/Bandeau_bar/set {"state":"ON"}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color_temp":0,"linkquality":42,"state":"ON"}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color_temp":0,"linkquality":42,"state":"ON"}
-->strip ON with warmest white
zigbee2mqtt/Bandeau_bar/set {"color":{"r":255,"g":0,"b":0}}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color":{"x":0.7006,"y":0.2993},"color_temp":37,"linkquality":44,"state":"ON"}
-->no color change but color temp change
zigbee2mqtt/Bandeau_bar/set {"state":"TOGGLE"}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color":{"x":0.7006,"y":0.2993},"color_temp":37,"linkquality":42,"state":"OFF"}
--> strip OFF
zigbee2mqtt/Bandeau_bar/set {"state":"TOGGLE"}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color":{"x":0.7006,"y":0.2993},"color_temp":37,"linkquality":44,"state":"ON"}
--> strip ON, color is red, white leds are off
zigbee2mqtt/Bandeau_bar/set {"state":"OFF"}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color":{"x":0.7006,"y":0.2993},"color_temp":37,"linkquality":42,"state":"OFF"}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color":{"x":0.7006,"y":0.2993},"color_temp":37,"linkquality":42,"state":"OFF"}
--> strip OFF
zigbee2mqtt/Bandeau_bar/set {"state":"ON"}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color":{"x":0.7006,"y":0.2993},"color_temp":37,"linkquality":42,"state":"ON"}
zigbee2mqtt/Bandeau_bar {"brightness":17,"color":{"x":0.7006,"y":0.2993},"color_temp":37,"linkquality":42,"state":"ON"}
--> strip ON, no color, warm white

2 - color temperature
For me, the values go from 0 (warm) to 511 (cold). So the range is different from the range mentionned in the info page (150 to 500).
example :
zigbee2mqtt/lidl/set {"color_temp": 0}
zigbee2mqtt/lidl {"brightness":17,"color_temp":0,"linkquality":39,"state":"ON"}

## Debug info
Zigbee2MQTT version:  Zigbee2MQTT version 1.17.1-dev (commit #9dd9f33)
Adapter hardware: CC2531
Adapter firmware version: ?

globulus2, 2021-02-17:
I noticed that if I send an effect like {"effect": "BLINK"}, the led strip changes the color to the previously set, even if state was on and only white ...

globulus2, 2021-02-22:
To be sure, I bought a second one. The new strip works as expected. So I decided to repair the first one. And .. it finally works as expected.
I don't know whaat happened, as I tried to repair the firsrt one almost 30 times.
But at least, everything is ok.

closed 2021-02-22 by globulus2

regenbieger, 2022-04-16:
found your solution and it works. repairing heals all trouble and only repairing does.
thx

evlo, 2022-09-03:
sad bad true, pairing again fixed issue described here https://github.com/Koenkk/zigbee2mqtt/issues/5716 for me too

MrHaroldA, 2023-10-21:
I've tried paring both of my Lidl HG06104A's like 50 times, and tried the 'reconfigure' option almost as much, but my LED strip controller never responds to color changes, except when choosing an effect; just like described above.

Is there anything I can try next?