Koenkk/zigbee2mqtt #6340: Support for Lightway/Mego (German Aldi Süd) bulbs

Velines, 2021-02-18:
## Information about the device + link
Aldi Süd started selling cheap TuYa-based bulbs today for EUR 9.99. They are labeled as Lightway, sold by a company called Megos.
I have the bought an E27 RGB, an E27 filament and an E14 RGB bulb. I got the first two running by essentially just copying the devices.js entries. I have not yet tried the E14 one (will try later or on the weekend).

## data/database.db entry of the device
E27 Filament:
`{"id":15,"type":"Router","ieeeAddr":"0xbc33acfffeed19a0","nwkAddr":53292,"manufId":4098,"manufName":"_TZ3000_g1glzzfk","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502B","manufacturerName":"_TZ3000_g1glzzfk","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1613645773922}`

E27 RGB:
`{"id":14,"type":"Router","ieeeAddr":"0x60a423fffe64efc2","nwkAddr":13512,"manufId":4098,"manufName":"_TZ3000_kohbva1f","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3000_kohbva1f","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1613645287889}`

Working after adding those configs to devices.js:

```
    {                                                                       
        fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3000_kohbva1f'}],               
        model: 'LIGHTWAY E27 RGB',                                                            
        vendor: 'MEGOS',                                                                          
        description: 'MEGOS LIGHTWAY E27 RGB',                              
        extend: preset.light_onoff_brightness_colortemp_colorxy(),                               
        meta: {applyRedFix: true},                                                                                                
    },
```

```
    {                                                                                                         
        fingerprint: [{modelID: 'TS0502B', manufacturerName: '_TZ3000_g1glzzfk'}],                     
        model: 'LIGHTWAY E27 Filament',                                                                       
        vendor: 'MEGOS',                                                                                       
        description: 'MEGOS LIGHTWAY E27 Filament',                                                 
        extend: preset.light_onoff_brightness_colortemp({disableColorTempStartup: true}),        
    },
```

TheOriginal92, 2021-02-18:
How did you get them in to pairing mode?

Velines, 2021-02-18:
Turn them off (longer than 5 seconds), than on-off-on-off-on with approx 2 or 3 seconds in each state. After that the bulb should "breath" while pairing.

maichai, 2021-02-18:
G10 RGB, https://api.eures-gmbh.de/storage/product_pdfs/de/249.description.pdf

{"dateCode":"","description":"-","friendly_name":"0xredacted","hardwareVersion":1,"ieeeAddr":"0xredacted","lastSeen":1613665196483,"manufacturerID":4098,"manufacturerName":"_TZ3000_j0gtlepx","model":"TS0505B","modelID":"TS0505B","networkAddress":redacted,"powerSource":"Mains (single phase)","type":"Router","vendor":"-"}],"type":"devices"}'

E14 RGB, https://api.eures-gmbh.de/storage/product_pdfs/de/248.description.pdf

tbd

Velines, 2021-02-18:
Here's the E14 bulb, also a TS0505B:

`{"id":18,"type":"Router","ieeeAddr":"0xbc33acfffeeba1ba","nwkAddr":8755,"manufId":4098,"manufName":"_TZ3000_iivsrikg","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3000_iivsrikg","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613665585918}`

```
    {                                                                       
        fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3000_iivsrikg'}],               
        model: 'L122AA63H11A6.5W',                                                            
        vendor: 'MEGOS',                                                                          
        description: 'MEGOS LIGHTWAY E14 RGB',                              
        extend: preset.light_onoff_brightness_colortemp_colorxy(),                               
        meta: {applyRedFix: true},                                                                                                
    },
```

Does anyone know if these support or can be made to support a power on state?

TheOriginal92, 2021-02-18:
> G10 RGB, https://api.eures-gmbh.de/storage/product_pdfs/de/249.description.pdf
> 
> {"id":36,"type":"Router","ieeeAddr":"redacted","nwkAddr":redacted,"manufId":4098,"manufName":"_TZ3000_j0gtlepx","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3000_j0gtlepx","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613665196483}

Thank you very much!

What should i add to the "devices.js" file to support the GU10? Particularly i'm missing the "model" Part

Velines, 2021-02-18:
Model for the GU10 is "L122FF51A114.5W"

Besenreiter, 2021-02-18:
Does the E27 has Routerfunctionality?

Besenreiter, 2021-02-18:
What are the MQTT-tipics to change the Values for the Bulbs (openhab)?

BMOD89, 2021-02-18:
I have created a PR here https://github.com/Koenkk/zigbee-herdsman-converters/pull/2261   
It is pretty similar to your configs. I have set vendor to Aldi, because the Lidl bulbs are also labeled as 'Livarno Lux', but the vendor here is Lidl. 
The other difference is the model name. I didn't find this string, just another number which seems to be the same on all bulbs. 
And as I have no E14 bulb I haven't included it...

BMOD89, 2021-02-18:
Regarding the model, I think you are right. 
I have changed my PR and added also the E14 bulb. 
Maybe this manual is also helpful https://api.eures-gmbh.de/storage/megos_anleitung_de/F122SB62H22A4.5W.pdf

TheOriginal92, 2021-02-18:
> Hello,
> what are the MQTT-topics to change the Values for the Bulbs (openhab)?
> Thanks!

mqtt.things:
Thing mqtt:topic:BadSpot_5 "MQTT Thing: BadSpot_5" (mqtt:broker:openhab3) @ "Bad" {
	Channels:
		Type switch : power "BadSpot_5 State" [
			stateTopic="zigbee2mqtt/BadSpot_5/state",
			commandTopic="zigbee2mqtt/BadSpot_5/set/state",
			on="ON",
			off="OFF"
		]
		Type dimmer : brightness "BadSpot_5 brightness" [
			stateTopic="zigbee2mqtt/BadSpot_5/brightness",
			transformationPattern="JS:getZigbeeBrightness.js",
			commandTopic="zigbee2mqtt/BadSpot_5/set",
			transformationPatternOut="JS:setZigbeeBrightness.js"
		]
		Type number : color_temp "BadSpot_5 color_temp" [
			commandTopic="zigbee2mqtt/BadSpot_5/set",
			transformationPatternOut="JS:setZigbeeCT.js"
		]
		Type number : color_temp_startup "BadSpot_5 color_temp_startup" [
			commandTopic="zigbee2mqtt/BadSpot_5/set",
			transformationPatternOut="JS:setZigbeeCTstartup.js"
		]
		Type colorHSB : color "BadSpot_5 color" [
			stateTopic = "zigbee2mqtt/BadSpot_5/state",
			commandTopic = "zigbee2mqtt/BadSpot_5/set",
			transformationPatternOut = "JS:HSBtoRGB.js"
		]	
		Type string : effect "BadSpot_5 Effect" [
			commandTopic="zigbee2mqtt/BadSpot_5/set",
			transformationPatternOut="JS:setZigbeeEffect.js"
		]
		Type number : signal "BadSpot_5 Signal" [
			stateTopic="zigbee2mqtt/BadSpot_5/linkquality"
		]
		Type datetime : lastseen "BadSpot_5 Last Seen" [
			stateTopic="zigbee2mqtt/BadSpot_5/last_seen"
		]
		Type switch : available "BadSpot_5 Available" [
			stateTopic="zigbee2mqtt/BadSpot_5/available",
			on="ON",
			off="OFF"
		]
}

lights.items:
Group gBadSpot_5							"Bad Spot 5"								<light>				(gBad)															//{alexa="Endpoint.Light"}
Switch BadSpot_5_switch						"Bad Spot 5 [%s]"							<switch>			(gBadSpot_5)													{channel="mqtt:topic:BadSpot_5:power"}
Dimmer BadSpot_5_brightness					"Bad Spot 5 Helligkeit [%.0f]"				<light>				(gBadSpot_5,gBLicht)											{channel="mqtt:topic:BadSpot_5:brightness"}
Dimmer BadSpot_5_ct							"Bad Spot 5 Farbtemperatur [%.0f]"			<colortemp>			(gBadSpot_5,gCtBad)												{channel="mqtt:topic:BadSpot_5:color_temp"}
Color BadSpot_5_color						"Bad Spot 5 Farbe"							<colorwheel>		(gBadSpot_5,gBad_ColorLED)										{channel="mqtt:topic:BadSpot_5:color"}
String BadSpot_5_effect						"Bad Spot 5 Effekt [%s]"					<text>				(gBadSpot_5)													{channel="mqtt:topic:BadSpot_5:effect"}
Number BadSpot_5_linkquality				"Bad Spot 5 Linkquality [%s]"				<qualityofservice>	(gBadSpot_5)													{channel="mqtt:topic:BadSpot_5:signal"}
DateTime BadSpot_5_LastSeen					"Bad Spot 5 zuletzt gesehen [%1$td.%1$tm - %1$tH:%1$tM]"<time-on>(gBadSpot_5)													{channel="mqtt:topic:BadSpot_5:lastseen"}

Thats what i use for now, but there will be changes.

Besenreiter, 2021-02-18:
Cool, Thanks!

TheOriginal92, 2021-02-19:
> Hello @TheOriginal92,
> could you also provide the Javascript Transformations you used in your example.

Hey, sure thing
HSBtoRGB.js:

(function (x) {
    if (x=="ON" || x=="INCREASE") {
        return '{"state":"ON"}'
    } else if (x=="OFF" || x=="DECREASE") {
        return '{"state":"OFF"}'
    } else {
        var tex = x.split(',');
        s = parseInt(tex[1]);
        v = parseInt(tex[2]);
        h = parseInt(tex[0]);
    
        h=(!h ? 0 : h/360.0);
        s=(!s ? 0 : s/100.0);
        v=(!v ? 0 : v/100.0);

        i = Math.floor(h * 6);
        f = h * 6 - i;
        p = v * (1 - s);
        q = v * (1 - f * s);
        t = v * (1 - (1 - f) * s);
        switch (i % 6) {
            case 0: r = v, g = t, b = p; break;
            case 1: r = q, g = v, b = p; break;
            case 2: r = p, g = v, b = t; break;
            case 3: r = p, g = q, b = v; break;
            case 4: r = t, g = p, b = v; break;
            case 5: r = v, g = p, b = q; break;
        }
        return '{"brightness":'+ Math.round(v * 255)+',"color":{"rgb":"'+ Math.round(r * 255)+','+ Math.round(g * 255)+','+Math.round(b * 255)+'"}}'
    }
})(input)

setZigbeeCT.js:
(function(x){
	var color_temp = (x * 350) + 150;		// 0-100 = 150-500
	//var color_temp = (500 - (x * 350));		// 0-100 = 500-100
	if(color_temp > 500) color_temp = 500
	if(color_temp < 150) color_temp = 150
    var result = new Object();
    result.color_temp = color_temp;
    return JSON.stringify(result);
})(input)

setZigbeeBrightness.js:
(function(x){
    var brightness = (x/100)*255;
	if(brightness > 255) brightness = 255
	if(brightness < 0) brightness = 0
    var result = new Object();
    result.brightness = brightness;
    return JSON.stringify(result);
})(input)

etZigbeeEffect.js:
(function(x){
	var effectid = x;
	var result = new Object();
    result.effect = effectid;
    return JSON.stringify(result);
})(input)

All the others are basicly just variations of those above :)

JoLander, 2021-02-19:
@TheOriginal92
Thank you 😊

maichai, 2021-02-20:
Sorry If my questions seems weird: I am using the "latest" container and there was no update yet to be installed. How or when do we get those changes to use the integration for the Aldi Lightway lights?

BMOD89, 2021-02-21:
I think you need to use the latest-dev tag. 
I think there is a release each first day of a month, so you would need to wait until 01. March...

maichai, 2021-02-22:
Thanks, they are detected as Aldi Lightway now. The icon looks still ugly. If someone has the time, they could use the same icons as for the Lidl Livarno products, as I think they are all Tuya Zigbee devices.

BMOD89, 2021-02-23:
Do you see the zigbee logo or some kind of bulbs? The pictures are indeed ugly, but these are the official ones 😄

n-pietrutschka, 2021-02-23:
Any chance to add the Megos panels as well?

Koenkk, 2021-02-23:
@n-pietrutschka megos panels have just been added.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Armageddit, 2021-02-23:
where must i add the `-dev`
in compose file?
i use Docker and IOTstack

petrows, 2021-02-27:
Thanks a lot, @Koenkk !

I have bought E27 Filament lamp in ALDI Süd offline shop today and tested it with current `latest-dev` docker image. I can confirm: pairs and works perfectly.

```
Zigbee2MQTT:info  2021-02-27 17:31:13: Device '0x60a423fffe4b9138' joined
Zigbee2MQTT:info  2021-02-27 17:31:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffe4b9138","ieee_address":"0x60a423fffe4b9138"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-02-27 17:31:13: Starting interview of '0x60a423fffe4b9138'
Zigbee2MQTT:info  2021-02-27 17:31:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffe4b9138","ieee_address":"0x60a423fffe4b9138","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-02-27 17:31:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x60a423fffe4b9138"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-02-27 17:31:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x60a423fffe4b9138"},"type":"pairing"}'
Zigbee2MQTT:info  2021-02-27 17:31:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffe4b9138","ieee_address":"0x60a423fffe4b9138"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-02-27 17:31:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x60a423fffe4b9138"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-02-27 17:31:27: Successfully interviewed '0x60a423fffe4b9138', device has successfully been paired
Zigbee2MQTT:info  2021-02-27 17:31:27: Device '0x60a423fffe4b9138' is supported, identified as: Aldi LIGHTWAY smart home LED-lamp - filament (F122SB62H22A4.5W)
Zigbee2MQTT:info  2021-02-27 17:31:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"LIGHTWAY smart home LED-lamp - filament","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":150}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"description":"Configure lighting cluster options attribute.","features":[{"access":7,"description":"Execute color/color_temp changes even when light is off","name":"execute_if_off","property":"execute_if_off","type":"binary","value_off":false,"value_on":true}],"name":"color_options","property":"color_options","type":"composite"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"F122SB62H22A4.5W","supports_ota":false,"vendor":"Aldi"},"friendly_name":"0x60a423fffe4b9138","ieee_address":"0x60a423fffe4b9138","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-02-27 17:31:27: Configuring '0x60a423fffe4b9138'
Zigbee2MQTT:info  2021-02-27 17:31:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"LIGHTWAY smart home LED-lamp - filament","friendly_name":"0x60a423fffe4b9138","model":"F122SB62H22A4.5W","supported":true,"vendor":"Aldi"},"type":"pairing"}'
Zigbee2MQTT:info  2021-02-27 17:31:28: Successfully configured '0x60a423fffe4b9138'
```

maichai, 2021-03-01:
> Do you see the zigbee logo or some kind of bulbs? The pictures are indeed ugly, but these are the official ones 😄

previously I just saw a wifi symbol of some sort (i think it is the default picture)

now I see proper pictures that let you recognize the kind of bulb/socket, good enough for me, nothing to complain about.

Benny93, 2021-03-21:
When I set a color, the bulb becomes very dimm. 

Like 25 percent of its max brightness. Does someone here know how to fix this?
I used the documentation from here https://www.zigbee2mqtt.io/devices/L122CB63H11A9.0W.html
and set the color with mqttFx by publishing 
{"color":{"x":0.3227,"y":0.329}}
to
zigbee2mqtt/WZCouchLamp/set

When setting the color_temp the bulb returns to its max brightness but setting the brightness directly does nothing.
{"color_temp":"warm"} to zigbee2mqtt/WZCouchLamp/set

{"brightness":254} to zigbee2mqtt/WZCouchLamp/set -> does not increase brightness
{"brightness":1} to zigbee2mqtt/WZCouchLamp/set -> lowers the brightness even further

Armageddit, 2021-03-21:
> When I set a color, the bulb becomes very dimm.
> 
> Like 25 percent of its max brightness. Does someone here know how to fix this?
> I used the documentation from here https://www.zigbee2mqtt.io/devices/L122CB63H11A9.0W.html
> and set the color with mqttFx by publishing
> {"color":{"x":0.3227,"y":0.329}}
> to
> zigbee2mqtt/WZCouchLamp/set
> 
> When setting the color_temp the bulb returns to its max brightness but setting the brightness directly does nothing.
> {"color_temp":"warm"} to zigbee2mqtt/WZCouchLamp/set
> 
> {"brightness":254} to zigbee2mqtt/WZCouchLamp/set -> does not increase brightness
> {"brightness":1} to zigbee2mqtt/WZCouchLamp/set -> lowers the brightness even further
> 
> I am using zigbee2mqtt 1.18.1-dev
> 
> Is this a hardware limitation of these bulbs that they cannot reach their full brightness in color mode?

Yes the rgb LEDs are not so bright! 
The white leds are much brighter! 
If you use warm at 1 brightness it's like color at 255 brightness ;)

closed 2021-04-29 by github-actions[bot]