Koenkk/zigbee2mqtt #6350: Pairing device with install code

ozer59, 2021-02-18:
Hello
I try to pairing BOSCH motion detector RFPR-ZB-SH-EU , but pairing process need install code
In Home assistant with ZAH it's work just no battery info

How to process withzigee2mqtt ?
Tanks in advance

link ha forum:
https://community.home-assistant.io/t/request-qr-code-tags-scanning-for-joining-new-zigbee-3-0-device-to-home-assistant-zha-integration-component-via-ui-and-companion-apps/231185

cluster info:
{
"node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4403, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
"endpoints": {
"1": {
"profile_id": 260,
"device_type": "0x0402",
"in_clusters": [
"0x0000",
"0x0001",
"0x0003",
"0x0020",
"0x0400",
"0x0402",
"0x0500",
"0x0b05"
],
"out_clusters": [
"0x0019"
]
}
},
"manufacturer": "Bosch",
"model": "RFPR-ZB-SH-EU",
"class": "zigpy.device.Device"
}

## Debug info
Zigbee2MQTT version: lastest
Adapter hardware:  CC1352P-2
Adapter firmware version:lastest

ozer59, 2021-02-22:
i have 2 install 
1 in production PI 4/ with CC1352P-2 pairing failed motion sensor blinking orange  slow 
"friendly_name":"0x000d6f0012f05d1f","ieee_address":"0x000d6f0012f05d1f","interview_completed":false,"interviewing":false,"network_address":7375,"supported":false,"type":"Unknown"},

2 for test synology docker with CC2652 Stick pairing ok without install code
{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f0012f05d1f","nwkAddr":48403,"manufId":4403,"manufName":"Bosch","powerSource":"Battery","modelId":"RFPR-ZB-SH-EU","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1024,1026,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"RFPR-ZB-SH-EU","manufacturerName":"Bosch","powerSource":3,"zclVersion":1,"appVersion":21,"stackVersion":84,"hwVersion":19,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0021600685","zoneState":1,"currentZoneSensitivityLevel":136}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":84,"hwVersion":19,"dateCode":"","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1613942652398}

pairing ok but i don't know how to create link in docker for devices.js to try support 
i use =>/volume1/docker/app/node_modules/zigbee-herdsman-converters zigbee2mqtt  start but change in devices.js have no effect
folder zigbee-herdsman-converters exist with all files download on git 
in log 
Zigbee2MQTT:warn  2021-02-22 18:40:03: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

if i use /app/node_modules/zigbee-herdsman-converters 
won't start 
in log 
Error: Cannot find module 'https-proxy-agent' | stderr

[image]

877dev, 2021-02-24:
@ozer59 hi, I'd like to keep my issue uncluttered so Koenkk can read it more clearly. 

Can you post your setup here, Docker type and version and I'll do what I can to help 👍

877dev, 2021-02-24:
@ozer59 have you tried this:

## Docker Compose

To setup a local copy of zigbee-herdsman-converters so that you can modify e.g. `devices.js` and from`Zigbee.js`; use the following instructions:

`cd /opt`
`git clone https://github.com/Koenkk/zigbee-herdsman-converters.git`
`cd zigbee-herdsman-converters`
`sudo npm ci`

Now add the volume to your `docker-compose.yml ` 
Example:
```
zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    volumes:
      - ./volumes/zigbee2mqtt/data:/app/data
      - /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters
    devices:
      #- /dev/ttyAMA0:/dev/ttyACM0
      - /dev/ttyACM0:/dev/ttyACM0
    restart: unless-stopped
    network_mode: host
```

To remove local changes simply remove the line from your docker-compose. Don't forget to run `docker-compose up -d ` after making changes to `docker-compose.yml`

closed 2021-04-03 by github-actions[bot]

smekac95, 2023-04-12:
This issues is still here. I bought this model motion sensor, paring with coordinator CC1352P-2 is not succesfull, same error (variant 1) - as @ozer59 also with Z2M version 1.30.3 where is support available this device.

`2023-04-12 20:29:25Failed to interview '0x000d6f0011f8692d', device has not successfully been paired`