Koenkk/zigbee2mqtt #6353: Cannot start zigbee2mqtt with Slaesh's CC2652RB stick: Error while starting zigbee-herdsman

wonx, 2021-02-18:
## What happened
I just bought a Slaesh's CC2652RB adapter and I flashed it with the latest coordinator firmware (https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC2652RB_coordinator_20210120.zip). I tested it using this https://github.com/slaesh/cc2652-stick/tree/master/coordinator-test and seems to work fine. 

The device seems to be mapped at /dev/ttyUSB0:

```
pi@openhab:~/cc2538-bsl $ ls -lia /dev/serial/by-id/*

7341 lrwxrwxrwx 1 root root 13 Feb 18 23:17 /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_4B_7B-if00-port0 -> ../../ttyUSB0
```

So I tried to install zigbee2mqtt using a docker container, following these instructions: https://www.zigbee2mqtt.io/information/docker.html

I pulled the image and ran the following command:

```
docker run \
   -it \
   -v $(pwd)/data:/app/data \
   --device=/dev/ttyUSB0 \
   -e TZ=Europe/Paris \
   -v /run/udev:/run/udev:ro \
   --privileged=true \
   koenkk/zigbee2mqtt
```

But for some reason, the container does not start. 

## What did you expect to happen

It should have start as expected, but instead I get the following error:

```
pi@openhab:~/cc2538-bsl $ docker run -it -v $(pwd)/data:/app/data --device=/dev/ttyUSB0 -e TZ=Europe/Paris -v /run/udev:/run/udev:ro --privileged=true koenkk/zigbee2mqtt
Using '/app/data' as data directory
Creating configuration file...

> zigbee2mqtt@1.17.1 start /app
> node index.js

Zigbee2MQTT:info  2021-02-19 00:11:37: Logging to console and directory: '/app/data/log/2021-02-19.00-11-21' filename: log.txt
Zigbee2MQTT:info  2021-02-19 00:11:38: Starting Zigbee2MQTT version 1.17.1 (commit #99274c7)
Zigbee2MQTT:info  2021-02-19 00:11:38: Starting zigbee-herdsman (0.13.59)
Zigbee2MQTT:error 2021-02-19 00:11:45: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-19 00:11:45: Failed to start zigbee
Zigbee2MQTT:error 2021-02-19 00:11:45: Exiting...
Zigbee2MQTT:error 2021-02-19 00:11:45: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.17.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-02-18T23_11_46_432Z-debug.log

```

I see that the most relevant error refers to `No such file or directory, cannot open /dev/ttyACM0'` but in my case the device should be `/dev/ttyUSB0`.

Any ideas what could have gone wrong?

Any help would be appreciated.

## Debug info
Zigbee2MQTT version: 1.17.1 (commit #99274c7)
Adapter hardware: CC2652RB
Adapter firmware version: CC2652RB_coordinator_20210120

m0wlheld, 2021-02-19:
That‘s not a bug. 

    Creating configuration file...
    
You started from scratch, using default configuration. It assumes to find the coordinator-device at /dev/ACM0. 

Either you configure 'serial' in configuration.yaml or map the device to /dev/ACM0. 

```
docker run \
   -it \
   -v $(pwd)/data:/app/data \
   --device=/dev/ttyUSB0:/dev/ACM0 \
   -e TZ=Europe/Paris \
   koenkk/zigbee2mqtt
```

PS: it‘s even better to use the full /dev/serial/... path, as other paths might change depending on the count and plug in order of your USB devices.

wonx, 2021-02-19:
Hi, thank you for your response. 

I already tried that option, but I still receive the same error. When I run:

```
docker run \
   -it \
   -v $(pwd)/data:/app/data \
   --device=/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_4B_7B-if00-port0:/dev/ACM0 \
   -e TZ=Europe/Paris \
   --name=zigbee2mqtt \
   koenkk/zigbee2mqtt
```

I get: 

`Zigbee2MQTT:error 2021-02-19 16:06:02: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'`

 It's as if the docker image does not get that parameter with the device.

m0wlheld, 2021-02-19:
Sorry, my bad, the target device is /dev/ttyACM0, not /dev/ACM0 (as I proposed). So it's:

```
docker run \
   -it \
   -v $(pwd)/data:/app/data \
   --device=/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_4B_7B-if00-port0:/dev/ttyACM0 \
   -e TZ=Europe/Paris \
   --name=zigbee2mqtt \
   koenkk/zigbee2mqtt
```

... for testing purposes. For productive use discard the "-it" argument and add "-d --restart=always"

wonx, 2021-02-19:
Great! It starts now. Now it's time have fun configuring it.

Thank you!

closed 2021-02-19 by wonx