Koenkk/zigbee2mqtt #6354: Moes ZK-EU-2U - wrong detection

ROEEROEEROEE, 2021-02-19:
## What happened 
so i've added Moes ZK-EU-2U trough z2m | i've try to do this with both regular and the edge z2m versions
and it's been recognized as Nue / 3A HGZB-01 #
https://www.zigbee2mqtt.io/devices/HGZB-01.html
and it's shown in home assistant as a light and when i turn off the main plug [the socket] 
it turns off also the two usb port as well, making them useless 

## What did you expect to happen - 
to z2m add it as Moes ZK-EU-2U and expose the socket and 2 other usb ports
and to have control on all of this product features [1 main plug and 2 additional usb ports]
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, 
Adapter firmware version:

ROEEROEEROEE, 2021-02-20:
also, worth to mention - 
i've try to do as wrote in #6228 and i'm working with zigbee2mqtt edge [updated - un and re install]
and nothing, the device is Moes ZK-EU-2U and it's recognize as Nue / 3A HGZB-01 

when i'm switching off from ha, it's shut down all 3 . main plug and 2 other usb - making them worthless
since i've purchased this device in order to have the ability to control each one of the 3 individualy

### this is what is in /share/zigbee2mqtt/database.db -

{"id":21,"type":"Router","ieeeAddr":"0x086bd7fffe3f8e95","nwkAddr":19866,"manufId":4478,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN59-1S7LX1.0","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"LXN59-1S7LX1.0","manufacturerName":"3A Smart Home DE","powerSource":0,"zclVersion":3,"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20200519"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e2b5b4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20200519","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1613857663016}

### this is what it expose -
    },
    "0x086bd7fffe3f8e95": {
        "state": "ON"
    }

any idea on how to fix this?
thanks in advance, well appreciated.

Koenkk, 2021-02-21:
Do you know if it's possible to control the USB separately using the original gateway? Given the fact that this device only has 1 endpoint this looks unlikely. (if the usb ports could be controlled separately I would expect 3 endpoints)

ROEEROEEROEE, 2021-02-22:
quoting from #6228 
I just added a Moes ZK-EU-2U to my network. It has 2 exposes: Endpoint: l1 and Endpoint: l2
Unfortunately the Endpoint: l2 doesn't work via Z2M. It means when I try to switch it wia Z2M it gives an error and when I switch it on the device it gives "state_undefined" instead of "state_l2".
The logs:
MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe932122', payload '{"linkquality":123,"state_l1":"OFF","state_undefined":"ON"}'

Error Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'getDevice' of undefined at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.17.1/lib/extension/publish.js:174:43) at Controller.callExtensionMethod (/zigbee2mqtt-1.17.1/lib/controller.js:382:44) at runMicrotasks () at processTicksAndRejections (internal/process/task_queues.js:93:5))

end of quote

as per my understanding, it should expose the usb as endpoint.
this is what i've understood from the manufacturer spec.

any idea, based on #6228 ?

ROEEROEEROEE, 2021-02-22:
also, 

in - Fixing ZigBee3.0 dual USB wireless socket plug #4503

it's mentioned the usb control is doable [in the copied line " socket on, usb off"] 
so i am taking into consideration it is possible, yet i try with no successes.

i would really appreciated if you could help me solve it [reading here in git, it seems you are a magician :) ]

thanks.

ROEEROEEROEE, 2021-02-23:
Oh no, 
i see your point. so basically this one [i owned] is not as described, meaning what dose it worth if i only can
switch all of them together. 

anyway, i do appreciate your help it's not obvious.

p.s. can you recommend another device that supported by Zigbee2mqtt that have a main socket and usb port/s 
to be controlled separately ?

Koenkk, 2021-02-23:
@ROEEROEEROEE I don't know of such device. But you can check on the supported device page. 

If you look e.g. at https://www.zigbee2mqtt.io/devices/E220-KR4N0Z0-HA.html you see it exposes 4 endpoints (l1 till l4) and it has 4 outlets so probably each endpoint controls 1 outlets

closed 2021-02-23 by Koenkk

ROEEROEEROEE, 2021-02-24:
Thanks a lot