Koenkk/zigbee2mqtt #6359: Philips RWL022

mabosch74, 2021-02-19:
Device has already been added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2202.

It's supported in the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

_Originally posted by @Koenkk in https://github.com/Koenkk/zigbee2mqtt/issues/6247#issuecomment-778759440_

Koenkk, 2021-02-19:
If you paired it before you switched to the dev branch try removing the device and pairing it again.

cthybert, 2021-02-19:
> If you paired it before you switched to the dev branch try removing the device and pairing it again.

same issue here, tried to remove and pair again, no error message and no mqtt messages sent while clicking : 

here info log : 
```

info  2021-02-19 21:22:32: Device '0x0017880109258650' joined
info  2021-02-19 21:22:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880109258650","ieee_address":"0x0017880109258650"},"type":"device_joined"}'
info  2021-02-19 21:22:32: Starting interview of '0x0017880109258650'
info  2021-02-19 21:22:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880109258650","ieee_address":"0x0017880109258650","status":"started"},"type":"device_interview"}'
info  2021-02-19 21:22:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0017880109258650"},"type":"device_connected"}'
info  2021-02-19 21:22:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0017880109258650"},"type":"pairing"}'
info  2021-02-19 21:22:36: Successfully interviewed '0x0017880109258650', device has successfully been paired
info  2021-02-19 21:22:36: Device '0x0017880109258650' is supported, identified as: Philips Hue dimmer switch (324131092621)
info  2021-02-19 21:22:36: MQTT publish: topic 'homeassistant/sensor/0x0017880109258650/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880109258650"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880109258650","sw_version":"Zigbee2MQTT 1.17.1-dev"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0x0017880109258650","name":"0x0017880109258650 battery","state_topic":"zigbee2mqtt/0x0017880109258650","unique_id":"0x0017880109258650_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2021-02-19 21:22:36: MQTT publish: topic 'homeassistant/sensor/0x0017880109258650/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880109258650"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880109258650","sw_version":"Zigbee2MQTT 1.17.1-dev"},"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x0017880109258650","name":"0x0017880109258650 action","state_topic":"zigbee2mqtt/0x0017880109258650","unique_id":"0x0017880109258650_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
info  2021-02-19 21:22:36: MQTT publish: topic 'homeassistant/sensor/0x0017880109258650/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880109258650"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880109258650","sw_version":"Zigbee2MQTT 1.17.1-dev"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x0017880109258650","name":"0x0017880109258650 linkquality","state_topic":"zigbee2mqtt/0x0017880109258650","unique_id":"0x0017880109258650_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2021-02-19 21:22:36: MQTT publish: topic 'homeassistant/sensor/0x0017880109258650/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880109258650"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880109258650","sw_version":"Zigbee2MQTT 1.17.1-dev"},"icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x0017880109258650","name":"0x0017880109258650 update state","state_topic":"zigbee2mqtt/0x0017880109258650","unique_id":"0x0017880109258650_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
info  2021-02-19 21:22:36: MQTT publish: topic 'homeassistant/binary_sensor/0x0017880109258650/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880109258650"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880109258650","sw_version":"Zigbee2MQTT 1.17.1-dev"},"json_attributes_topic":"zigbee2mqtt/0x0017880109258650","name":"0x0017880109258650 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x0017880109258650","unique_id":"0x0017880109258650_update_available_zigbee2mqtt","value_template":"{{ value_json.update_available}}"}'
info  2021-02-19 21:22:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Hue dimmer switch","friendly_name":"0x0017880109258650","model":"324131092621","supported":true,"vendor":"Philips"},"type

```

Koenkk, 2021-02-19:
Can you try configuring the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceconfigure , right before sending this command wakeup the RWL022 by clicking a button on it.

Now check if it works and post the debug log here if it doesn't.

ferdyh, 2021-02-19:
I'm running into the same issues. It pairs, but it doesn't register the actions. If i do a reconfigure, I get the following messages:

`Zigbee2MQTT:error 2021-02-19 23:56:00: Failed to configure 'woonkamker_light_switch', attempt 4 (TypeError: Cannot read property 'write' of undefined
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices.js:3975:29)
    at async Configure.configure (/app/lib/extension/configure.js:131:13)
    at async Configure.onMQTTMessage (/app/lib/extension/configure.js:82:21)
    at async Controller.callExtensionMethod (/app/lib/controller.js:385:21))
Zigbee2MQTT:info  2021-02-19 23:56:00: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"woonkamker_light_switch"},"error":"Failed to configure (Cannot read property 'write' of undefined)","status":"error","transaction":"d73b0-2"}'`

cthybert, 2021-02-20:
still not working : 

I renamed the device : dimmer_bureau

```
debug 2021-02-20 11:21:52: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data 'dimmer_bureau'
info  2021-02-20 11:21:52: Configuring 'dimmer_bureau'
debug 2021-02-20 11:22:01: Received Zigbee message from 'dimmer_bureau', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID null
debug 2021-02-20 11:22:01: Received Zigbee message from 'dimmer_bureau', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":30,"transtime":9}' from endpoint 1 with groupID null
error 2021-02-20 11:22:03: Failed to configure 'dimmer_bureau', attempt 4 (TypeError: Cannot read property 'write' of undefined
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices.js:3975:29)
    at async Configure.configure (/app/lib/extension/configure.js:131:13)
    at async Configure.onMQTTMessage (/app/lib/extension/configure.js:82:21)
    at async Controller.callExtensionMethod (/app/lib/controller.js:385:21))
info  2021-02-20 11:22:03: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"dimmer_bureau"},"error":"Failed to configure (Cannot read property 'write' of undefined)","status":"error"}'

```

Koenkk, 2021-02-20:
The issue is `error 2021-02-20 11:22:03: Failed to configure 'dimmer_bureau', attempt 4 (TypeError: Cannot read property 'write' of undefined` .

Can you provide the data/database.db entry of this device?

mabosch74, 2021-02-20:
here's mine, with the same effect as by cthybert:
`{"id":81,"type":"EndDevice","ieeeAddr":"0x00178801092699c3","nwkAddr":38502,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Battery","modelId":"RWL022","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2096,"inClusterList":[0,1,3,64512,4096],"outClusterList":[25,0,3,4,6,8,5,4096],"clusters":{"genBasic":{"attributes":{"modelId":"RWL022","manufacturerName":"Signify Netherlands B.V.","powerSource":3,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20200622","swBuildId":"2.42.0_h1F5E860"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5fdea","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5fdea","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20200622","swBuildId":"2.42.0_h1F5E860","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1613859342812}`

cthybert, 2021-02-21:
same story 

`{"id":19,"type":"EndDevice","ieeeAddr":"0x0017880109258650","nwkAddr":36993,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Battery","modelId":"RWL022","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2096,"inClusterList":[0,1,3,64512,4096],"outClusterList":[25,0,3,4,6,8,5,4096],"clusters":{"genBasic":{"attributes":{"modelId":"RWL022","manufacturerName":"Signify Netherlands B.V.","powerSource":3,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20200622","swBuildId":"2.42.0_h1F5E860"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff065fa4","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00212effff065fa4","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20200622","swBuildId":"2.42.0_h1F5E860","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1613818941036}`

Koenkk, 2021-02-21:
Does it work after changing the definition in devices.js to:

```js
{
    zigbeeModel: ['RWL020', 'RWL021', 'RWL022'],
    model: '324131092621',
    vendor: 'Philips',
    description: 'Hue dimmer switch',
    fromZigbee: [fz.ignore_command_on, fz.ignore_command_off, fz.ignore_command_step, fz.ignore_command_stop,
        fz.legacy.hue_dimmer_switch, fz.battery],
    exposes: [e.battery(), e.action(['on-press', 'on-hold', 'on-hold-release', 'up-press', 'up-hold', 'up-hold-release',
        'down-press', 'down-hold', 'down-hold-release', 'off-press', 'off-hold', 'off-hold-release'])],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);

        const endpoint2 = device.getEndpoint(2) || device.getEndpoint(1);
        const options = {manufacturerCode: 0x100B, disableDefaultResponse: true};
        await endpoint2.write('genBasic', {0x0031: {value: 0x000B, type: 0x19}}, options);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['manuSpecificPhilips', 'genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint2);
    },
    endpoint: (device) => {
        return {'ep1': 1, 'ep2': 2};
    },
    ota: ota.zigbeeOTA,
},
```

make sure to follow https://github.com/Koenkk/zigbee2mqtt/issues/6359#issuecomment-782416732 after changing

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

mabosch74, 2021-02-21:
added " || device.getEndpoint(1);", pressing some key on the remote and make a configure, the result:
`debug 2021-02-21 11:18:21: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"0x00178801092699c3","transaction":"rrtr1-2"}'
info  2021-02-21 11:18:21: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x00178801092699c3"},"status":"ok","transaction":"rrtr1-2"}'`

pushed every key on the remote with this results:
`debug 2021-02-21 11:22:56: Received Zigbee message from '0x00178801092699c3', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2021-02-21 11:22:56: Received Zigbee message from '0x00178801092699c3', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":30,"transtime":9}' from endpoint 1 with groupID 0
debug 2021-02-21 11:22:57: Received Zigbee message from '0x00178801092699c3', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":30,"transtime":9}' from endpoint 1 with groupID 0
debug 2021-02-21 11:22:59: Received Zigbee message from '0x00178801092699c3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":33565184,"imageType":281,"manufacturerCode":4107}' from endpoint 1 with groupID 0
debug 2021-02-21 11:23:09: Received Zigbee message from '0x00178801092699c3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":33565184,"imageType":281,"manufacturerCode":4107}' from endpoint 1 with groupID 0
debug 2021-02-21 11:23:19: Received Zigbee message from '0x00178801092699c3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":33565184,"imageType":281,"manufacturerCode":4107}' from endpoint 1 with groupID 0`

no mqtt publising... :-(

I don't know if it's a good idea to group the RWL020 and RWL021 with the new RWL022, the firmware of the RWL022 is a Version 2.x the RWL020 and RWL021 is a Version 1.x. Or it's this okay for OTA-Updates?

markusbona, 2021-02-21:
Works for me now. I put your proposed changes as a separate block for only the device RWL022. After that, i reconfigured the device like you suggested.

mabosch74, 2021-02-21:
for me not.

@cthybert any feedback?

@Koenkk this new remote control, should be separated from the old RWL's. The correct modelno. for the RWL022 is 929001173761.

cthybert, 2021-02-21:
it's working doing @markusbona 's way.

I removed "RWL022" off the first block : 
```

    {
        zigbeeModel: ['RWL020', 'RWL021'],
        model: '324131092621',
        vendor: 'Philips',
        description: 'Hue dimmer switch',
        fromZigbee: [fz.ignore_command_on, fz.ignore_command_off, fz.ignore_command_step, fz.ignore_command_stop,
            fz.legacy.hue_dimmer_switch, fz.battery],
        exposes: [e.battery(), e.action(['on-press', 'on-hold', 'on-hold-release', 'up-press', 'up-hold', 'up-hold-release',
            'down-press', 'down-hold', 'down-hold-release', 'off-press', 'off-hold', 'off-hold-release'])],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);

            const endpoint2 = device.getEndpoint(2);
            const options = {manufacturerCode: 0x100B, disableDefaultResponse: true};
            await endpoint2.write('genBasic', {0x0031: {value: 0x000B, type: 0x19}}, options);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['manuSpecificPhilips', 'genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint2);
        },
        endpoint: (device) => {
            return {'ep1': 1, 'ep2': 2};
        },
        ota: ota.zigbeeOTA,
    },
    {
        zigbeeModel: ['RWL022'],
        model: '324131092621',
        vendor: 'Philips',
        description: 'Hue dimmer switch',
        fromZigbee: [fz.ignore_command_on, fz.ignore_command_off, fz.ignore_command_step, fz.ignore_command_stop,
            fz.legacy.hue_dimmer_switch, fz.battery],
        exposes: [e.battery(), e.action(['on-press', 'on-hold', 'on-hold-release', 'up-press', 'up-hold', 'up-hold-release',
            'down-press', 'down-hold', 'down-hold-release', 'off-press', 'off-hold', 'off-hold-release'])],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    
            const endpoint2 = device.getEndpoint(2) || device.getEndpoint(1);
            const options = {manufacturerCode: 0x100B, disableDefaultResponse: true};
            await endpoint2.write('genBasic', {0x0031: {value: 0x000B, type: 0x19}}, options);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['manuSpecificPhilips', 'genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint2);
        },
        endpoint: (device) => {
            return {'ep1': 1, 'ep2': 2};
        },
        ota: ota.zigbeeOTA,
    },
```

Koenkk, 2021-02-21:
Does anybody have the the original Hue gateway, if yes can you check if this remote can be configured to assign custom actions? (I know the old hue remotes can do this)

luchtgitaar, 2021-02-23:
> Does anybody have the the original Hue gateway, if yes can you check if this remote can be configured to assign custom actions? (I know the old hue remotes can do this)

I have a Hue Gateway (v2) and this new dimmer switch. What do you mean by custom actions, so I can check?

Koenkk, 2021-02-23:
@luchtgitaar actions like this: https://community.hueessentials.com/t/hue-dimmer-switch-hold-button/135

luchtgitaar, 2021-02-23:
> @luchtgitaar actions like this: https://community.hueessentials.com/t/hue-dimmer-switch-hold-button/135

In the official Hue app it is possible to assign action to single click On/Off button, Hold On/Off button and one to quadruple click Hue button.

877dev, 2021-02-23:
@cthybert configuration works for me too, thanks for posting to make it easier.

[hue_dimmer_v2_actions.log](https://github.com/Koenkk/zigbee2mqtt/files/6031759/hue_dimmer_v2_actions.log)

877dev, 2021-02-24:
@Koenkk that is working great for me in latest-dev, thanks!

```
Zigbee2MQTT:info  2021-02-24 17:48:39: Successfully interviewed '0x001788010925bb8e', device has successfully been paired
Zigbee2MQTT:info  2021-02-24 17:48:39: Device '0x001788010925bb8e' is supported, identified as: Philips Hue dimmer switch (929002398602)
Zigbee2MQTT:info  2021-02-24 17:48:39: MQTT publish: topic 'zigbee2mqtt_test/bridge/event', payload '{"data":{"definition":{"description":"Hue dimmer switch","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["on_press","on_hold","on_press_release","on_hold_release","off_press","off_hold","off_press_release","off_hold_release","up_press","up_hold","up_press_release","up_hold_release","down_press","down_hold","down_press_release","down_hold_release"]},{"access":1,"description":"Link quality (signalstrength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"929002398602","supports_ota":true,"vendor":"Philips"},"friendly_name":"0x001788010925bb8e","ieee_address":"0x001788010925bb8e","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-02-24 17:48:39: Configuring '0x001788010925bb8e'
Zigbee2MQTT:info  2021-02-24 17:48:39: MQTT publish: topic 'zigbee2mqtt_test/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Hue dimmer switch","friendly_name":"0x001788010925bb8e","model":"929002398602","supported":true,"vendor":"Philips"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-02-24 17:48:42: Received Zigbee message from '0x001788010925bb8e', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":145}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-24 17:48:42: MQTT publish: topic 'zigbee2mqtt_test/0x001788010925bb8e', payload '{"battery":72.5,"linkquality":147}'
Zigbee2MQTT:info  2021-02-24 17:48:42: Successfully configured '0x001788010925bb8e'
```

Koenkk, 2021-02-24:
1. Disabled OTA, so should be fixed now.
2. Can you provide the debug log when pressing the button? Could be that you see nothing, then try unbinding it
To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

877dev, 2021-02-25:
> 2\. Can you provide the debug log when pressing the button?

```
Zigbee2MQTT:debug 2021-02-25 12:02:49: Received Zigbee message from 'Hue_dimmer_main_bedroom', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0

Zigbee2MQTT:info  2021-02-25 12:02:49: MQTT publish: topic 'zigbee2mqtt/Hue_dimmer_main_bedroom', payload '{"action":"off_press","battery":100,"counter":1,"linkquality":39}'

Zigbee2MQTT:debug 2021-02-25 12:02:50: Received Zigbee message from 'Hue_dimmer_main_bedroom', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":2,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0

Zigbee2MQTT:info  2021-02-25 12:02:50: MQTT publish: topic 'zigbee2mqtt/Hue_dimmer_main_bedroom', payload '{"action":"off_press_release","battery":100,"counter":1,"linkquality":39}'
```

Koenkk, 2021-02-25:
Can you try to comment https://github.com/Koenkk/zigbee-herdsman-converters/blob/2395e4445a9beb605a823e4767fec6710d5658bd/devices.js#L4057 , factory reset and repair the device and try to bind now?

877dev, 2021-02-26:
I did a git pull which seems to have pulled down the latest devices.js with your updates:
```
git clone https://github.com/Koenkk/zigbee-herdsman-converters.git
cd zigbee-herdsman-converters
docker run -v "$PWD":/app -w /app --rm node:12 npm ci
```

Had to force remove the Hue dimmer v2, then reset and repair successfully.

Now I am sending the bind command but it's failing, I am pressing button just before:

```
Zigbee2MQTT:info  2021-02-26 10:23:16: MQTT publish: topic 'zigbee2mqtt/Hue_dimmer_main_bedroom', payload '{"action":"on_press_release","battery":69,"linkquality":47}'
Zigbee2MQTT:debug 2021-02-26 10:23:17: Received Zigbee message from 'Hue_dimmer_main_bedroom', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-02-26 10:23:17: Received MQTT message on 'zigbee2mqtt/bridge/request/device/bind' with data '{"from": "Hue_dimmer_main_bedroom", "to": "Hue_light_main_bedroom"}'
Zigbee2MQTT:debug 2021-02-26 10:23:17: binding cluster 'genScenes' from 'Hue_dimmer_main_bedroom' to 'Hue_light_main_bedroom'
Zigbee2MQTT:error 2021-02-26 10:23:27: Failed to bind cluster 'genScenes' from 'Hue_dimmer_main_bedroom' to 'Hue_light_main_bedroom' (Error: Bind 0x001788010925bb8e/1 genScenes from '0x00178801064d3257/11' failed (AREQ - ZDO - bindRsp after 10000ms))
Zigbee2MQTT:info  2021-02-26 10:23:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genScenes","from":"Hue_dimmer_main_bedroom","to":"Hue_light_main_bedroom"},"type":"device_bind_failed"}'
Zigbee2MQTT:debug 2021-02-26 10:23:27: binding cluster 'genOnOff' from 'Hue_dimmer_main_bedroom' to 'Hue_light_main_bedroom'
Zigbee2MQTT:error 2021-02-26 10:23:37: Failed to bind cluster 'genOnOff' from 'Hue_dimmer_main_bedroom' to 'Hue_light_main_bedroom' (Error: Bind 0x001788010925bb8e/1 genOnOff from '0x00178801064d3257/11' failed (AREQ - ZDO - bindRsp after 10000ms))
Zigbee2MQTT:info  2021-02-26 10:23:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"Hue_dimmer_main_bedroom","to":"Hue_light_main_bedroom"},"type":"device_bind_failed"}'
Zigbee2MQTT:debug 2021-02-26 10:23:37: binding cluster 'genLevelCtrl' from 'Hue_dimmer_main_bedroom' to 'Hue_light_main_bedroom'
Zigbee2MQTT:error 2021-02-26 10:23:47: Failed to bind cluster 'genLevelCtrl' from 'Hue_dimmer_main_bedroom' to 'Hue_light_main_bedroom' (Error: Bind 0x001788010925bb8e/1 genLevelCtrl from '0x00178801064d3257/11' failed (AREQ - ZDO - bindRsp after 10000ms))
Zigbee2MQTT:info  2021-02-26 10:23:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"Hue_dimmer_main_bedroom","to":"Hue_light_main_bedroom"},"type":"device_bind_failed"}'
Zigbee2MQTT:info  2021-02-26 10:23:47: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":["genScenes","genOnOff","genLevelCtrl"],"from":"Hue_dimmer_main_bedroom","to":"Hue_light_main_bedroom"},"error":"Failed to bind","status":"error"}'
Zigbee2MQTT:error 2021-02-26 10:23:47: Failed to bind
```

Koenkk, 2021-02-26:
@877dev did you also modify devices.js as I proposed in https://github.com/Koenkk/zigbee2mqtt/issues/6359#issuecomment-786203207 ?

877dev, 2021-02-26:
@Koenkk yes, the git clone version of devices.js is the same, including this:

`await endpoint.write('genBasic', {0x0031: {value: 0x000B, type: 0x19}}, options);`

Koenkk, 2021-02-26:
@877dev yes but you should comment (aka remove) this line and then factory reset + repair and bind again

877dev, 2021-02-26:
@Koenkk sorry I misunderstood, I will do it right now and report back ASAP.

877dev, 2021-02-26:
@Koenkk 

```
Zigbee2MQTT:debug 2021-02-26 18:34:04: Received Zigbee message from 'Hue_dimmer_main_bedroom', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-26 18:34:04: MQTT publish: topic 'zigbee2mqtt/Hue_dimmer_main_bedroom', payload '{"action":"off_press","battery":73.5,"linkquality":76}'
Zigbee2MQTT:debug 2021-02-26 18:34:04: Received Zigbee message from 'Hue_dimmer_main_bedroom', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":3,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-26 18:34:04: MQTT publish: topic 'zigbee2mqtt/Hue_dimmer_main_bedroom', payload '{"action":"off_press_release","battery":73.5,"linkquality":81}'
Zigbee2MQTT:debug 2021-02-26 18:34:05: Received Zigbee message from 'Hue_light_main_bedroom', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
Zigbee2MQTT:info  2021-02-26 18:34:05: MQTT publish: topic 'zigbee2mqtt/Hue_light_main_bedroom', payload '{"brightness":253,"color":{"x":0.459857791746717,"y":0.410598847355503},"color_temp":370,"color_temp_startup":65535,"linkquality":76,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

877dev, 2021-02-26:
Actually after some help on the node red forum, there is an excellent solution with node red function:

`[{"id":"f1e7cf45.48393","type":"inject","z":"be5275e8.e06c78","name":"off_press","props":[{"p":"payload.action","v":"off_press","vt":"str"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"zigbee2mqtt/Hue_dimmer_main_bedroom","x":1100,"y":860,"wires":[["43f0731d.f5780c"]]},{"id":"43f0731d.f5780c","type":"function","z":"be5275e8.e06c78","name":"","func":"// If payload matches\nif (msg.payload.action === \"off_press\")\n{\nvar opts1 = [\n    {\"color_temp\": \"coolest\", \"transition\":0.5},\n    {\"color_temp\": \"cool\", \"transition\":0.5},\n    {\"color_temp\": \"neutral\", \"transition\":0.5},\n    {\"color_temp\": \"warm\", \"transition\":0.5},\n    {\"color_temp\": \"warmest\", \"transition\":0.5}\n];\n\nvar opt1 = context.get(\"opt1\") || 0;\nmsg.payload = opts1[opt1];\n\nopt1++;\nif(opt1 >= opts1.length) opt1 = 0;\ncontext.set(\"opt1\", opt1); \n\nreturn msg;\n}\n\n\n\n// If payload matches\nif (msg.payload.action === \"off_hold\")\n{\nvar opts2 = [\n    {\"color\": {\"r\":255,\"g\":0,\"b\":0}, \"transition\":0.5},\n    {\"color\": {\"r\":0,\"g\":255,\"b\":0}, \"transition\":0.5},\n    {\"color\": {\"r\":0,\"g\":0,\"b\":255}, \"transition\":0.5}\n];\n\nvar opt2 = context.get(\"opt2\") || 0;\nmsg.payload = opts2[opt2];\n\nopt2++;\nif(opt2 >= opts2.length) opt2 = 0;\ncontext.set(\"opt2\", opt2); \n\nreturn msg;\n}\n\n\n\n\n\n\n\n\n\n\n\n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":1310,"y":880,"wires":[["c54e199.d982fe8","a30f27dc.d12dc8"]]},{"id":"c54e199.d982fe8","type":"debug","z":"be5275e8.e06c78","name":"","active":true,"tosidebar":true,"console":false,"tostatus":true,"complete":"payload","targetType":"msg","statusVal":"payload","statusType":"auto","x":1500,"y":880,"wires":[]},{"id":"ace5d94.01d6928","type":"inject","z":"be5275e8.e06c78","name":"off_hold","props":[{"p":"payload.action","v":"off_hold","vt":"str"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"zigbee2mqtt/Hue_dimmer_main_bedroom","x":1110,"y":910,"wires":[["43f0731d.f5780c"]]}]`

877dev, 2021-04-03:
Using latest 1.18.2 I am getting this when pressing the action/scene button (button 4):
Hopefully this means the button is freely configurable by the user?

```
Zigbee2MQTT:debug 2021-04-03 12:52:19: Received Zigbee message from 'Hue_dimmer_hallway', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-04-03 12:52:19: MQTT publish: topic 'zigbee2mqtt/Hue_dimmer_hallway', payload '{"action":"off_press","battery":100,"linkquality":117}'
Zigbee2MQTT:debug 2021-04-03 12:52:19: Received Zigbee message from 'Hue_dimmer_hallway', type 'commandRecall', cluster 'genScenes', data '{"groupid":58081,"sceneid":0}' from endpoint 1 with groupID 3
Zigbee2MQTT:debug 2021-04-03 12:52:19: No converter available for '929002398602' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":58081,"sceneid":0}'
Zigbee2MQTT:debug 2021-04-03 12:52:19: Received Zigbee message from 'Hue_dimmer_hallway', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":2,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-04-03 12:52:19: MQTT publish: topic 'zigbee2mqtt/Hue_dimmer_hallway', payload '{"action":"off_press_release","battery":100,"linkquality":117}'
```

grubyak, 2022-01-09:
@Koenkk both https://www.zigbee2mqtt.io/devices/929002398602.html and https://www.zigbee2mqtt.io/devices/324131092621.html dimmers have a scene toggle function (with consecutive press on "on" or "hue" buttons) that gets unavailable after pairing. Is it possible to get that functionality within the dimmer itself without using node red as suggested above?

After enabling debug I see that message from dimmer on consecutive press on "on" / "hue" is always:
`type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID 0` so seems that dimmer does not cycle on different scenes. Was it expected to have additional property in the message that would indicate cycling?

Commenting out `await endpoint2.write('genBasic', {0x0031: {value: 0x000B, type: 0x19}}, options);` does not help.