Koenkk/zigbee2mqtt #6366: 1.16.2 has broken by Tuya dimmer gq8b1uv compared to 1.16.1

peca89, 2021-02-19:
## What happened
Just noticed (using latest 1.17.1) that brightness controll on my Zigbee smart dimmer (gq8b1uv) does not work. Moving the slider in Home Assistant does not change brightness. In z2m logs, after every slider move, it just sends a message with brightness 254. On-off controll works fine. 

One more thing to note is that "Effect" propery is discovered through Home Assistant in the affecting z2m versions. This device does not support effects.

Comparing the versions, it seems that a large change was introduced to "tuya_dimmer_level" function within toZigbee.js file. I don't know if that is related.

## What did you expect to happen
Brightness to work properly as it did on 1.16.1 (last version where it worked)

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.16.2 through 1.17.1
Adapter hardware: CC2531

Koenkk, 2021-02-19:
- In the `tuya_dimmer_level` (`toZigbee.js`) converter, can you try changing `let dp = tuya.dataPoints.dimmerLevel;` to `let dp = 515;`
- Can you also provide the data/database.db entry of this device?
- For future reference, working version: https://github.com/Koenkk/zigbee-herdsman-converters/blob/c1593eac133130d54160ab46aa6a8486d5cb987b/converters/toZigbee.js#L2725

peca89, 2021-02-19:
- After changing to `let dp = 515;` the following error message pops up and it still does not work

````
Debug Received MQTT message on 'zigbee2mqtt/0xec1bbdfffec3e6d1/set' with data '{"state": "ON", "brightness": 188}'
Debug Publishing 'set' 'state' to '0xec1bbdfffec3e6d1'
Debug Publishing 'set' 'brightness' to '0xec1bbdfffec3e6d1'
Error Publish 'set' 'brightness' to '0xec1bbdfffec3e6d1' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 0xec1bbdfffec3e6d1/1 manuSpecificTuya.setData({"status":0,"transid":12,"dp":515,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,2,225]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 515)'
Debug RangeError [ERR_OUT_OF_RANGE]: Command 0xec1bbdfffec3e6d1/1 manuSpecificTuya.setData({"status":0,"transid":12,"dp":515,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,2,225]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 515) at writeU_Int8 (internal/buffer.js:555:11) at Buffer.writeUInt8 (internal/buffer.js:565:10) at BuffaloZcl.writeUInt8 (c:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\buffalo\buffalo.js:30:21) at BuffaloZcl.write (c:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\buffalo\buffalo.js:206:18) at BuffaloZcl.write (c:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\zcl\buffaloZcl.js:303:26) at ZclFrame.writePayloadCluster (c:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\zcl\zclFrame.js:144:21) at ZclFrame.toBuffer (c:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\zcl\zclFrame.js:74:18) at ZStackAdapter.<anonymous> (c:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\adapter\z-stack\adapter\zStackAdapter.js:317:163) at Generator.next (<anonymous>) at c:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\adapter\z-stack\adapter\zStackAdapter.js:27:71
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to '0xec1bbdfffec3e6d1' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 0xec1bbdfffec3e6d1/1 manuSpecificTuya.setData({\"status\":0,\"transid\":12,\"dp\":515,\"datatype\":2,\"length_hi\":0,\"length_lo\":4,\"data\":[0,0,2,225]}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (The value of \"value\" is out of range. It must be >= 0 and <= 255. Received 515)'","meta":{"friendly_name":"0xec1bbdfffec3e6d1"},"type":"zigbee_publish_error"}'
Debug Received Zigbee message from '0xec1bbdfffec3e6d1', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":11}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffec3e6d1', payload '{"brightness":254,"level":1000,"linkquality":126,"state":"ON"}'
````

- database.db entry:

````
{"id":13,"type":"Router","ieeeAddr":"0xec1bbdfffec3e6d1","nwkAddr":2135,"manufId":0,"manufName":"_TYST11_sgq8b1uv","powerSource":"Mains (single phase)","modelId":"gq8b1uv\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"gq8b1uv\u0000","manufacturerName":"_TYST11_sgq8b1uv","powerSource":1,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e22117","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e22117","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1613756513659}
````

Koenkk, 2021-02-19:
Could you provide the herdsman debug log when sending this command with both 1.16.1 and 1.17.1?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

peca89, 2021-02-19:
OK, this was real pain to collect under Windows :) I managed to set the environment variable and run z2m by entering `set DEBUG=zigbee-herdsman*&&node index.js` under normal command prompt. You should add this to the guide.

Anyway, non working 1.17.1
````
Zigbee2MQTT:debug 2021-02-19 21:11:14: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffec3e6d1/set' with data '{"state": "ON", "brightness": 170}'
Zigbee2MQTT:debug 2021-02-19 21:11:14: Publishing 'set' 'state' to '0xec1bbdfffec3e6d1'
  zigbee-herdsman:controller:endpoint Command 0xec1bbdfffec3e6d1/1 manuSpecificTuya.setData({"status":0,"transid":0,"dp":1,"datatype":1,"length_hi":0,"length_lo
":1,"data":[1]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":
0,"manufacturerCode":null,"transactionSequenceNumber":null}) +36s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xec1bbdfffec3e6d1:2135/1 (0,0,1) +36s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":2135,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":10,"options":0,"ra
dius":30,"len":10,"data":{"type":"Buffer","data":[17,3,0,0,0,1,1,0,1,1]}} +36s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,20,36,1,87,8,1,1,0,239,10,0,30,10,17,3,0,0,0,1,1,0,1,1,141] +36s
Zigbee2MQTT:debug 2021-02-19 21:11:14: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffec3e6d1/set' with data '{"state": "ON", "brightness": 170}'
Zigbee2MQTT:debug 2021-02-19 21:11:14: Publishing 'set' 'state' to '0xec1bbdfffec3e6d1'
  zigbee-herdsman:controller:endpoint Command 0xec1bbdfffec3e6d1/1 manuSpecificTuya.setData({"status":0,"transid":1,"dp":1,"datatype":1,"length_hi":0,"length_lo
":1,"data":[1]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":
0,"manufacturerCode":null,"transactionSequenceNumber":null}) +22ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +36s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +4ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +36s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,10,204] +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,10,204] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,10] - 204 +4ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":10} +36s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xec1bbdfffec3e6d1:2135/1 (0,0,1) +64ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":2135,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":11,"options":0,"ra
dius":30,"len":10,"data":{"type":"Buffer","data":[17,4,0,0,1,1,1,0,1,1]}} +67ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,20,36,1,87,8,1,1,0,239,11,0,30,10,17,4,0,0,1,1,1,0,1,1,138] +69ms
Zigbee2MQTT:debug 2021-02-19 21:11:14: Publishing 'set' 'brightness' to '0xec1bbdfffec3e6d1'
  zigbee-herdsman:controller:endpoint Command 0xec1bbdfffec3e6d1/1 manuSpecificTuya.setData({"status":0,"transid":2,"dp":3,"datatype":2,"length_hi":0,"length_lo
":4,"data":[0,0,2,155]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reserv
edBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +67ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,87,8,0,221] +32ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,87,8,0,221] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [87,8,0] - 221 +3ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":2135,"relaycount":0,"relaylist":[]} +47ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3,68,128,0,1,11,205] +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3,68,128,0,1,11,205] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +5ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +96ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,11,205] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,11] - 205 +7ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":11} +43ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xec1bbdfffec3e6d1:2135/1 (0,0,1) +87ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":2135,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":12,"options":0,"ra
dius":30,"len":13,"data":{"type":"Buffer","data":[17,5,0,0,2,3,2,0,4,0,0,2,155]}} +85ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,1,87,8,1,1,0,239,12,0,30,13,17,5,0,0,2,3,2,0,4,0,0,2,155,23] +85ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,16,214] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,16,214] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,16] - 214 +3ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":16} +71ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +7ms
````

Koenkk, 2021-02-20:
Thanks, I've investigated but I don't see the problem yet. The produced bytes are exactly the same. Would you be also able to sniff the traffic when sending this command? 

When doing so make sure to set exactly the same brightness in each sniff and do it in the following sequence (to make sure variables like transaction id which is incremented by one with every command is the same).
- Start z2m and sniffing
- Send one time brightness 170
- Stop sniffing

https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

peca89, 2021-02-20:
OK, will do, but I will be away for a few days. I should have the capture ready by next weekend.

Thanks for your effort so far.

peca89, 2021-02-25:
While preparing for capture, I was manually sending the MQTT command to topic `zigbee2mqtt/0xec1bbdfffec3e6d1/set` with payload `{"brightness": 170}` and found out that this actually works with 1.17.1. So it's just a slider in HA GUI that does not work.

However, publishing payload `{"state": "ON", "brightness": 170}` does not work, and this appears to be the exact MQTT message HA is sending while moving the slider.

Can you make something out of this or would you still require packet captures?
What's the difference between "level" and "brightness". IKEA bulbs don't seem to have "level" attribute...

````
{"state": "ON", "brightness": 170}

Debug Received MQTT message on 'zigbee2mqtt/0xec1bbdfffec3e6d1/set' with data '{"state": "ON", "brightness": 170}'
Debug Publishing 'set' 'state' to '0xec1bbdfffec3e6d1'
Debug Publishing 'set' 'brightness' to '0xec1bbdfffec3e6d1'
Debug Received Zigbee message from '0xec1bbdfffec3e6d1', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":43}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffec3e6d1', payload '{"brightness":38,"level":157,"linkquality":70,"state":"ON"}'
Debug Received Zigbee message from '0xec1bbdfffec3e6d1', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":43}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffec3e6d1', payload '{"brightness":38,"level":157,"linkquality":70,"state":"ON"}'
Debug Received Zigbee message from '0xec1bbdfffec3e6d1', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":43}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffec3e6d1', payload '{"brightness":38,"level":157,"linkquality":70,"state":"ON"}'

{"brightness": 170}

Debug Received MQTT message on 'zigbee2mqtt/0xec1bbdfffec3e6d1/set' with data '{"brightness": 170}'
Debug Publishing 'set' 'brightness' to '0xec1bbdfffec3e6d1'
Debug Received Zigbee message from '0xec1bbdfffec3e6d1', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,2,155],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":45}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffec3e6d1', payload '{"brightness":169,"level":667,"linkquality":73,"state":"ON"}'
````

peca89, 2021-02-25:
OK, please find attached two packet captures using 1.16.1 and 1.17.1.  These are results of publishing `{"state": "ON", "brightness": 170}` to `zigbee2mqtt/0xec1bbdfffec3e6d1/set`. Detailed sequence was:

1. Downgrade to 1.16.1
2. Start z2m
3. Turn the light ON and set max brightness via slider to achieve consistent state
4. Restart z2m
5. Start capture
6. Fire the MQTT command from HA. Light dims.
5. Stop capture
6. Move the slider back to max from HA. Light goes back to max.
7. Stop z2m
8. Upgrade to 1.17.1
9. Start z2m
10. Start capture
11. Fire the MQTT command from HA. Light does nothing, stays on max.
12. Stop capture

[capture.zip](https://github.com/Koenkk/zigbee2mqtt/files/6044591/capture.zip)

Koenkk, 2021-02-25:
I did not look at the sniff yet. But given it work when setting only the `brightness` it looks like the it's something from z2m and not in the actual send data.

Above https://github.com/Koenkk/zigbee-herdsman-converters/blob/2395e4445a9beb605a823e4767fec6710d5658bd/lib/tuya.js#L396 can you try adding: `sendDataPoint.transId = 16;`

Koenkk, 2021-02-26:
Does it only work with `16` or also e.g. `sendDataPoint.transId = 1;`?

peca89, 2021-02-26:
It also works fine with setting that to 1, 0 or seemingly any other value.

Koenkk, 2021-02-27:
Integrated the fix now, please let me know if it works.

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

peca89, 2021-02-28:
It works now, thanks!! 👍

closed 2021-02-28 by peca89