Koenkk/zigbee2mqtt #6367: MAC transaction expired' (240) with Moes Tuya TS0601 radiator valve

877dev, 2021-02-19:
## What happened

This is possibly two separate issues.

1) I woke up and all my zigbee devices are offline, I fixed this by restarting the Docker container and power cycling the CC2531 co-ordinator. See attached log: (https://github.com/Koenkk/zigbee2mqtt/files/6009650/zigbee2mqtt_logs_19-02-2021.log)

2) Now all working, except sending mqtt command to the radiator valve, it does not respond, and in Portainer logs I get these errors in the log:
[tuya_error.log](https://github.com/Koenkk/zigbee2mqtt/files/6009669/tuya_error.log)

## What did you expect to happen
1) the cc2531 not to crash randomly.
2) the valve to operate as requested, and the valve to report back to mqtt.

## How to reproduce it (minimal and precise)
Not sure why this happened. My system backs up at 4am but this crash has never happened before. Maybe the cc2531 crashing has caused the radiator valve to need re-pairing?

## Debug info
Zigbee2MQTT version: Docker latest-dev 14.15.4
Adapter hardware: CC2531 coordinator and several CC2531 routers
Adapter firmware version: Coordinator FIRMWARE_CC2531_DEFAULT_20190608, Routers CC2531_router_2019_02 (router-cc2531-std.hex)

Koenkk, 2021-02-19:
CC2531 crashes are not uncommon, this is because the adapter is not very powerful. Switching to the source routing firmware might provide more stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing . Else a more powerful adapter is needed: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter

877dev, 2021-02-19:
@Koenkk thank you, i have added more devices recently so that's why it's crashing perhaps. 

I will flash source routing, so I also need to reflash my cc2531 routers?

And when I swap to zzh stick, can I still use all the cc2531 routers, or best to change them all to zzh?

Thank you

Koenkk, 2021-02-19:
> I will flash source routing, so I also need to reflash my cc2531 routers?

No repairing needed

> And when I swap to zzh stick, can I still use all the cc2531 routers, or best to change them all to zzh?

You can still use your cc2531 routers but you will have to repair all devices when switching to the zzh

https://www.zigbee2mqtt.io/information/FAQ.html#what-does-and-does-not-require-repairing-of-all-devices

877dev, 2021-02-20:
Thank you! 
For some reason I did have to repair all my devices, but it was quite a quick process.

All is working again now, I will be purchasing a zzh or similar as soon as stock is available.

closed 2021-02-20 by 877dev