Koenkk/zigbee2mqtt #6371: Failed to disable reporting for

maciekb, 2021-02-19:
## What happened
After removing the "report: true" option from the configuration file, the logs show the following errors:
https://pastebin.com/vzwVdhAH

## What did you expect to happen
Logs free of any errors and fast startup of frontend, now it takes about two minutes.

## How to reproduce it (minimal and precise)
Remove "report: true" option from config file, restart zigbee2mqtt

## Debug info
Zigbee2MQTT version: 1.17.1-dev (commit #a16635d)
Adapter hardware: Conbee II
Adapter firmware version: 0x26680700

Koenkk, 2021-02-19:
@chrishae what does error 136 mean?

```
Failed to disable reporting for '0x000d6ffffe5b9c19' - Error: Unbind 0x000d6ffffe5b9c19/1 genOnOff from '0x00212effff05dc37/1' failed (Error: status: 136)
```

ChrisHae, 2021-02-22:
ZDP status of 136 (0x88) means:
```
ZB_ZDP_STATUS_NO_ENTRY 	
The unbind request was unsuccessful due to the coordinator or source device not having an entry in its binding table to unbind
```

Koenkk, 2021-02-22:
@ChrisHae thanks!

@maciekb you can fix the error by editing database.db and removing `"reporting": 1` for this device.

closed 2021-02-22 by Koenkk

maciekb, 2021-02-22:
After removing "reporting":1 from database.db errors are gone and everything works as it should. Thanks!