Koenkk/zigbee2mqtt #6374: TuYa TS0201 unsupported device with CC2652RB

ced1442, 2021-02-19:
## What happened
I moved from CC2531 to CC2652RB and cannot pair my temperature sensor TuYa TS0201. It was working correctly on CC2531.

on the .db I got this now:
{"id":40,"type":"EndDevice","ieeeAddr":"0x5c0272fffecbed71","nwkAddr":5134,"manufId":4098,"powerSource":"Battery","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,1026,1029],"outClusterList":[25],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2740}},"genBasic":{"attributes":{"powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1}},"msRelativeHumidity":{"attributes":{"measuredValue":3410}}},"binds":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613749864962}

When working on CC2531 :
{"id":38,"type":"EndDevice","ieeeAddr":"0x5c0272fffecbed71","nwkAddr":14743,"manufId":4098,"manufName":"_TZ2000_a476raq2","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,1026,1029],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0201","manufacturerName":"_TZ2000_a476raq2","appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryVoltage":32,"batteryPercentageRemaining":200}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1950}},"msRelativeHumidity":{"attributes":{"measuredValue":4550}}},"binds":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","interviewCompleted":true,"meta":{},"lastSeen":1613720761826}

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2652RB
Adapter firmware version: I don't know how to get it as already flashed

Thanks for your help!

RubenKelevra, 2021-02-24:
`"batteryPercentageRemaining":200` does this mean 20% or 200%?

If 20%, the battery might just be dead. :)

ced1442, 2021-02-25:
Sorry, it's not so easy. The 200 come from the working setup and the battery is ok.

closed 2021-04-05 by github-actions[bot]