Koenkk/zigbee2mqtt #6375: Lidl Lidl HG06104A being detected as TS0505A

PiteousHonking, 2021-02-19:
## What happened
When pairing device it pairs but is not detected correctly
hardware is a zzh! Co-ordinatior attempting to pair With
https://www.zigbee2mqtt.io/devices/HG06338.html
Device is detected as (supported) device but then says not supported (see log below)

## What did you expect to happen
support device to pair and be recognised

## How to reproduce it (minimal and precise)
Pair device 
## Debug info
Zigbee2MQTT version:
Adapter hardware: Zzh!
Adapter firmware version:

log:
Zigbee2MQTT:info  2021-02-19 17:14:22: Starting interview of '0x588e81fffecf443a'
Zigbee2MQTT:info  2021-02-19 17:14:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x588e81fffecf443a"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-02-19 17:14:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x588e81fffecf443a"},"type":"pairing"}'
Zigbee2MQTT:info  2021-02-19 17:14:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x588e81fffecf443a"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-02-19 17:14:24: Successfully interviewed '0x588e81fffecf443a', device has successfully been paired
Zigbee2MQTT:warn  2021-02-19 17:14:24: Device '0x588e81fffecf443a' with Zigbee model 'TS0505A' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Koenkk, 2021-02-19:
Make sure you are on zigbee2mqtt 1.17.1 . If that doesn't help share the data/database.db entry

PiteousHonking, 2021-02-20:
Upgraded from 1.14 to 1.17 and now zigbee2mqtt is unable to start and I have lost access to all devices. Debug log attached. 

Zigbee2MQTT:debug 2021-02-20 09:05:09: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6746},"serialPort":{"path":"/dev/ttyUSB0"}}'
Zigbee2MQTT:error 2021-02-20 09:06:11: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-20 09:06:11: Failed to start zigbee
Zigbee2MQTT:error 2021-02-20 09:06:11: Exiting...
Zigbee2MQTT:error 2021-02-20 09:06:11: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.17.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-02-20T09_06_11_444Z-debug.log

PiteousHonking, 2021-02-20:
zzh fw is CC26X2R1_20200805 does it need to be newer as well?

PiteousHonking, 2021-02-20:
zzh updated to CC2652R_coordinator_20210120 firmware and issue #2893 encountered and resolved.

PiteousHonking, 2021-02-20:
Had to repair all devices but this is now resolved

closed 2021-02-20 by PiteousHonking