Koenkk/zigbee2mqtt #6377: Danalock battery drain

Varming73, 2021-02-19:
The last two days the Danalock was drained from around 53% battery to zero. I've read elsewhere that polling the device too much can lead to this. Is there any way to control that somehow?

closed 2021-02-19 by Varming73

MattWestb, 2021-02-20:
Its have pity large batteries so not very likely not problem with the normal pull control.
But its have one EFR32 Zigbee chip and it can being that the configuration of the long pull interval is not being OK and its cant going down to long pull and staying in short pull like Samsung devices can doing.
Trying configure the device or repairing it for getting it settedup OK.

Varming73, 2021-02-20:
Repairing it..?

MattWestb, 2021-02-20:
I dont knowing if reconfiguration is setting up the pull control OK or not i is sniffing and analyzing how its working in ZHA for the moment but its working little different way.
If you repairing (open the network for joining and resetting the device so it can joining the network) Z2M shall setting all OK but i dont knowing wot its default setting is.