Koenkk/zigbee2mqtt #6379: Please add support E-Life Zigbee door lock module

GD-Dal, 2021-02-20:
Hi
I've bought myself a new door lock, and with it came a zigbee module to control it.
Unfortunately, Zigbee2MQTT does not support it yet.
Z2MQTT recognizes the module as easyCodeTouch_v1, but the manual says it is E-Life Zigbee module from Onesti Products.
Manual to the module is attached. I'm hoping it contains all needed to support this module.
[E-life Zigbee Modul User Manual v2.0.pdf](https://github.com/Koenkk/zigbee2mqtt/files/6015013/E-life.Zigbee.Modul.User.Manual.v2.0.pdf)

Thank you very much

GD-Dal, 2021-02-22:
Hi.
I tried to add some preliminary code for this in devices.js, but I get an error.
And then i see this in the change log:
- Deprecated devices.js method for adding support for a new device
In what way can we try to add devices ourselves now?

The code I was trying to use was partly copied from another door lock:
`    {
        zigbeeModel: ['easyCodeTouch_v1'],
        model: 'easyCodeTouch_v1',
        description: 'EasyAcess EasyFingerTouch',
        vendor: 'Onesti Products AS',
        fromZigbee: [fz.ZNMS12LM_ZNMS13LM_closuresDoorLock_report, fz.ZNMS12LM_low_battery, fz.xiaomi_battery],
        toZigbee: [],
        exposes: [
            e.battery(), e.voltage(), e.battery_low(), exposes.binary('state', ea.STATE, 'UNLOCK', 'LOCK'),
            exposes.binary('reverse', ea.STATE, 'UNLOCK', 'LOCK'),
            exposes.enum('action', ea.STATE, [
                'finger_not_match', 'password_not_match', 'reverse_lock', 'reverse_lock_cancel', 'locked', 'lock_opened',
                'finger_add', 'finger_delete', 'password_add', 'password_delete', 'lock_opened_inside', 'lock_opened_outside',
                'ring_bell', 'change_language_to', 'finger_open', 'password_open', 'door_closed',
            ]),
        ],
        meta: {configureKey: 1, battery: {voltageToPercentage: '4LR6AA1_5v'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            // Device advertises itself as Router but is an EndDevice
            device.type = 'EndDevice';
            device.save();
        },
    },`

The cluster looks like this:
`0xf4ce367919d08145
Endpoint 11
 - Output clusters
 -- genOta
 - Input clusters
 -- genBasic
 -- genPowerCfg
-- genIdentify
-- genScenes
-- genGroups
-- closuresDoorLock`

I wish there was a web gui or something to add new devices in an easier way.
It's not that I don't want to do it myself, I just don't have the faintest clue how.

GD-Dal, 2021-03-25:
Hi
Any news on this?
What can I do to help?

Thanks

Koenkk, 2021-03-26:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/6551

closed 2021-03-26 by Koenkk