Koenkk/zigbee2mqtt #6380: Document Tradfri-Conbee pairing: multiple lights at a time don't work

canton7, 2021-02-20:
I've just moved from a CC2531 to a Conbee II, and went through the process of pairing all of my Tradfri lights again.

With the CC2531 I was able to reset multiple lights at a time, and Z2M would discover them all. With the Conbee II, this didn't work, and there was no indication in any log (Z2M or zigbee-herdsman) that anything was happening. However when I unplugged the lights so that only one was reset at a time, everything worked.

I suspect this is a Combee II issue and nothing to do with Z2M, but I imagine I probably won't be the only one caught out by this. Would it be worth adding a note to the docs (either on the docs for the Tradfri lights, or on the adapter docs) giving a hint for others?

pavax, 2021-02-23:
I was able to pair multiple ikea devices using the conbee-II at one time (3-4 devices I´d say) - but can't compare it to a different zigbee adapter. 

I do remember that several devices got recognized at the same time: But I needed a couple of reset-cycle retries (6 times on/off) to recognize every light (got approx.: 16). Also, one spotlight refused to get recognized: That is when I also unplugged the others to get that one recognized after the next reset-cycle.

Lights that I integrated:
- 7xLED1537R6/LED1739R5
- 3xLED1837R5
- 2xL1527 
- 1xL1528
- 1xL1529
- 2xLED1732G11

Zigbee2MQTT version: 1.17.1
Adapter hardware: Conbee II
Integration: Home Assistant Addon
Adapter firmware version: 0x26680700

canton7, 2021-02-24:
Interesting. I tried it both with LED1649C5's and LED1837R5's. The difference perhaps was that in both cases, all of the lights being reset were plugged into the same fixture, so they were all reset at exactly the same instant.

With the LED1837R5's, I tried resetting 3 at a time, and also 2 at a time, and Z2M only appeared to notice when I went down to 1 at a time. For the LED1649C5's I gave up after resetting 3 at a time didn't work.

Tried with both the stock adapter firmware version, and with 0x26680700.

MattWestb, 2021-03-28:
CC-253X is normally running on HA1.2 and its have not implanting the Zigbee 3 security.
One Zigbee network (or more correct the underlying IEEE 801.15.4) is limiting to 8 broadcasts in 9 seconds and much of the Zigbee 3 paring is done with broadcast.

So the deCONZ is one Zigbee 3 coordinator and making it right.

CC-253X is something else that is doing much things little out of standard and dont care of standard things and its firmware is very well patched for doing that.

So is you is paring 15 devices direct to one CC-253X its likely its possible to do it OK but if the pairing is going thru routers its hitting the 8/9 seconds and you is blocking the network more or less.

closed 2021-05-06 by github-actions[bot]