Koenkk/zigbee2mqtt #6383: Lidl HG06106B color red no light

luca-iodice, 2021-02-21:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

luca-iodice, 2021-02-21:
When I set the light of Lidl HG06106B to red 255 the lamp doesn't turn on.
Sometimes trying to change the color to reddish color doesn't do anything ( I have to trigger multiple times)

Chekedaudau, 2021-03-14:
Hey, I fixed it temporary with the applyredfix: https://github.com/Koenkk/zigbee2mqtt/issues/5011
But thats not a final solution.

Koenkk, 2021-03-14:
@Chekedaudau `applyRedFix` is enabled by default in the latest zigbee2mqtt

closed 2021-04-14 by Koenkk