Koenkk/zigbee2mqtt #6386: TS0503B color calibration problem

FaBRiK74, 2021-02-21:
## What happened
The colors are not correct. E.g. total red contains even blue and green. Total Blue contains a very little red.
can't set yellow at all.
When I try to set colors manually with the RGB there are some interesting issues: Below a given value the related color switches off, then sets back the value to 0 or 255: R below 43, G below 58 and B below 75 switches off. Thet means the following RGB codes gives totally switch off: 42,0,0; 0,57,0; 0,0,74
When I set 255,0,0 it goes back 255,0,17

The module and the LED strip itself have no any problem. Withouh Z2M (with its button) I can set full Red, Green and Blue as well.

Zigbee2MQTT version: Z2M Edge (reinstalled for the latest)
Adapter hardware: CC2531 (Sonoff)

FaBRiK74, 2021-03-02:
Is there any solution for this issue?

FaBRiK74, 2021-04-02:
The issue still exists.

FaBRiK74, 2021-05-04:
The problem still exists

FaBRiK74, 2021-06-04:
The issue still exists

FaBRiK74, 2021-07-08:
Unfortunately still exists.

closed 2021-08-16 by github-actions[bot]

pawelbudynek, 2021-10-06:
Maybe the hue_correction parameter could fix that? I have similar problem but I am not sure how to use this parameter from documentation...