Koenkk/zigbee2mqtt #6387: Awox based LED ceiling lamp recognized as E27 AwoX bulb

bouyssic, 2021-02-21:
Hello team!

I'm hoping you can help me. I'm based in France and I just purchased 2 ceiling lights that seem to be based on AwoX controllers. I use zigbee2mqtt together with Home Assistant (both running as docker and rocking latest versions).

I'm having an issue with this ceiling lamp, which has a led strip inside, that zigbee2mqtt is recognizing as a E27 Awox bulb. All in all not too much of a deal, since I can still switch it on/off and control brightness, but **I lose temp control**.

I've gone through [how to support new devices page](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html) but here it's a bit different seems the identification is "just wrong". I don't understand how to prevent the "false identification" or if there is anything we can do to prevent it.

Can someone walk me through how to "correct the identification" and add temperature control to this device? 

I'm fairly technical literate and I can help if you can walk me through the next steps

## Logs when pairing
`Zigbee2MQTT:info  2021-02-21 10:31:37: Starting interview of '0xa4c138ccc2e45fef'
Zigbee2MQTT:info  2021-02-21 10:31:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ccc2e45fef","ieee_address":"0xa4c138ccc2e45fef","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-02-21 10:31:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138ccc2e45fef"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-02-21 10:31:39: Received Zigbee message from '0xa4c138ccc2e45fef', type 'readResponse', cluster 'genBasic', data '{"modelId":"TLSR82xx"}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-02-21 10:31:39: Received Zigbee message from '0xa4c138ccc2e45fef', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"AwoX"}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-02-21 10:31:39: Received Zigbee message from '0xa4c138ccc2e45fef', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-02-21 10:31:40: Received Zigbee message from '0xa4c138ccc2e45fef', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-02-21 10:31:40: Received Zigbee message from '0xa4c138ccc2e45fef', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-02-21 10:31:41: Received Zigbee message from '0xa4c138ccc2e45fef', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-02-21 10:31:41: Received Zigbee message from '0xa4c138ccc2e45fef', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-02-21 10:31:41: Received Zigbee message from '0xa4c138ccc2e45fef', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-02-21 10:31:42: Received Zigbee message from '0xa4c138ccc2e45fef', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"0122052017"}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-21 10:31:42: Successfully interviewed '0xa4c138ccc2e45fef', device has successfully been paired
Zigbee2MQTT:info  2021-02-21 10:31:42: Device '0xa4c138ccc2e45fef' is supported, identified as: AwoX LED E27 white (33951)
`

## Information about the device + link
..
I can't seem to find it in their Eglos or AwoX page :man_facepalming: 

This is the reseller page, not sure it will help a lot : https://www.castorama.fr/plafonnier-connecte-led-integree-2250lm-ip20-23w-40cm-blanc/9002759339579_CAFR.prd

I'm available if you need any additional information

## data/database.db entry of the device
`{21-02-21 10:48}neo:~/docker_containers/home_assistant chris% cat zigbe2mqtt/database.db| grep 0xa4c138ccc2e45fef
{"id":23,"type":"Router","ieeeAddr":"0xa4c138ccc2e45fef","nwkAddr":10939,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613899902239}
`
Looking forward to your help

bouyssic, 2021-02-21:
I just noticed that I ran Z2A debug logs only. I'll enable Zigbee-herdsman debug logging and report back

Koenkk, 2021-02-21:
Does it work after changing the entry in devices.js to:

```js
{
    zigbeeModel: ['TLSR82xx'],
    model: '33951',
    vendor: 'AwoX',
    description: 'LED E27 white',
    extend: preset.light_onoff_brightness_colortemp(),
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

bouyssic, 2021-02-21:
Hi @Koenkk ,

Thanks for the ultra-fast feedback. I'll give it a try and I'll let you know.

I'm adding herdsamn debugg to this in case it's needed later:
`zigbee2mqtt      | 2021-02-21T10:20:14.686Z zigbee-herdsman:deconz:adapter RECEIVING NODE_DESCRIPTOR FAILED - addr: 0x2b85 waiting for response TIMEOUT
zigbee2mqtt      | 2021-02-21T10:20:14.686Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0xa4c138ccc2e45fef', attempt 5
zigbee2mqtt      | 2021-02-21T10:20:14.734Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x48cb EP:0 SeqNr. 103 request id: 13
zigbee2mqtt      | 2021-02-21T10:20:14.737Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 13 status: 0
zigbee2mqtt      | 2021-02-21T10:20:14.737Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2021-02-21T10:20:15.637Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 104
zigbee2mqtt      | 2021-02-21T10:20:15.639Z zigbee-herdsman:deconz:frameParser device state: 10100010
zigbee2mqtt      | 2021-02-21T10:20:15.640Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2021-02-21T10:20:18.128Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
zigbee2mqtt      | 2021-02-21T10:20:18.128Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2021-02-21T10:20:18.147Z zigbee-herdsman:deconz:driver query aps data indication
zigbee2mqtt      | 2021-02-21T10:20:18.151Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 105
zigbee2mqtt      | 2021-02-21T10:20:18.155Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 105 srcAddr: 0x49f7 destAddr: 0xfffd profile id: 0x0 cluster id: 0x13
zigbee2mqtt      | 2021-02-21T10:20:18.156Z zigbee-herdsman:deconz:frameParser response payload: 0,247,73,239,95,228,194,204,56,193,164,142
zigbee2mqtt      | 2021-02-21T10:20:18.156Z zigbee-herdsman:controller:log Device '0xa4c138ccc2e45fef' joined
zigbee2mqtt      | 2021-02-21T10:20:18.156Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2021-02-21T10:20:18.156Z zigbee-herdsman:controller:log Device '0xa4c138ccc2e45fef' accepted by handler
zigbee2mqtt      | 2021-02-21T10:20:18.157Z zigbee-herdsman:controller:log New device '0xa4c138ccc2e45fef' joined
zigbee2mqtt      | 2021-02-21T10:20:18.157Z zigbee-herdsman:controller:log Creating device '0xa4c138ccc2e45fef'
zigbee2mqtt      | Zigbee2MQTT:info  2021-02-21 11:20:18: Device '0xa4c138ccc2e45fef' joined
zigbee2mqtt      | Zigbee2MQTT:info  2021-02-21 11:20:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ccc2e45fef","ieee_address":"0xa4c138ccc2e45fef"},"type":"device_joined"}'
zigbee2mqtt      | 2021-02-21T10:20:18.164Z zigbee-herdsman:controller:log Interview '0xa4c138ccc2e45fef' start
zigbee2mqtt      | Zigbee2MQTT:info  2021-02-21 11:20:18: Starting interview of '0xa4c138ccc2e45fef'
zigbee2mqtt      | Zigbee2MQTT:info  2021-02-21 11:20:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ccc2e45fef","ieee_address":"0xa4c138ccc2e45fef","status":"started"},"type":"device_interview"}'
zigbee2mqtt      | 2021-02-21T10:20:18.165Z zigbee-herdsman:controller:device:log Interview - start device '0xa4c138ccc2e45fef'
zigbee2mqtt      | Zigbee2MQTT:info  2021-02-21 11:20:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138ccc2e45fef"},"type":"device_connected"}'
zigbee2mqtt      | Zigbee2MQTT:info  2021-02-21 11:20:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138ccc2e45fef"},"type":"pairing"}'
zigbee2mqtt      | 2021-02-21T10:20:18.250Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x49f7 EP:0 SeqNr. 106 request id: 14
zigbee2mqtt      | 2021-02-21T10:20:18.253Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 14 status: 0
zigbee2mqtt      | 2021-02-21T10:20:18.254Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2021-02-21T10:20:18.261Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
zigbee2mqtt      | 2021-02-21T10:20:18.262Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2021-02-21T10:20:18.276Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110
zigbee2mqtt      | 2021-02-21T10:20:18.276Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2021-02-21T10:20:18.349Z zigbee-herdsman:deconz:driver query aps data confirm
zigbee2mqtt      | 2021-02-21T10:20:18.351Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 107
zigbee2mqtt      | 2021-02-21T10:20:18.353Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x49f7 request id: 14 confirm status: 0
[... cut ...]
zigbee2mqtt      | 2021-02-21T10:20:24.888Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2021-02-21T10:20:25.585Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 118
zigbee2mqtt      | 2021-02-21T10:20:25.589Z zigbee-herdsman:deconz:frameParser device state: 10100010
zigbee2mqtt      | 2021-02-21T10:20:25.589Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
`

bouyssic, 2021-02-21:
@Koenkk it worked like a charm ! 

Several questions if I may : 

1. Do you need any output or should I do something else now to add this (a PR)?
2. What is the process to "go back" to default conatiner (i.e. removing the volume I had to mount to modify the device.js). I guess I have to wait for a new release and just remove the volume see it it keeps working?

I appreciate your help so far

Koenkk, 2021-02-21:
1. Added it now.
2. You remove your changes and switch to the latest dev or wait for the next release on 1 march

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-21 by Koenkk

piotrwl, 2021-04-02:
Can you add support for Awox 33946 model?