Koenkk/zigbee2mqtt #6390: Awox 33948 E27 bulb support

tdaime, 2021-02-21:
The awox 33948 (https://www.castorama.fr/ampoule-led-connectee-e27-st64-500lm-41w-blanc-chaud-awox/9002759339487_CAFR.prd )  is actually paired in my Home Assistant / Zigbee2mqtt configuration as 33951 (https://www.zigbee2mqtt.io/devices/33951.html#awox-33951) device,  which is OK as both devices seem to expose the same properties. 
Just submitting this issue to update documentation as/if needed. 
BTW, I didn't find any Awox entry in devices.js: 

## Information about the device + link
..

## data/database.db entry of the device
..

Koenkk, 2021-02-21:
Changed the model to `33951/33948`

closed 2021-02-21 by Koenkk

Omega-Phone, 2021-03-14:
Hi @tdaime, sorry to re-open this issue but I bought the same hardware (awox 33948) and cannot figure out how to pair it with my Home Assistant / Zigbee2mqtt installation. Maybe you could tell me the steps you took ? 
Thank's in advance !