Koenkk/zigbee2mqtt #6391: support sensor-gas GASSensor-EF-3.0 m414-5E

maverik80, 2021-02-21:
hello, unfortunately this device is detected as unsupported and I tried to follow the guide for the support of new products but I did not succeed, I kindly ask if it is possible to support this product, I leave product links to facilitate recognition

https://it.aliexpress.com/item/32840620167.html?spm=a2g0s.9042311.0.0.395e4c4dBt1RC8

thank you very much

sjorge, 2021-02-21:
You can try with adding this to devices.js

```javascript
    {
        zigbeeModel: ['GASSensor-EF-3.0'],
        model: 'HS1CG-E',
        vendor: 'HEIMAN',
        description: 'Combustible gas sensor',
        fromZigbee: [fz.ias_gas_alarm_1],
        toZigbee: [],
        exposes: [e.gas(), e.battery_low(), e.tamper()],
    },
```

maverik80, 2021-02-21:
ok soon I'll try this and let you know

maverik80, 2021-02-21:
sorry if I stress you but I have recently approached this world even I know I already have 20napparecchiature connected, where can I find the file device.js in raspbian in docker?

sjorge, 2021-02-21:
I'm not 100% as I don't use docker, but I'd assume a `find /path/to/docker/root -name devices.js` should find it.

maverik80, 2021-02-21:
I try to see if I find it because I have no idea where to put the code you passed me

sjorge, 2021-02-21:
Once you find the file, just add it somewhere along the Heiman section. Order shouldn’t  matter.

maverik80, 2021-02-21:
Unfortunately I'm trying to look for it but I don't know where it is, I hope to be able to find it

maverik80, 2021-02-21:
can the device.js file be the device.yaml file? but inside I see only the friendly names, then I have the addon's configuration.yaml file of z2m

sjorge, 2021-02-21:
No devices.yaml is not the correct file

maverik80, 2021-02-21:
I guessed it wasn't the right one but I really can't find this file, I start to get nervous 😂

sjorge, 2021-02-21:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#2-adding-your-device

Maybe this is helpful

maverik80, 2021-02-21:
I am following the guide but I stop when I need to add volumes to the container, I enter" docker run -v /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters" and I get this

root@raspberrypi:/opt/zigbee-herdsman-converters# docker run -v /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters
"docker run" requires at least 1 argument.
See 'docker run --help'.

Usage:  docker run [OPTIONS] IMAGE [COMMAND] [ARG...]

Run a command in a new container
-------------------------
i don't really understand how to enter this command and add volumes to container

maverik80, 2021-02-22:
after all my trials to add volume to the container i always get this, i don't know what to do,
[image]

sjorge, 2021-02-22:
Hopefully someone with docker experience comes along, as I have no idea on the error.

maverik80, 2021-02-22:
hopefully, I'm going crazy

BFFonseca, 2021-03-22:
> You can try with adding this to devices.js
> 
> ```js
>     {
>         zigbeeModel: ['GASSensor-EF-3.0'],
>         model: 'HS1CG-E',
>         vendor: 'HEIMAN',
>         description: 'Combustible gas sensor',
>         fromZigbee: [fz.ias_gas_alarm_1],
>         toZigbee: [],
>         exposes: [e.gas(), e.battery_low(), e.tamper()],
>     },
> ```

Would this work on the HomeAssistant Z2M addon?
Do you know where I can find the 'devices.js' file?

BFFonseca, 2021-03-22:
Was able to make it work on the Z2M HA addon.
Created the file '/config/zigbee2mqtt/devices.js' and added 
```
"external_converters": [
    "devices.js"
  ]
```

to the Z2M 'config.yaml' file.

The "devices.js" file contains:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const device = {
    zigbeeModel: ['GASSensor-EF-3.0'],
    model: 'HS1CG-E',
    vendor: 'HEIMAN',
    description: 'Combustible gas sensor',
    fromZigbee: [fz.ias_gas_alarm_2],
    toZigbee: [],
    exposes: [e.gas(), e.battery_low(), e.tamper()],
};
module.exports = device;
```
With `fz.ias_gas_alarm_1` it wasn't reporting the status when I pressed the "test" button on the device. Changing this to `fz.ias_gas_alarm_2` made it work

maverik80, 2021-03-22:
the problem is that I work in supervised and I am unable to mount it

Koenkk, 2021-03-22:
@BFFonseca added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-03-22 by Koenkk

maverik80, 2021-03-22:
i entered manually as you advised me and it worked, thanks a lot BBFonseca