Koenkk/zigbee2mqtt #6398: Devices not recognized with Conbee 1

M-Gregoire, 2021-02-21:
## What happened

I'm currently migrating from Deconz to Zigbee2MQTT and I'm using a Conbee 1 (`  adapter: deconz` specified in config).

Only Conbee 2 is written as a supported adapter but I've read on some issues of this repo that the API with Conbee 1 is the same as Conbee 2, and, as such, should work the same with a Conbee II. I'm sorry if this not the case (and If it actually is, might it be worth it to add it to the supported adapters page?)

Both Deconz and Zigbee2MQTT run in Hassio.

I've first removed my devices from Deconz, stopped the container, started Zigbee2MQTT and did a reset on the Zigbee devices of my network.

My device appears is the Zigbee2MQTT web ui (which leads me to believe Conbee is indeed supported) with the following line in the log:
```
 Failed to interview '0x00158d0002c9e846', device has not successfully been paired
```

The friendly name is the same as the IEE adress and the manufacturer is written as `Unsupported`, On the device picture, a warning sign says "Interview failed". 

However, my device is supported: it's as [Xiaomi WXKG03LM_rev2](https://www.zigbee2mqtt.io/devices/WXKG03LM_rev2.html#xiaomi-wxkg03lm_rev2) and it was working properly in Deconz, excluding a hardware issue.

I tried following all the steps described in the [FAQ
](https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails) without success.

Here is an extract from the DEBUG log: https://pastebin.com/NtTkffVp

Thank you for any help you might provide

## Debug info
Zigbee2MQTT version: 1.17.1-4
Adapter hardware: Conbee I

M-Gregoire, 2021-02-26:
Update: After doing a firmware update (using Ubuntu) on my Conbee, devices are now recognized and working properly!
This confirms Conbee 1 is supported by Zigbee2MQTT. Would you consider adding it to the documentation ? I can submit a PR.

Koenkk, 2021-02-26:
Please submit a PR. Assuming this can be closed.

closed 2021-02-26 by Koenkk