Koenkk/zigbee2mqtt #6402: Osram colors broken again due to preferred xy mode in homeassistant.js

jasperro, 2021-02-21:
## What happened
Commit acfd938e48f693c5040935e765fc700c9a049259 made it so that xy color is always preferred in home assistant. This unfortunately breaks previous fixes for color rendering that forced on hsv mode. On AC03647 and AC03645 colors are again incorrectly set, because xy color on the device is broken on a firmware level. It seems that https://github.com/Koenkk/zigbee-herdsman-converters/issues/2129 has something to do with this issue too, is that the reason for the change?

## What did you expect to happen
HSV color advertised to home assistant as the primary way to set colors on devices AC03647, AC03645 and other devices with broken XY support.

## How to reproduce it (minimal and precise)
Change colors in home assistant of a affected device, see only the xy values change, and wrong color bulb.

## Possible fix
Add a extra flag in the code to check for preferred mode.
Instead of
```
xy: supportsXY,
hs: !supportsXY && supportsHS
```
something like
```
xy: supportsXY || !prefersHS
hs: (prefersHS) || (!supportsXY && supportsHS)
```
?

## Debug info
Zigbee2MQTT version: 1.17.1-dev
Adapter hardware: CC2531
Adapter firmware version: not relevant

sjorge, 2021-02-21:
I guess a way to indicate which bulbs should use hs over xy would be a good start.

osram and hue definitely look way better with hs (with the gamma correct applied),

innr looks slightly different between hs and xy, both look objectively the same, as they don’t do enhancedHue they look the most consistent with XY between normal and scenes (they need enhancedHue for using hs in scenes)

ikea only supports XY

tint again does not do enhancedHue but does look a bit better with hs, but when using scenes with fallback to xy and looks about the same as hs outside of scenes... xy outside of scenes looks really bad. Which makes me wonder if it internally applies some sort of correct to the xy values when using scenes. These are by far the worst.

Hue and Osram being the better looking ones if using hs.

Koenkk, 2021-02-22:
Should be fixed now.

The colors of the bulbs are indeed very off, these are in fact the only two devices which were forced to hs instead of xy (https://github.com/Koenkk/zigbee2mqtt/blob/1bccc5d9f17be93c1f680f8ed2d37cbbc0aef0fd/lib/extension/homeassistant.js#L1395)

jasperro, 2021-02-22:
It works better than ever before with the color being set correctly and everything being reported back in home assistant. Thanks!

Koenkk, 2021-02-23:
@jasperro this can be solved after https://github.com/home-assistant/architecture/discussions/519 . For now stop z2m, edit `data/state.json` and remove the xy color for this device.

closed 2021-02-23 by Koenkk

salopette, 2021-02-27:
Hello, there has been something new in the frontend since the update, but it has no function.
It is there for three different types of lamps.

-Gardenpole Mini RGBW OSRAM
-PAR16 RGBW Z3
-Outdoor Lantern W RGBW OSRAM

Debug Received Zigbee message from 'BWM Terrasse', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":24672}' from endpoint 2 with groupID 0
Debug Received MQTT message on 'zigbee2mqtt/Terrassen Licht/get' with data '{"color_options":""}'
Debug Received MQTT message on 'zigbee2mqtt/Terrassen Licht/get' with data '{"color_options":{"execute_if_off":""}}'
Debug Received MQTT message on 'zigbee2mqtt/Terrassen Licht/set' with data '{"color_options":{"execute_if_off":true}}'
Debug Received Zigbee message from 'Terrassen Licht', type 'readResponse', cluster 'lightingColorCtrl', data '{}' from endpoint 3 with groupID 0
Error Publish 'get' 'color_options' to 'Terrassen Licht' failed: 'Error: Read 0x7cb03eaa00b13545/3 lightingColorCtrl(["options"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'color_options' to 'Terrassen Licht' failed: 'Error: Read 0x7cb03eaa00b13545/3 lightingColorCtrl([\"options\"], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"Terrassen Licht"},"type":"zigbee_publish_error"}'
Info MQTT publish: topic 'zigbee2mqtt/Terrassen Licht', payload '{"brightness":254,"color_options":{"execute_if_off":false},"last_seen":"2021-02-27T08:12:33.963Z","linkquality":105,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'color_options' to 'Terrassen Licht' failed: 'Error: Read 0x7cb03eaa00b13545/3 lightingColorCtrl([\"options\"], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"Terrassen Licht"},"type":"zigbee_publish_error"}'

[image]
[image]
[image]

sjorge, 2021-02-27:
@koenkk, maybe like with level_options and power_on_behavior we should not expose this? (As originally propose) Support seems to vary between different models/brands.

~ sjorge

Koenkk, 2021-02-27:
@sjorge agree, could you make a PR?

sjorge, 2021-02-27:
> @sjorge agree, could you make a PR?

I'll make one tonight.