Koenkk/zigbee2mqtt #6403: Ikea 30W LED driver not working after firmware update

nielsi, 2021-02-21:
## What happened
Together with some other lights I OTA updated to the February 9th release of Ikea firmware. All updates where successful, however, both my 30W LED drivers now don't work (don't produce light) anymore. Communication with Zigbee over zigbee2mqtt still works. Brightness status even updates after changing it with an MQTT command, however, with any value there is no light.

## What did you expect to happen
To update and the lights keep functioning. 

Since there seem to be other rare complaints from people who don't use Zigbee2MQTT of their 30W drivers not working correctly anymore, it could be a firmware issue. So a roll back or manual flashing of an old binary file could be a solution but I don't know if this is possible with Zigbee2MQTT and I don't know if previous firmware versions are still possible to find. Or am I missing something? Right now, the only solution I can think of to light my kitchen again is to buy two new 30W Ikea drivers and hope they come out of stock with the old (working) firmware because the update effectively bricked them for me until Ikea might come with a next version.

## Debug info
Zigbee2MQTT version: latest
Adapter hardware: TI LAUNCHXL-CC26X2R1

diplix, 2021-02-21:
ikea activated a built-in voltage sensor and the drivers only work with a proper 24 volt supply. there’s a discussion about this on [reddit](https://www.reddit.com/r/tradfri/comments/lgo4yq/heads_up_update_to_led_driver_breaks_12v/) with a solution, too. you can try to change one of the resistors on the board. if you use an unmodified driver, may be the power supply is not working properly.

nielsi, 2021-02-22:
You're absolutely correct. I was running it with 12V. I've now replaced the resistor to 120k ohm and everything is working again. Thank you so much for sharing that information.

I had already left a feature request for a rollback function of firmware in Zigbee2MQTT that always saves the binaries of the last update so you could flash a previously flashed version back (would only work after the seconds update but possibly better than nothing). I'll leave that feature request where it is.

Since this is solved I consider the issue closed!

closed 2021-02-22 by nielsi