Koenkk/zigbee2mqtt #6408: No white channel on RGBW controller 'TS0504B'

rmickael62, 2021-02-22:
I just bought the same model of controller "TS0504B" but I do not understand how to control the white channel? 
[image]

_Originally posted by @rmickael62 in https://github.com/Koenkk/zigbee2mqtt/issues/6299#issuecomment-783249778_

rmickael62, 2021-02-23:
devices.js
{
        fingerprint: [{modelID: 'TS0504B', manufacturerName: '_TZ3000_ukuvyhaa'}],
        model: 'TS0504B',
        vendor: 'TuYa',
        description: 'Zigbee smart mini led strip controller 5V/12V/24V RGBW',
        extend: preset.light_onoff_brightness_colorxy(), <=  **_preset.light_onoff_brightness_colortemp_colorxy() ?_**
        meta: {applyRedFix: true},
    },

Koenkk, 2021-02-23:
The Zigbee ZCL spec does not define anything like a `white_value`. Did you try with `preset.light_onoff_brightness_colortemp_colorxy() ` does it work then? 

Can you also share the data/database.db entry of this device?

rmickael62, 2021-02-23:
preset.light_onoff_brightness_colortemp_colorxy() => does not work.

database.db :
{"id":52,"type":"Router","ieeeAddr":"0x60a423fffee28dac","nwkAddr":60195,"manufId":4098,"manufName":"_TZ3000_ukuvyhaa","powerSource":"Mains (single phase)","modelId":"TS0504B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0504B","manufacturerName":"_TZ3000_ukuvyhaa","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1614102988277}

Koenkk, 2021-02-23:
Given that is only has 1 endpoint. It seems the white channel is controlled in some non standard way. You will need to sniff the traffic with the original gateway while controlling the white channel and post it here: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

rmickael62, 2021-02-23:
I don't have an original gateway, is it possible with a sonoff zigbee bridge? knowing that the device is compatible with tuya and alexa.

Koenkk, 2021-02-24:
I don't know if you can set it with a sonoff zigbee bridge, you can try.

jti989, 2021-03-09:
Hi @rmickael62,
I have the same problem. Have you been able to solve it?

rmickael62, 2021-03-09:
Hi
I think that the white channel cannot be controlled directly in zigbee (obligation to use the button) I would use this module only for RGB.

bluefish007, 2021-03-16:
Hopefully a solution can be found. Unfortunately, I can not control white either.

jti989, 2021-03-23:
I spent some time with the controller and recognised that there are zigbee messages that may be useful for implementing this.
When I first changed the color mode to white (coming from RGB) with the hardware button, there was this message:

`Zigbee2MQTT:debug 2021-03-23 12:30:26: Received Zigbee message from '0x60a423fffe94d1d8', type 'attributeReport', cluster 'lightingColorCtrl', data '{"tuyaRgbMode":2}' from endpoint 1 with groupID 0`

When changing colors afterwards there are messages like this:
`Zigbee2MQTT:debug 2021-03-23 12:40:43: Received Zigbee message from '0x60a423fffe94d1d8', type 'raw', cluster 'lightingColorCtrl', data '{"data":[24,15,10,3,240,72,10,2,0,14,13,0,15,0,0,232,232],"type":"Buffer"}' from endpoint 1 with groupID 0`

And when switching to white again the message is longer:
`Zigbee2MQTT:debug 2021-03-23 12:40:51: Received Zigbee message from '0x60a423fffe94d1d8', type 'raw', cluster 'lightingColorCtrl', data '{"data":[24,17,10,3,240,72,18,4,3,70,70,120,240,232,232,0,0,70,70,120,192,232,10,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0`

I don't know how to decode these messages, but I have the feeling that the information we need is in it.
Could we reopen this topic @Koenkk  and @rmickael62 ?

mramirezd, 2021-03-31:
Me too! Thank you

bluefish007, 2021-05-05:
Problem still exists :(

bluefish007, 2021-06-05:
Unfortunately it still does not work!

bluefish007, 2021-07-07:
Still not working!

MattiaFochesato, 2021-07-17:
I managed to get it to work with this code:

``` 
const definition = {
        fingerprint: [{modelID: 'TS0504B', manufacturerName: '_TZ3000_ukuvyhaa'}],
        model: 'TS0504B',
        vendor: 'TuYa',
        description: 'Zigbee RGBW light',
        toZigbee: [tz.on_off, tz.tuya_led_control],
        fromZigbee: [fz.on_off, fz.tuya_led_controller, fz.brightness, fz.ignore_basic_report],
        exposes: [e.light_brightness_colortemp_colorhs([498, 500]).removeFeature('color_temp_startup')],
        meta: {applyRedFix: true, enhancedHue: false},
};
```

Just set any value as color temperature and the controller deactivates the RGB output and activates the W output with the respective brightness set.

It's not the best solution but at least it works. Better than nothing

bluefish007, 2021-07-17:
@MattiaFochesato 

Thanks for this idea! But how to go back to RGB? I do not come back when I enter a color...

And I can't control brightness after this change...

fastplayer95, 2021-07-20:
Hey there @MattiaFochesato,

first of all, thanks for your solution.
For me it works!

However i do get Errors:

'''
Zigbee2MQTT:error 2021-07-20 10:05:27: Publish 'set' 'color_temp' to 'led_strip' failed: 'TypeError: Cannot read property 'hasOwnProperty' of undefined'
Zigbee2MQTT:error 2021-07-20 10:07:38: No converter available for 'color_temp' (500)
Zigbee2MQTT:error 2021-07-20 10:09:44: No converter available for 'color_temp' (499)
Zigbee2MQTT:error 2021-07-20 10:09:46: No converter available for 'color_temp' (498)
'''

Omirax, 2021-07-30:
Better than nothing, but we need full integration.

untitledH, 2021-08-13:
> const definition = {
>         fingerprint: [{modelID: 'TS0504B', manufacturerName: '_TZ3000_ukuvyhaa'}],
>         model: 'TS0504B',
>         vendor: 'TuYa',
>         description: 'Zigbee RGBW [...]

This works within zigbee2mqtt but not with hass, something is wrong when it convert the colorxy value to a colorhs value.

rodrigoscoelho, 2021-08-14:
Same problem here

rodrigoscoelho, 2021-09-06:
> I managed to get it to work with this code:
>
> ```
> const definition = {
>         fingerprint: [{modelID: 'TS0504B', manufacturerName: '_TZ3000_ukuvyhaa'}],
>         model: 'TS0504B',
>         vendor: [...]

Did you get it working with HA? On mine, I don't have option for Color temperature on HA.

Thanks.

deeja, 2021-09-22:
> I managed to get it to work with this code:
>
> ```
> const definition = {
>         fingerprint: [{modelID: 'TS0504B', manufacturerName: '_TZ3000_ukuvyhaa'}],
>         model: 'TS0504B',
>         vendor: [...]

@MattiaFochesato I'm new here. Where does this code go?

deeja, 2021-09-22:
> @MattiaFochesato
> 
> Thanks for this idea! But how to go back to RGB? I do not come back when I enter a color...
> 
> And I can't control brightness after this change...

`zigbee2mqtt/[ID here]/set`
To white:
```json
{"color_temp": 1}
```
to get [back to color](https://www.zigbee2mqtt.io/devices/TS0504B.html):
```json
{"color":{"hex":"#547CFF"}}
````

pawelbudynek, 2021-10-05:
The problem still exist, I do have original gateway, so I can try to help, but I need instructions :)

rodrigoscoelho, 2021-10-05:
Yes. It is not working as expected. And it is a product with a high number
os sells. It would be great to have it working 100%.

Enviado utilizando um Galaxy S10...

th3d00d, 2021-11-06:
any solution to use it in Home Assistant so far?

rodrigoscoelho, 2021-11-07:
Nope. We really need an expert help on this

Enviado utilizando um Galaxy S10...

pchrzaszczewski, 2021-11-11:
I have the same problem with 7 controllers and I have no idea how to solve it.
Please help!

ferrarii1, 2021-11-18:
I have a controller like this and a zigbee sniffer. When you press the button on the controller to turn on white, it doesn't send anything. How else can you catch the inclusion of white?

ferrarii1, 2021-11-19:
sniff zigbee traffic from the zigbee gateway tuya to the led stripe.
link to the video that I clicked [https://youtu.be/c2Sy4lzHor0](url)
link yo pcap file [https://drive.google.com/file/d/1lWOpJUaLwiAmXNT-6Kf3t8kBtMO2b-ng/view?usp=sharing](url)

Koenkk, 2021-11-19:
@ferrarii1 do you have the network key for me?

ferrarii1, 2021-11-19:
@Koenkk 6f9c97b9bb168b7236017d86be11ee61

Koenkk, 2021-11-19:
What happens when you change the color temperature when using the following external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0504B', manufacturerName: '_TZ3000_ukuvyhaa'}],
    model: 'TS0504B',
    vendor: 'TuYa',
    description: 'RGB+CCT LED _ EXTERNAL CONVERTER',
    toZigbee: [tz.on_off, tz.tuya_led_control],
    fromZigbee: [fz.on_off, fz.tuya_led_controller, fz.brightness, fz.ignore_basic_report],
    exposes: [e.light_brightness_colortemp_colorhs([153, 500]).removeFeature('color_temp_startup')],
},;

module.exports = definition;

```

(after adding the converter and restarting z2m, make sure it loaded it by going to the frontend, the description of this device should now be: `RGB+CCT LED _ EXTERNAL CONVERTER`)

How to user external converter: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions

If that doesn't help, you will need to reverse engineer the TuYa commands that control the white mode, examples in your sniff are no. 41, no. 63

pawelbudynek, 2021-11-19:
Guys that's a breakthrough indeed!

rodrigoscoelho, 2021-11-19:
THE GUY is on. Now it will work. 😂

ferrarii1, 2021-11-19:
Do you need a pcap? or tell how the LED strip behaves?

ferrarii1, 2021-11-19:
> What happens when you change the color temperature when using the following external converter?
>
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

Do you need a pcap? or tell how the LED strip behaves?

Koenkk, 2021-11-19:
tell how the strip behaves when changing the color temperature

ferrarii1, 2021-11-19:
does not react to temperature changes. 
but at least from the color mode can be displayed in white. Ideally, make sure that when you press white, it turns on not the RGB white, but white LEDs.
[image]
the converter is written for RGB+CCT, and this controller is RGB+W.

Koenkk, 2021-11-22:
@ferrarii1 could you provide a which only contains setting the white channel from the TuYa gateway?

ferrarii1, 2021-11-22:
pcap file where did I include white?

Koenkk, 2021-11-22:
@ferrarii1 yes, produced by:
- start sniffing
- control white channel **once**
- stop sniffing

ferrarii1, 2021-11-23:
no. 163 on white channel
no 174-192  white channel from 0% minimum to 100% maximum
no 210 off
(https://drive.google.com/file/d/1bxewenq0aso8AGhMUt816zOme78IxVSB/view?usp=sharing)

Koenkk, 2021-11-23:
Coud you
- update to the latest dev branch
- add the following external converter: (see https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions how to do this)
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    test_1: {
        key: ['test_1'],
        convertSet: async (entity, key, value, meta) => {
            await entity.command('lightingColorCtrl', 'tuyaWhite', {data: [0x31, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x33, 0x33, 0x34, 0x30, 0x33, 0x65, 0x38]});
        },
    },
    test_2: {
        key: ['test_2'],
        convertSet: async (entity, key, value, meta) => {
            await entity.command('lightingColorCtrl', 'tuyaRgbMode2', {data: [0]});
        },
    },
}

const definition = {
    fingerprint: [{modelID: 'TS0504B', manufacturerName: '_TZ3000_ukuvyhaa'}, {modelID: 'TS0504B', manufacturerName: '_TZ3210_ttkgurpb'}],
    model: 'TS0504B',
    vendor: 'TuYa',
    description: 'Zigbee RGBW light - EXTERNAL DEFINTION',
    exposes: extend.light_onoff_brightness_color().exposes,
    toZigbee: [...extend.light_onoff_brightness_color().toZigbee, tzLocal.test_1, tzLocal.test_2],
    fromZigbee: extend.light_onoff_brightness_color().fromZigbee,
    meta: {applyRedFix: true},
};

module.exports = definition;
```
- open the frontend, make sure the external converter work, the of the device should be `Zigbee RGBW light - EXTERNAL DEFINTION'` now
- if yes, what happens when you send to `zigbee2mqtt/DEVICE_FRIENDLY_NAME/set`
  - payload: `{"test_1":""}`
  - payload: `{"test_2":""}`

ferrarii1, 2021-11-24:
debug Received MQTT message on 'zigbee2mqtt/0x9035eafffe583b6d/set' with data '{"test_1":""}'
error No converter available for 'test' (undefined)
debug Received MQTT message on 'zigbee2mqtt/0x9035eafffe583b6d/set' with data '{"test_2":""}'
error Device '0x9035eafffe583b6d' has no endpoint '2'
[image]
[image]

Koenkk, 2021-11-24:
@ferrarii1 can you check with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    test_1: {
        key: ['test1'],
        convertSet: async (entity, key, value, meta) => {
            await entity.command('lightingColorCtrl', 'tuyaWhite', {data: Buffer.from([0x31, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x33, 0x33, 0x34, 0x30, 0x33, 0x65, 0x38])});
        },
    },
    test_2: {
        key: ['test2'],
        convertSet: async (entity, key, value, meta) => {
            await entity.command('lightingColorCtrl', 'tuyaRgbMode2', {data: Buffer.from([0])});
        },
    },
}

const definition = {
    fingerprint: [{modelID: 'TS0504B', manufacturerName: '_TZ3000_ukuvyhaa'}, {modelID: 'TS0504B', manufacturerName: '_TZ3210_ttkgurpb'}],
    model: 'TS0504B',
    vendor: 'TuYa',
    description: 'Zigbee RGBW light - EXTERNAL DEFINTION',
    exposes: extend.light_onoff_brightness_color().exposes,
    toZigbee: [...extend.light_onoff_brightness_color().toZigbee, tzLocal.test_1, tzLocal.test_2],
    fromZigbee: extend.light_onoff_brightness_color().fromZigbee,
    meta: {applyRedFix: true},
};

module.exports = definition;

```

ferrarii1, 2021-11-25:
@Koenkk, ok.
[test1.txt](https://github.com/Koenkk/zigbee2mqtt/files/7599801/test1.txt)
[test2.txt](https://github.com/Koenkk/zigbee2mqtt/files/7599802/test2.txt)

Koenkk, 2021-11-25:
And with:

```js
const Buffer = require('buffer');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    test_1: {
        key: ['test1'],
        convertSet: async (entity, key, value, meta) => {
            await entity.command('lightingColorCtrl', 'tuyaWhite', {data: Buffer.from([0x31, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x33, 0x33, 0x34, 0x30, 0x33, 0x65, 0x38])});
        },
    },
    test_2: {
        key: ['test2'],
        convertSet: async (entity, key, value, meta) => {
            await entity.command('lightingColorCtrl', 'tuyaRgbMode2', {data: Buffer.from([0])});
        },
    },
}

const definition = {
    fingerprint: [{modelID: 'TS0504B', manufacturerName: '_TZ3000_ukuvyhaa'}, {modelID: 'TS0504B', manufacturerName: '_TZ3210_ttkgurpb'}],
    model: 'TS0504B',
    vendor: 'TuYa',
    description: 'Zigbee RGBW light - EXTERNAL DEFINTION',
    exposes: extend.light_onoff_brightness_color().exposes,
    toZigbee: [...extend.light_onoff_brightness_color().toZigbee, tzLocal.test_1, tzLocal.test_2],
    fromZigbee: extend.light_onoff_brightness_color().fromZigbee,
    meta: {applyRedFix: true},
};

module.exports = definition;
```

ferrarii1, 2021-11-25:
@Koenkk 

debug Received MQTT message on 'zigbee2mqtt/0x9035eafffe583b6d/set' with data '{"test2":""}'
debug Publishing 'set' 'test2' to '0x9035eafffe583b6d'
error Publish 'set' 'test2' to '0x9035eafffe583b6d' failed: 'TypeError: Buffer.from is not a function'
debug TypeError: Buffer.from is not a function at Object.convertSet (/opt/zigbee2mqtt/dist/util/externally-loaded.js:20:85) at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:247:52) at EventEmitter.emit (events.js:412:35) at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:95:22) at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:113:27) at MqttClient.emit (events.js:400:28) at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1277:12) at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:410:12) at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:321:12) at Writable.writable._write (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:335:5)
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'test2' to '0x9035eafffe583b6d' failed: 'TypeError: Buffer.from is not a function'","meta":{"friendly_name":"0x9035eafffe583b6d"},"type":"zigbee_publish_error"}'

ferrarii1, 2021-11-25:
const Buffer = require('buffer').Buffer;
work

Koenkk, 2021-11-25:
Is it possible to change the white brightness when setting it in white mode now?

ferrarii1, 2021-11-26:
yes, with the slider the brightness in the exposes

Anatol6868, 2021-11-26:
Command: Unknown (0xf0)  
159	22.067717	0x0000	0xb77a	ZigBee HA	50	ZCL Level Control: Unknown Command, Seq: 52
[https://github.com/Koenkk/zigbee2mqtt/issues/8575](url)
Help Wanted ?

ferrarii1, 2021-11-27:
[image]
the dimmer works in this model

Koenkk, 2021-11-27:
@ferrarii1 So if I understand correctly test2 is enough to control the white channel? e.g.
- Device is in rgb mode
- send test 2
- rgb mode is now disabled, white activated
- white brightness can be controlled now with brightness slider
- to switch back to rgb mode simply set a color

ferrarii1, 2021-11-27:
Yes that's right. I think that will be enough. 

but how can you solve the issue with color rendition?
when you turn on the red, pink is displayed on the LED strip. yellow with a tint of green.

Koenkk, 2021-11-29:
@ferrarii1 could you provide me an aliexpress link where you bought the device? I believe it will be too hard to integrate this device without having it.

bluefish007, 2021-11-29:
@Koenkk I ordered here aliexpress.com/item/1005001865233095.html

pchrzaszczewski, 2021-11-29:
@Koenkk I ordered this one https://aliexpress.com/item/1005002554523847.html

Koenkk, 2021-11-29:
thanks, ordered it.

[19 more replies]

closed 2021-12-27 by Koenkk