Koenkk/zigbee2mqtt #6409: CC2538 + CC2592 Z-stack 3 - Non responsive Zigbee after power failure or disconnect.

Multipage, 2021-02-22:
Anyone experiencing the same thing?

How to reproduce:
Have a working installation of zigbee2mqtt. Have paired ZigBee devices. Now WHILE the machine is running pull the USB-cord so the coordinator is disconnected. Wait for 3 minutes or so. Reconnect coordinator. Wait untill Zigbee2MQTT reconnects to the coordinator. 
Result: ALL your paired devices all still there but unresponsive. Doesn't make a difference if you wait 10 minutes or 48 hours, multiple reboots/restarts. They will never respond again unless you re-pair them.
This also applies on power loss of the whole machine. You will have to re-pair ALL ZigBee devices. You will have to force remove the battery powered ones  them first before adding them.

Pairings WILL survive a reboot. They willl respond after reboot. NOT after power interruption

I allready tried to move to channel 20 and use different firmwares ( all latest 2538+2592 Z-stack 3.0.x ones)

Zigbee2MQTT version: ALWAYS latest
Adapter hardware: CC2538 + CC2592
Adapter firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20201010,"transportrev":2},"type":"zStack30x"}'
(JH_2538_2592_ZNP_USB_20201010)

When i try to send a configure command ( not respondig ) :
 Failed to configure '0xec1bbdfffe6b70c1', attempt 1 (Error: Bind 0xec1bbdfffe6b70c1/1 genOnOff from '0x00124b0018e300f7/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:554:17) at processTimers (internal/timers.js:497:7))
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0xec1bbdfffe6b70c1"},"error":"Failed to configure (Bind 0xec1bbdfffe6b70c1/1 genOnOff from '0x00124b0018e300f7/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"mhbty-1"}'

This is what i see whemn zigbee2mqtt starts:
Zigbee2MQTT:info  2021-02-22 13:42:12: Currently 1 devices are joined:
Zigbee2MQTT:info  2021-02-22 13:42:12: 0xec1bbdfffe6b70c1 (0xec1bbdfffe6b70c1): TS0001 - TuYa 1 gang switch (Router)
Zigbee2MQTT:info  2021-02-22 13:42:12: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2021-02-22 13:42:12: Started frontend on port 0.0.0.0:8099
Zigbee2MQTT:info  2021-02-22 13:42:12: Connecting to MQTT server at mqtt://192.168.0.20:1883
Zigbee2MQTT:info  2021-02-22 13:42:12: Connected to MQTT server
Zigbee2MQTT:info  2021-02-22 13:42:12: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2021-02-22 13:42:12: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe6b70c1', payload '{"linkquality":31,"state":"OFF"}'

So yes....it should still be paired..it is...but unresponsive

Multipage, 2021-02-22:
Just tried an old CC2531 with Z-stack 3.0.x FW. Did the same thing. Unplugged while powered on...replugged back in...all stil works.

So ...why isn't this working with the CC2538

Koenkk, 2021-02-22:
@0x3EC is this a known issue?

0x3EC, 2021-02-22:
This reminds me of this issue https://github.com/Koenkk/Z-Stack-firmware/issues/225#issuecomment-759060054
@Multipage, show your **coordinator_backup.json** file

Multipage, 2021-02-23:
Hi @0x3EC 
I reinstalled the CC2538+CC2592, paired a device. Disconnected it. Waited 5 minutes and reconnected. After that i copied the coordinator_backup.json file
Here is the file : https://pastebin.com/CrAwrt4y

If it's the same problem it's unfixable ?

Multipage, 2021-02-28:
@0x3EC : Any news/comment
If this is unfixable, all 2538 and 2562's are useless with Zigbee2mqtt. That's a huge problem

0x3EC, 2021-03-01:
Hi, @Multipage 

This should help you https://github.com/Koenkk/zigbee2mqtt/blob/master/scripts/zStackEraseAllNvMem.js

But before erasing, can you save memory dump of the stick?
Do you have a zigbee sniffer?

TroLoos, 2021-03-03:
I have exactly the same issue... and for me as well - it wasn't present when I used CC2531. This night I had a power outage and my z2m host (RPi) went off.

After booting back up, all of my 63 devices are disconnected. My network consists of ~30 routers (ac powered wall switches) and ~30 end devices (battery powered).

I have CC2538 w/ CC2592 RPi HAT Module (zStack30x, revision 20201010). Actually, I had exactly the same issue when my network was running under ZigBee stick v4 (CC2652P E72-2G4M20S1E).

0x3EC, 2021-03-03:
@TroLoos, show your coordinator_backup.json file

If you have a CC2531, you can make a sniffer out of it. This information is needed to solve the problem.

TroLoos, 2021-03-03:
Sure, there you go:

`https://pastebin.com/NAmDx0A8`

0x3EC, 2021-03-03:
@TroLoos, I have changed your file. You need to do this:  

  0. Stop z2m
  1. Erase the NV_MEM area (using a script https://github.com/Koenkk/zigbee2mqtt/blob/master/scripts/zStackEraseAllNvMem.js)
  2. Reload the stick.
  3. Replace the cordinator_backup.json file with this https://pastebin.com/BUfQdud1
  4. Start z2m.
  5. Check the devices on the network.

TroLoos, 2021-03-03:
@0x3EC ,

I did the following:

*[yes] Stop z2m*
*[yes] Erase the NV_MEM area (using a script https://github.com/Koenkk/zigbee2mqtt/blob/master/scripts/zStackEraseAllNvMem.js)*

`Detected znp version 'zStack30x' ({"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20201010})
Clearing all NVMEM items, from 0 to 831
NVMEM item #58 - deleting, size: 17
NVMEM item #59 - deleting, size: 17
NVMEM item #130 - deleting, size: 24
Clearing all NVMEM items finished, deleted 3 items`

*[yes/no] Reload the stick.*

It is a HAT for my RPi - I cannot remotely unplug it and plug it back in. I restarted my RPi hoping that it would fullfill this step.

*[yes] Replace the cordinator_backup.json file with this https://pastebin.com/BUfQdud1*
*[yes] Start z2m.*

Check the devices on the network.

And the result is like this:
`Mar 03 13:37:00 zigbeeserver systemd[1]: Started zigbee2mqtt.
Mar 03 13:37:05 zigbeeserver npm[494]: > zigbee2mqtt@1.18.0-dev start /opt/zigbee2mqtt
Mar 03 13:37:05 zigbeeserver npm[494]: > node index.js
Mar 03 13:37:11 zigbeeserver npm[494]: Zigbee2MQTT:debug 2021-03-03 13:37:11: Loaded state from file /opt/zigbee2mqtt/data/state.json
Mar 03 13:37:11 zigbeeserver npm[494]: Zigbee2MQTT:info  2021-03-03 13:37:11: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-03-03.13-37-08' filename: log.txt
Mar 03 13:37:11 zigbeeserver npm[494]: Zigbee2MQTT:debug 2021-03-03 13:37:11: Removing old log directory '/opt/zigbee2mqtt/data/log/2021-03-03.07-21-12'
Mar 03 13:37:11 zigbeeserver npm[494]: Zigbee2MQTT:info  2021-03-03 13:37:11: Starting Zigbee2MQTT version 1.18.0-dev (commit #b9c69f9)
Mar 03 13:37:11 zigbeeserver npm[494]: Zigbee2MQTT:info  2021-03-03 13:37:11: Starting zigbee-herdsman (0.13.74)
Mar 03 13:37:11 zigbeeserver npm[494]: Zigbee2MQTT:debug 2021-03-03 13:37:11: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"baudRate":115200,"path":"/dev/ttyAMA0","rtscts":false}}'
Mar 03 13:37:14 zigbeeserver npm[494]: Zigbee2MQTT:error 2021-03-03 13:37:14: Error while starting zigbee-herdsman
Mar 03 13:37:14 zigbeeserver npm[494]: Zigbee2MQTT:error 2021-03-03 13:37:14: Failed to start zigbee
Mar 03 13:37:14 zigbeeserver npm[494]: Zigbee2MQTT:error 2021-03-03 13:37:14: Exiting...
Mar 03 13:37:14 zigbeeserver npm[494]: Zigbee2MQTT:error 2021-03-03 13:37:14: Error: SREQ '--> SYS - osalNvWrite - {"id":1,"offset":0,"osal":true,"product":-1,"value":[109,42,131,31,0,75,18,0],"len":8}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)')
Mar 03 13:37:14 zigbeeserver npm[494]:     at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
Mar 03 13:37:14 zigbeeserver npm[494]:     at Generator.next (<anonymous>)
Mar 03 13:37:14 zigbeeserver npm[494]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
Mar 03 13:37:14 zigbeeserver npm[494]: npm ERR! code ELIFECYCLE
Mar 03 13:37:14 zigbeeserver npm[494]: npm ERR! errno 1
Mar 03 13:37:14 zigbeeserver npm[494]: npm ERR! zigbee2mqtt@1.18.0-dev start: `node index.js`
Mar 03 13:37:14 zigbeeserver npm[494]: npm ERR! Exit status 1
Mar 03 13:37:14 zigbeeserver npm[494]: npm ERR!
Mar 03 13:37:14 zigbeeserver npm[494]: npm ERR! Failed at the zigbee2mqtt@1.18.0-dev start script.
Mar 03 13:37:14 zigbeeserver npm[494]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Mar 03 13:37:14 zigbeeserver npm[494]: npm ERR! A complete log of this run can be found in:
Mar 03 13:37:14 zigbeeserver npm[494]: npm ERR!     /home/pi/.npm/_logs/2021-03-03T12_37_14_730Z-debug.log
Mar 03 13:37:14 zigbeeserver systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Mar 03 13:37:14 zigbeeserver systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Mar 03 13:37:14 zigbeeserver systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Mar 03 13:37:14 zigbeeserver systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 1.
Mar 03 13:37:14 zigbeeserver systemd[1]: Stopped zigbee2mqtt.
`

0x3EC, 2021-03-03:
_It is a HAT for my RPi - I cannot remotely unplug it and plug it back in. I restarted my RPi hoping that it would fullfill this step._
Need a cold restart of the stick

TroLoos, 2021-03-03:
Ok, I did exactly that: stop, clear nvram, update coordinator_backup.json, reset device (there's a reset button on my HAT), and finally start. It loads up just fine. But there's no single message coming from any of the devices.

It loads a lot of information about my devices, but they don't speak to the coordinator, any of them does :-(.

0x3EC, 2021-03-03:
Stop z2m and show again the file coordinator_backup.json

TroLoos, 2021-03-03:
> Stop z2m and show again the file coordinator_backup.json

There you go: [https://pastebin.com/9nLWQHZF](https://pastebin.com/9nLWQHZF)

I am willing to give you remote console (via TeamViewer for example) if you feel like battling with this issue :-)

0x3EC, 2021-03-04:
@TroLoos 
Thanks for the invitation, but it won't work)
If you can listen to the radio channel with a zigbee sniffer, you will know exactly what the matter is. I think your network started with a different PANID.

The problem was originally caused by migrating from CC2531 to CC2538. In your file, the NIB table was not the correct size. Or when you first launched z2m, the stick had another firmware in which the NIB table was 110 bytes.
If you are tired of these problems, then you can reconnect the devices with the file that I sent you and there should be no problems.

0x3EC, 2021-03-04:
@Multipage 
You also have an incorrect NIB table size.
If your problem is not resolved, then I can try to fix your file and try to repair your network.

TroLoos, 2021-03-04:
> @TroLoos
> Thanks for the invitation, but it won't work)
> If you can listen to the radio channel with a zigbee sniffer, you will know exactly what the matter is. I think your network started with a [...]

I would like to thank you for time you've spent on investigating this issue.

Right now I decided to go ahead and re-install it from scratch to break the CC2531 migration path and start clear. I have a CC2652P zigbee stick I used for this purpose.

I will observe if this setup survives unexpected shutdowns and restarts and if it works - that's just great :-). If it doesn't - then we will have some base knowledge after this thread to start debuging process. Your explanation sounds very reasonable so I hope it's gonna be easier next time, although I hope that there will be no next time ;-).

Thank you again!

Multipage, 2021-03-15:
@0x3EC . I'm sorry . i was very busy lately. Since i had 2 CC2538's in the same drawer laying around i mixed them up. I didn't which one was connected. So  i can't snif. I cleaned the CC2538, stopt and removed zigbee2mqtt. Reinstalled it and paired the devices. While running i disconnected the CC2538 by disconnecting the USB cable. Reconnected it and all my devices were still responsive. 
So the cleaning worked! Thanks!
I hope it will continue to work.

closed 2021-03-31 by Multipage