Koenkk/zigbee2mqtt #6416: Docker Compose instructions

877dev, 2021-02-22:
## Is your feature request related to a problem? Please describe
I use [IOTstack ](https://sensorsiot.github.io/IOTstack/) which uses Docker compose, and find the Docker instructions don't work for me, such as here: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#docker

## Describe the solution you'd like
I would like the Docker compose commands to be shown.

## Describe alternatives you've considered
I tried following the guide to support new devices, but always get stuck at:
`docker run -v "$PWD":/app -w /app --rm node:12 npm ci`
I don't know the Docker Compose equivalent.

## Additional context
My docker-compose.yml looks like this:
```
zigbee2mqtt:
    devices:
    - /dev/ttyACM0:/dev/ttyACM0
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest-dev
    volumes:
    - ./volumes/zigbee2mqtt/data:/app/data
    - /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters
    restart: unless-stopped
    network_mode: host
```

ozer59, 2021-02-23:
same for me link for zigbee-herdsman-converters don't work but for /app/data is ok 
https://github.com/Koenkk/zigbee2mqtt/issues/6350#issuecomment-783558621

877dev, 2021-02-23:
@ozer59 are you using docker compose? As that is what I am using and raised the issue for.

Koenkk, 2021-02-23:
Your docker compose looks OK to me (`- /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters`) Make sure that `/opt/zigbee-herdsman-converters` exists on your system (execute `ls /opt/zigbee-herdsman-converters`).

To remove local changes simply remove this line from your docker-compose.

877dev, 2021-02-23:
Thanks for the help, I'm afraid problems persist:

[image]

docker-compose.yml:
```
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    volumes:
      - ./volumes/zigbee2mqtt/data:/app/data
      - /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters
    devices:
      #- /dev/ttyAMA0:/dev/ttyACM0
      - /dev/ttyACM0:/dev/ttyACM0
    restart: unless-stopped
    network_mode: host
```

Portainer zigbee2mqtt debug output:
[zigbee.log](https://github.com/Koenkk/zigbee2mqtt/files/6031449/zigbee.log)

m0wlheld, 2021-02-23:
> I don't know the Docker Compose equivalent.

There is none, that‘s a one-time task and the container created is removed immediately after.

m0wlheld, 2021-02-23:
> Also regarding supporting new devices, how to revert back from using local devices.js

remove the 2nd volume bind from compose file

877dev, 2021-02-23:
@m0wlheld thank you! Any idea on the error's I am having only  when mapping the 2nd volume ?

` - /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters`

[zigbee.log](https://github.com/Koenkk/zigbee2mqtt/files/6031548/zigbee.log)

ozer59, 2021-02-23:
yes i try it but same issue 
for me app/data is ok with : - /volume1/docker/zigbee2mqttstable:/app/data
but not for converters
    zigbee2mqtt:
      container_name: zigbee2mqtt
      image: koenkk/zigbee2mqtt:latest
      volumes:
        - /volume1/docker/zigbee2mqttstable:/app/data
        - /volume1/docker/zigbee2mqttstable/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters
        - /run/udev:/run/udev:ro
      devices:
        - /dev/ttyACM0:/dev/ttyACM0
      restart: always
      privileged: true
      environment:
        - TZ=Europe/Paris

877dev, 2021-02-23:
@ozer59 looks like the same issue after all.
Maybe something to do with this? https://github.com/Koenkk/zigbee2mqtt/issues/3291

877dev, 2021-02-23:
Some success, seems to be working now!

Try:
```
cd  /opt/zigbee-herdsman-converters
sudo npm ci
```

Then restart zigbee container..

ozer59, 2021-02-23:
sorry but i don't unterstand this command : sudo docker run -v "$PWD":/app -w /app --rm node:12 npm ci
ozer@HOUSE:/volume1/docker/zigbee2mqttstable/zigbee-herdsman-converters$ sudo docker run -v "$PWD":/app -w /app --rm node:12 npm ci
npm WARN prepare removing existing node_modules/ before installation

> @serialport/bindings@9.0.4 install /app/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> serialport@9.0.6 postinstall /app/node_modules/zigbee-herdsman/node_modules/serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

added 699 packages in 18.982s

877dev, 2021-02-23:
If you read through the comments the answer is there already...

Did you try the commands I posted?

m0wlheld, 2021-02-24:
The docker command created the directories and files in the current directory (which - following the guide — should be /opt/zigbee-herdsman-converters), so you can bind-mount it to the Z2M container using docker-compose. 

It is only required, if you want to add unsupported devices.

877dev, 2021-02-24:
@Koenkk here is the solution for docker compose which I can confirm working, could it be added to the "supporting new devices" guide?

## Docker Compose

To setup a local copy of zigbee-herdsman-converters so that you can modify e.g. `devices.js` and from`Zigbee.js`; use the following instructions:

`cd /opt`
`git clone https://github.com/Koenkk/zigbee-herdsman-converters.git`
`cd zigbee-herdsman-converters`
`sudo npm ci`

Now add the volume to your `docker-compose.yml `
Example:
```
zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    volumes:
      - ./volumes/zigbee2mqtt/data:/app/data
      - /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters
    devices:
      #- /dev/ttyAMA0:/dev/ttyACM0
      - /dev/ttyACM0:/dev/ttyACM0
    restart: unless-stopped
    network_mode: host
```

m0wlheld, 2021-02-24:
> cd /opt
git clone https://github.com/Koenkk/zigbee-herdsman-converters.git
cd zigbee-herdsman-converters
sudo npm ci

That is basically the result of the `docker run`-call with the requirement to have npm installed locally. I don't see no benefit for an environment, if Docker is already installed.

877dev, 2021-02-24:
https://github.com/Koenkk/zigbee2mqtt/issues/3291
> After the clone a `npm ci` has to be done to install the dependencies. I've updated the instructions accordingly. thanks!

I'm not sure , but I do know it works. I never use `docker-run` as I am using Docker Compose.

m0wlheld, 2021-02-24:
> I never use `docker-run` as I am using Docker Compose.

... which orchestrates docker run commands ...

877dev, 2021-02-24:
> That is basically the result of the `docker run`-call with the requirement to have npm installed locally. I don't see no benefit for an environment, if Docker is already installed.

I think I am understanding what you are saying. So how would you recommend getting the dependencies to be installed when up'ing the container in compose?

Koenkk, 2021-02-24:
Does it work with:

```yaml
zigbee2mqtt:
    devices:
    - /dev/ttyACM0:/dev/ttyACM0
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest-dev
    volumes:
    - ./volumes/zigbee2mqtt/data:/app/data
    - /opt/zigbee-herdsman-converters/index.js:/app/node_modules/zigbee-herdsman-converters/index.js
    - /opt/zigbee-herdsman-converters/devices.js:/app/node_modules/zigbee-herdsman-converters/devices.js
    - /opt/zigbee-herdsman-converters/converters:/app/node_modules/zigbee-herdsman-converters/converters
    - /opt/zigbee-herdsman-converters/lib:/app/node_modules/zigbee-herdsman-converters/lib
    restart: unless-stopped
    network_mode: host
```

877dev, 2021-02-24:
> Does it work with:

Yes! I started from fresh and that worked perfectly, thank you! No need to do `npm ci`.
Can it be added to the guide please?

Koenkk, 2021-02-24:
Ah wait, this all shouldn't be necessary when following the instructions from https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#docker . Did you try these?

m0wlheld, 2021-02-24:
Koenkk, that guide has been correct from the beginning, no need to change.

877dev, 2021-02-24:
> Ah wait, this all shouldn't be necessary when following the instructions from https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#docker . Did you try these? Based on your error I think you missed the last line: `docker run -v "$PWD":/app -w /app --rm node:12 npm ci`

That last line always gave me an error, however adding your lines to docker-compose.yml worked for me.

m0wlheld, 2021-02-25:
Final attempt to help:

First, please understand that docker-compose is just a tool to orchestrate Docker containers, you need to have the Docker daemon and the Docker CLI (the `docker` command) to be installed. 

Second, please understand, that the `docker run` command from the How-To is a one-time-run only. It will create the required files, that you can edit and later use to "overlay" the original files in the koenkk/zigbee2mqtt image. Running the command again will infact reset those files and remove your  changes / additions. Therefore, it **must not be present** in your docker-compose file.

So, let's step through the howto. 

    cd /opt

Change to the /opt directory. That would actually be my only criticism of the how-to, because the directory "/opt" can only be written to by the root user. So you have to execute all commands of the how-to as root.

    git clone https://github.com/Koenkk/zigbee-herdsman-converters.git

Create directory /opt/zigbee-herdsman-converters  and download source from Git to it.

    cd zigbee-herdsman-converters

Change into the recently created directory /opt/zigbee-herdsman-converters

    docker run -v "$PWD":/app -w /app --rm node:12 npm ci

* download Docker image "node:12"
* `run`: create a container from it
* `-v`: make the current directory ("$PWD") available *inside* the container at `/app`
* `-w`: change the working directory *inside* the container to `/app`
* `--rm`: remove the container instance, as soon as the process inside the container ends
* use command `npm ci` as entrypoint process for the container

Because the working directory inside the container is `/app` and at the same time `/app` is "mapped" to the host's directory `/opt/zigbee-herdsman-converters`, any output from `npm ci` will end in the host directory `/opt/zigbee-herdsman-converters`. Effectively, this creates directory `/opt/zigbee-herdsman-converters/node_modules`

The same thing happens when you run `npm ci` directly on the host, but you need npm to be installed on the host. 

Finally, by adding 

    -v /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters

to your docker-compose file, you overlay the directory `/app/node_modules/zigbee-herdsman-converters` *inside* the Zigbee2MQTT container by the host's directory `/opt/zigbee-herdsman-converters`. Changes to the files in your local directory are reflected in the Zigbee2MQTT container upon restart. 

I do hope that now it is clear, that the steps of the How-To work and there is no error in it. In #6102 you wrote

> The docker run command did run but I got an error regarding node:12 not found or similar

Let me guess: `Unable to find image 'node:12' locally`? That's not an error, just a message that the container image needs to be download first.

877dev, 2021-02-25:
> Final attempt to help:
Actually this is first helpful thing you have posted :)

But seriously thanks for the detailed post, very helpful and it worked as expected this time.

> Let me guess: `Unable to find image 'node:12' locally`? That's not an error, just a message that the container image needs to be download first.

Yes I believe that was the error, which led me to believe the command was not compatible with my docker compose setup.  Should the command `docker run -v "$PWD":/app -w /app --rm node:12 npm ci` download the node:12 container image? As there is no instructions on downloading it prior to running the command.

m0wlheld, 2021-02-25:
>     1. Should the command `docker run -v "$PWD":/app -w /app --rm node:12 npm ci` download the node:12 container image? (There are no instructions on downloading it prior to running the command).

It will "pull" (download) the container image automatically. You can track the download progress on screen.

>     2. Is the command a 'one time only', or does it need to be run after making any changes?

Not 100% sure. I'll assume that you have to re-run it whenever a new Zigbee2MQTT version is published. Because the library "node_modules" must match the Zigbee2MQTT app inside the container. 

Better ask @Koenkk for feedback.

Koenkk, 2021-02-25:
To be on the safe side, run `docker run -v "$PWD":/app -w /app --rm node:12 npm ci` every time you do a `git pull` in `/opt/zigbee-herdsman-converters`.

closed 2021-03-28 by Koenkk