Koenkk/zigbee2mqtt #6422: Connection errors occuring spontaneously

fabian494, 2021-02-23:
## What happened

After a few weeks of normal operation suddenly the Coordinator seems to loose connection to the device (only one 2 output switch in the network) after it's been running a while.
As a result the switch randomly switches states, resulting in a disco-light like behaviour, until it finally converges to either the state it should be set to or both outputs turned off after a few minutes.
Right now it happens after it's been running for like 30 minutes.
Nothing has changed in the actual setup btw.

## What did you expect to happen

Stable connection.

## How to reproduce it (minimal and precise)

The only things I have in my network are the Coordinator (ConBee II) and one Lupus 12127 2 channel switch.
I always set the state of both outputs at the same time `{"state_l1": "ON", "state_l2": "OFF"}` or `{"state_l1": "OFF", "state_l2": "ON"}` (it switches some kind of traffic light) via MQTT to _zigbee2mqtt/0x00124b001fff2ec1/set_.
The ConBee II stick is plugged into the USB2 port on a Raspberry Pi 4B running the latest Raspbian/RaspiOS via an extension cable.
This even happens when I put the ConBee II stick and the Lupus directly next to each other.

configuration.yaml   
```   
homeassistant: false
advanced:
  log_level: debug
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM0
devices:
  '0x00124b001fff2ec1':
    friendly_name: '0x00124b001fff2ec1'
```

## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: Conbee II
Adapter firmware version: 26680700 (latest)
Device: Lupus 12127 2 channel switch

The debug log shows:   
```
Zigbee2MQTT:error 2021-02-22 18:00:12: Publish 'set' 'state' to '0x00124b001fff2ec1' failed: 'Error: Command 0x00124b001fff2ec1/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
Zigbee2MQTT:debug 2021-02-22 18:00:12: Error: Command 0x00124b001fff2ec1/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)
    at DeconzAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:606:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:25:65)
Zigbee2MQTT:info  2021-02-22 18:00:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x00124b001fff2ec1' failed: 'Error: Command 0x00124b001fff2ec1/2 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (no response received)'","meta":{"friendly_name":"0x00124b001fff2ec1"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:debug 2021-02-22 18:00:12: Received MQTT message on 'zigbee2mqtt/0x00124b001fff2ec1/set' with data '{"state_l1": "ON", "state_l2": "OFF"}'
Zigbee2MQTT:debug 2021-02-22 18:00:12: Publishing 'set' 'state' to '0x00124b001fff2ec1'
Zigbee2MQTT:debug 2021-02-22 18:00:13: Received Zigbee message from '0x00124b001fff2ec1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-22 18:00:13: MQTT publish: topic 'zigbee2mqtt/0x00124b001fff2ec1', payload '{"linkquality":67,"state_l1":"ON","state_l2":"OFF"}'
```

fabian494, 2021-02-24:
Right now it works fine for about 5 minutes after I restart zigbee2mqtt but then the problem occures.
I can see that the states of the _zigbee2mqtt/0x00124b001fff2ec1_ topic flap all the time (`{"linkquality":133,"state_l1":"ON","state_l2":"OFF"}`). The link quality seems to be good though.

Koenkk, 2021-02-24:
Could be interference: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html

closed 2021-03-27 by Koenkk