Koenkk/zigbee2mqtt #6423: Request: MQTT - Homeassistant

forfrossen, 2021-02-23:
Would it be possible to populate z2m properties of devices via homeassistant? Values like "last_seen" and "enlapsed".

I tried auto discovery on openhab for a couple of contact sensors and realized they only have their 'exposes' properties.
In openhab you're not able manually edit auto-discovered things. So I could not get the values from above.

In the discord channel I was told to create an issue here asking for it to be implemented.

RubenKelevra, 2021-02-24:
You can enable this in Home Assistant's integration of zigbee2mqtt:

[image]

forfrossen, 2021-02-26:
I'm afraid, I think that was misleading from my side. I already did that.
The MQTT topics exist and when I manually configure them in openhab everything works.
I was asking to get them to be auto-discovered. As far as I can see the channels are not added to the homeassistant config topics.

RubenKelevra, 2021-03-29:
> I'm afraid, I think that was misleading from my side. I already did that.
> The MQTT topics exist and when I manually configure them in openhab everything works.
> I was asking to get them to be auto-discovered. As far as I can see the channels are not added to the homeassistant config topics.

Not sure what you're referring to, the timestamps are being displayed and can be used in HA:

[image]

closed 2021-05-07 by github-actions[bot]