Koenkk/zigbee2mqtt #6424: Zigbee Groups having multiple Lights assigned don't change change/report state when triggered by remote

pavax, 2021-02-23:
## What happened
If I create a group with more than two lights and bind a remote to the group, the state of the lights doesn't always update. They do however if I change the group state directly (i.e sending a group/set mqtt command)

I've got several zigbee groups that consist of several ikea based lights. In the same group are always the same kind of lights. I've also got different remotes ( 5Btn / 2Btn). The very same problem can be found in various constellations.  I also played around with the `retrieve_state` attribute (set to true) on the devices and/or the groups - without luck though.... 

## How to reproduce it

- I defined a Group consisting of the same type of ikea bulbs

    ```
    '4':
      friendly_name: Esstisch
      devices:
        - 0x5c0272fffe9d957d/1
        - 0x5c0272fffe9f261e/1
        - 0x5c0272fffea29e52/1
    ```

- I bound the Ikea Remote (5btn) to the group (also I made sure it's not bound to the default group according to the device setup page)

    Source Ep | Destination |  Clusters
    ------------ | ------------- | -------------
    1 | Coordinator | PowerCfg 
    1 | Esstisch | OnOff/LevelCtrl
    
- when I press the toggle button on the remote the lights all toggle, but their state is often not changed as I see it in the MQTT Topic. Since at least one light might not correctly toggle the group also stays in that state (which is what I'd also expect) 

- when I change state of the group (by sending a mqtt message or using Home-assistant UI) the lights behave correct

## Current Workaround/Hack
several seconds after I press the toggle button (on the 5-Btn-Remote) or the on-off button (on the 2-Btn-Remote) I'm going to fetch the current value of every light associated with that group.  Eventually the state of the group also changes in z2m after some seconds... I guess this hack works similar to the concept of  the  `retrieve_state`  attribute  but with a longer delay (3sec) in my case

**Home-Assitant Automation:**
```
alias: '[zigbee] sync  esstisch '
description: ''
use_blueprint:
  path: custom/z2m-ikea-remote-controls.yaml
  input:
    remote: sensor.esstisch_remote_action
    toggle:
      - delay: '00:00:03'
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/Esstisch Spot 01/get
          payload_template: '{"state": "", "color_temp": "", "brightness": ""}'
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/Esstisch Spot 02/get
          payload_template: '{"state": "", "color_temp": "", "brightness": ""}'
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/Esstisch Spot 03/get
          payload_template: '{"state": "", "color_temp": "", "brightness": ""}'
```
Blueprint: https://gist.github.com/pavax/32581f0d4ed860264159df1d79d8c849

## Debug info
Zigbee2MQTT version: 1.17.1 
Adapter hardware:  Conbee II
Integration: Home Assistant Addon
Adapter firmware version:  0x26680700
Remotes: E1524/E1810, E1743
Lights:  LED1537R6/LED1739R5, L1529, LED1837R5 
Groups: Tradfri_Scene (65289)

canton7, 2021-02-24:
I assume you mean that the lights themselves physically change, but their state as reported in their topic by Z2M doesn't?

I hit exactly this: even with a group of 3 lights, half the time only 2 would report a change in status when I used the remote. I solved it by setting `report: true` in the config, even though the docs say this is deprecated.

```
advanced:
  report: true
```

pavax, 2021-02-24:
> I assume you mean that the lights themselves physically change, but their state as reported in their topic by Z2M doesn't?

Exactly they physically change, but not their z2m state (in the topic / ui)

Yeah, I stumbled across this property but since it was highlighted as deprecated, I skipped it.

I tried to add in my configuration (within the home-assitant addon) and restarted it. Unfortunately, the problem still exits....  what is your conbee II adapter firmware ... Have you enabled any other configuration options (baudrate etc) ?

pavax, 2021-02-25:
Still playing around with the `report: true` setting. I unpaired the lights, I unpaired the remote and deleted the group. Then I joined the lights again and saw the following message for one of the lights:

```
Zigbee2MQTT:error 2021-02-25 20:06:01: Failed to setup reporting for '0x14b457fffe4b2708' - Error: Bind 0x14b457fffe4b2708/1 genLevelCtrl from '0x00212effff06dde3/1' failed (waiting for response TIMEOUT)
    at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:710:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:25:65)
```

pavax, 2021-03-02:
It looks like it works much more reliable in the newest z2m version: 1.18.0-1 / adapter: conbee II

Steps:

- deleted the flag `report: true`
- upgraded to the version 1.18.0-1.
- pay attention to the log file and note the devices that are causing Errors
- unpaired said lights
- paired the lights again
- assigned the lights again to their group

canton7, 2021-03-06:
I tried those steps and my Z2M isn't reporting the states for my lights at all. I've gone back to `report: true`.

closed 2021-04-13 by github-actions[bot]